linux4sam/at91bootstrap #32: Stuck in compile

Dark1X, 2016-04-15:
Hello, I am using the sama5d3_xplained. When I tied to run these command:
$ make mrproper
$ make at91sam9x5ekdf_linux_defconfig
$ make
I was stucked in "make at91sam9x5ekdf_linux_defconfig", I have changed this command to "make sama5d3_xplainedsd_uboot_defconfig", but it didn't help. Finally, I found "sama5d3_xplainedsd_uboot_defconfig" in "~/at91bootstrap/board/sama5d3_xplained", I tried copy all files in this folder to ~/at91bootstrap directory, then I tried to run "make sama5d3_xplainedsd_uboot_defconfig", but this is also not work. Help me please, many thx.

noglitch, 2016-04-15:
Documentation is here: http://www.at91.com/linux4sam/bin/view/Linux4SAM/AT91Bootstrap#Build_AT91Bootstrap_from_sources

at91sam9x5 is definitively not related to sama5d3 Xplained, it's another product so you mustn't use this defconfig file.

Dark1X, 2016-04-15:
OK. I will try it right now. Thanks again.

closed 2016-06-07 by noglitch