linux4sam/at91bootstrap #44: SAMA5D2 Xplained SD card boot fails

baruchsiach, 2016-10-06:
Current master `sama5d2_xplainedsd_uboot_defconfig` fails to load u-boot.bin as follows:

    AT91Bootstrap 3.8.6-00008-gb04fe6ffd399 (Thu Oct  6 15:29:25 IDT 2016)

    EEPROM: Loading AT24xx information ...
    EEPROM: BoardName | [Revid] | VendorName
      #0  SAMA5D2-XULT [AA1]      ATMEL-RF0

    EEPROM: Board sn: 0xd300000 revision: 0x200000

    SD/MMC: Image: Read file u-boot.bin to 0x26f00000
    *** FATFS: f_open, filename: [u-boot.bin]: error
    SD/MMC: Failed to load image

Switching back to `v3.8.6` fixed this issue for me. Successful boot log follows:

    AT91Bootstrap 3.8.6 (Thu Oct  6 17:13:00 IDT 2016)

    EEPROM: Loading AT24xx information ...
    EEPROM: BoardName | [Revid] | VendorName
      #0  SAMA5D2-XULT [AA1]      ATMEL-RF0

    EEPROM: Board sn: 0xd300000 revision: 0x200000

    SD/MMC: Image: Read file u-boot.bin to 0x26f00000
    SD: Card Capacity: High or Extended
    SD: Specification Version 3.0X
    SD/MMC: Done to load image

alexandrebelloni, 2016-10-06:
Does reverting https://github.com/linux4sam/at91bootstrap/commit/752daf7b759f25a6ba24b77592d032da9b01fe7a fix the issue?

baruchsiach, 2016-10-06:
That commit was my first suspect. Reverting 752daf7 doesn't help, however.

alexandrebelloni, 2016-10-06:
I came to the same conclusion but we needed to make sure as we can't reproduce the issue. Did you try bisecting?

baruchsiach, 2016-10-07:
For some reason I can't reproduce this issue now. The exact same image boots correctly.

```
AT91Bootstrap 3.8.6-00008-gb04fe6ffd399 (Thu Oct  6 15:29:25 IDT 2016)

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
  #0  SAMA5D2-XULT [AA1]      ATMEL-RF0

EEPROM: Board sn: 0xd300000 revision: 0x200000

SD/MMC: Image: Read file u-boot.bin to 0x26f00000
SD: Card Capacity: High or Extended
SD: Specification Version 3.0X
SD/MMC: Done to load image
```

closed 2016-10-07 by baruchsiach

alexandrebelloni, 2016-10-07:
Did you change the SDcard?

baruchsiach, 2016-10-07:
No. I used the same SD card for all my tests.

dumblob, 2021-07-07:
Just for reference - could someone post an example of a longer successful boot log of a buildroot image compiled with default settings (ideally for SAMA5D4 Xplained, but I think any SAMA5Dx is fine if it's an **SD card** boot log)?

On the wiki https://www.linux4sam.org/ I couldn't find any such complete log.

I'm having a trouble running either of the yocto/buildroot/openwrt **SD Card image** [demo archives](https://www.linux4sam.org/bin/view/Linux4SAM/Sama5d4XplainedMainPage#Demo_archives ) as well as running my own compiled buildroot from github.com/linux4sam (I've tried the latest tagged versions as well as the bleeding edge `master`). I always get the following boot log on my **SAMA5D4 Xplained**:
```
RomBOOT

AT91Bootstrap 3.9.0 (7. 7. 2021, 10:33:16 CEST)

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName

EEPROM: Using default information

EEPROM: Board sn: 0x1012420 revision: 0x680820

SD/MMC: Image: Read file u-boot.bin to 0x26f00000
SD: Card Capacity: Standard
SD: Specification Version 2.00
SD/MMC: Done to load image

```
And nothing else. Any ideas?

ehristev, 2021-07-07:
@dumblob you reached the point where bootstrap jumps to u-boot . 
Can you check if your u-boot.bin image is valid and correct ? It may crash very early thus you see nothing on the console.
Here is a boot log example:

`
AT91Bootstrap 3.9.2 (Fri May 29 01:16:13 EEST 2020)

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
  #0  SAMA5D2-XULT [AC1]      ATMEL-RFO
EEPROM: BoardDate | Year | Week
EEPROM:             2014    1

EEPROM: Board sn: 0xd300000 revision: 0x600000

SD/MMC: Image: Read file u-boot.bin to 0x26f00000
MMC: ADMA supported
SD: Card Capacity: High or Extended
SD: Specification Version 3.0X
SD/MMC: Done to load image
<debug_uart>

U-Boot 2020.01-linux4sam-2020.04 (May 29 2020 - 01:16:25 +0300)

CPU: SAMA5D27-CU`

Note that the line 
`<debug_uart>`
is printed by u-boot early console.

dumblob, 2021-07-07:
Thanks for the example boot log.

How could I check validity and correctness of the u-boot.bin image? Just for reference this is latest tagged buildroot from https://github.com/linux4sam with upstream default options after mounting an SD card with the flashed image:
```
$ LC_ALL=C ls -al /media/sdb1-usb-Mass_Storage_Dev/
total 4.6M
drwxr-xr-x 2 user00 user00  16K Jan  1  1970 .
drwxr-xr-x 3 root  root  4.0K Jul  7 13:01 ..
-rw-r--r-- 1 user00 user00  46K Jul  7  2021 at91-sama5d4_xplained.dtb
-rw-r--r-- 1 user00 user00  19K Jul  7  2021 boot.bin
-rw-r--r-- 1 user00 user00 519K Jul  7  2021 u-boot.bin
-rw-r--r-- 1 user00 user00 4.0M Jul  7  2021 zImage
$ wc /media/sdb1-usb-Mass_Storage_Dev/*
     44     298   46916 /media/sdb1-usb-Mass_Storage_Dev/at91-sama5d4_xplained.dtb
    140     909   18645 /media/sdb1-usb-Mass_Storage_Dev/boot.bin
   5427   20444  530864 /media/sdb1-usb-Mass_Storage_Dev/u-boot.bin
  11179   72122 4147824 /media/sdb1-usb-Mass_Storage_Dev/zImage
  16790   93773 4744249 total
```

ehristev, 2021-07-07:
i assume you can do a checksum from your boot media vs the image build by buildroot... 
but in your case if u-boot crashes at boot time, I begin to suspect something else, maybe there is something wrong with your board, and this is visible when u-boot starts and tries to perform something ( i don't know what...) 
do you have a different board to try ?

dumblob, 2021-07-07:
Thanks for an expert suggestion - yes, I have two other boards I didn't try yet (they're in a different package which can't be easily disassembled without destroying the package so I won't have serial console, but I'll try a blind test with some scripts simply persisting some information about a running system and see where this'll get me - if nowhere, then I'll break the packages and connect to right pads to access the serial console).

ehristev, 2021-07-07:
as you mention this, it looks like you don't have the sama5d4_xplained board but rather something else ?
you could try to debug u-boot to see what is happening, but you won't have even access to the early console as it something happens before that... so it would be low level initialization debugging. you could use a debugger like lauterbach or arm ds-5 to see what's going on.
you could also try to rebuild bootstrap with a configuration to load linux kernel directly and see if linux boots. or at least it might print something that would be of interest to you (could also enable DEBUG_LL and early printk in kernel)

dumblob, 2021-07-07:
Thanks - those are good ideas. Actually the board says it's sama5d4_xplained but when I checked some pictures it seems my boards are not identical but slightly different.

Looking at the other boards I found some contents on SD card and there was some old FAT partition with `boot.bin`. I've tried to swap my `boot.bin` with this one and voila, strangely it worked - it ran my new compiled u-boot (which actually got stuck when it was about to start Linux because of some DTB issue, but next to the old `boot.bin` I've found also an old DTB file which surprisingly worked with my new kernel and ran the whole system so that I could log in as root and do a quick lookaround).

Would you have any idea why the newest tagged at91boostrap doesn't work and this 5 years old one does? `cmp old/boot.bin new/boot.bin` shows lots of bytes and bisecting the commit might take quite some time (because my setup is not fully automated).

The old `boot.bin` reports this:
```
AT91Bootstrap 3.8-alpha7-00005-gbb52e9f (Tue Jun 21 13:21:35 EDT 2016)

All interrupts redirected to AIC
SD/MMC: Image: Read file u-boot.bin to 0x26f00000
SD: Card Capacity: Standard
SD: Specification Version 2.00
SD/MMC: Done to load image
;D0?�K7��{�KOp�C�0�

U-Boot 2021.04-linux4sam-2021.04-rc3 (Jul 07 2021 - 00:12:05 +0200)

CPU: SAMA5D42
Crystal frequency:       12 MHz
CPU clock        :      444 MHz
Master clock     :      148 MHz
DRAM:  512 MiB
NAND:  1024 MiB
MMC:   Atmel mci: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial@fc00c000
Out:   serial@fc00c000
Err:   serial@fc00c000
Net:   
Error: ethernet@f8020000 address not set.
No ethernet found.

Hit any key to stop autoboot:  0 
26594 bytes read in 9 ms (2.8 MiB/s)
4147824 bytes read in 527 ms (7.5 MiB/s)
## Flattened Device Tree blob at 21000000
   Booting using the fdt blob at 0x21000000
   Loading Device Tree to 3fb5a000, end 3fb637e1 ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.19.78-linux4sam-6.2 (user00@ntb) (gcc version 9.3.0 (Buildroot 2020.08-14-ge5a2a90)) #1 Wed Jul 7 10:53:09 CEST 2021
...
```
(notice the weird characters right after `SD/MMC: Done to load image`)

I'd like to get it work with the new DTB because after I disassembled the old one and compared to the new one, there are literally hundreds of important changes. One important thing I immediately noticed is that the booted system reports **no sound cards** despite there should be one if I'm not mistaken. Not sure it's related to DTB but it could... In which case I could try to bisect which commit made the DTB not working. Would you have some pointers where to start to find it out quicker :wink:?

And thanks a lot for your support - asking here on GitHub was a last-resort idea as I didn't expect anyone to even respond.

ehristev, 2021-07-08:
> Thanks - those are good ideas. Actually the board says it's sama5d4_xplained but when I checked some pictures it seems my boards are not identical but slightly different.

it may be that you have an old version of the board. it may be an earlier revision... I cannot say without comparing the boards.
> Looking at the other boards I found some contents on SD card and there was some old FAT partition with `boot.bin`. I've tried to swap my `boot.bin` with this one and voila, strangely it worked - it [...]

It may be that you have an old revision of the board that we do not have on our desk anymore such that we can try it . 
We may have done several improvements or fixes over the years, and this may break something which is not broken in the board that we have on our desk. So we could not find this regression when testing...
> 
> The old `boot.bin` reports this:
> 
> ```
> AT91Bootstrap 3.8-alpha7-00005-gbb52e9f (Tue Jun 21 13:21:35 EDT 2016)

This is indeed pretty old . 5 years old.

> All interrupts redirected to AIC
> SD/MMC: Image: Read file u-boot.bin to 0x26f00000
> SD: Card Capacity: Standard
> SD: Specification Version 2.00
> SD/MMC: Done to load image
> ;D0?�K7��{�KOp�C�0�
>
> U-Boot [...]
Yes, those are the early console of the U-boot, the string `<debug serial>` but it's printed in a wrong baud rate. This is because the clock tree of the CPU is not configured in the same way as U-boot expects . The early serial is hardcoded at a baudrate by assuming that the master clock of the CPU is e.g. 133 Mhz. If it's 100 Mhz or some other  frequency, the baud rate calculator will be wrong, thus you will see these bogus characters. After U-boot loads it's own devicetree, it will reconfigure all the clock tree and then the serial will be working fine. The issue is just with the very early debug console.

> I'd like to get it work with the new DTB because after I disassembled the old one and compared to the new one, there are literally hundreds of important changes. One important thing I immediately [...]
I cannot say for sure.. what I would suggest is to look for 'i2s' or 'sound' inside the DTB that works, and then try to use git blame to see which commit added the sound card. But from what I remember now (I am not sure, just the bottom of my head), sama5d4_xplained has no sound at all.

> 
> And thanks a lot for your support - asking here on GitHub was a last-resort idea as I didn't expect anyone to even respond.
I am the maintainer for at91bootstrap and u-boot-at91, so you reached the right place

ehristev, 2021-07-08:
> it may be that you have an old version of the board. it may be an earlier revision... I cannot say without comparing the boards.
>
>
> It may be that you have an old revision of the board that we do not [...]

ehristev, 2021-07-08:
> > Crystal frequency:       12 MHz
> > CPU clock        :      444 MHz
> > Master clock     :      148 MHz

about the debug serial,

u-boot expects CPU @ 600 Mhz, master clock at 200 I believe... thus the bad baud rate

ehristev, 2021-07-08:
> It may be that you have an old revision of the board that we do not have on our desk anymore such that we can try it .
> We may have done several improvements or fixes over the years, and this may [...]

If we can identify which board you have, I can test the latest bootstrap on a board similar to yours. Then I can debug to see what's wrong. 
There are plenty of changes since 3.8 until 3.10.2 or 4.0.0. I cannot pinpoint any specific reason until I can reproduce the issue and debug it

dumblob, 2021-07-08:
Sorry for the delay. Too busy these days.

Thanks for the baud rate explanation - that absolutely didn't appear to me! I'm glad it's nothing more serious.

I've taken a closer look at my s oard and it's not Sama5d4 Xplained. It's just a cut down version of it. It seems to mimick all the pin connections etc. but it lacks ethernet and many other things (the board is much smaller and has almost no connectors). On the other hand it has an LTE modem, audio codec `tlv320aic3254` (I've checked it and if I didn't miss anything Sama5d4 Xplained has no sound card/codec), and a GPS chip. The old Linux device tree file has these interesting sections:

```dts
	tobyl210 {
		compatible = "tobyl210";
		pinctrl-0 = <0x5c>;
		pinctrl-names = "default";
		irq-gpios = <0x08 0x04 0x01>;
	};

	sound {
		compatible = "atmel,tlv320aic32x4-audio";
		atmel,model = "tlv320aic32x4 @ ssc0";
		atmel,audio-routing = "IN1_R\0Mic\0IN1_L\0Mic";
		atmel,ssc-controller = <0x5d>;
		atmel,audio-codec = <0x5e>;
	};

	quectel-l86 {
		compatible = "quectel,l86";
		comm-usart = <0x5f>;
		reset-gpio = <0x38 0x07 0x00>;
		forceon-gpio = <0x38 0x06 0x00>;
		pinctrl-0 = <0x60>;
		pinctrl-names = "default";
		status = "disabled";
	};
```

But `modprobe snd-soc-tlv320aic32x4_spi; modprobe snd-soc-tlv320aic32x4_i2c` reports:
```
tlv320aic32x4 0-0018: Failed getting the mclk. The current implementation does not support the usage of this codec without mclk
tlv320aic32x4: probe of 0-0018 failed with error -2
```
It seems the kernel driver for the codec has changed and nowadays requires `mclk` unlike 5 years ago...

Could you maybe try to run the newest at91bootstrap on a ~6 years old Sama5d4 Xplained? Or if you already did, then the only option seems to be for me to bisect which version starts to fail for my boards.