linux4sam/at91bootstrap #62: Add packages after boot

AnneNJ, 2017-07-04:
Hello,

I know this may not be the best place to ask that but , I give it a try.
I am using a SAMA5D2 board.

I was wondering how I can add additionnal packages and functionnalities to my board. Since the demo Image provides a very basic version of Linux. The only package manager I have is opkg and there is no Atmel opkg repository specified in opkg.conf

Has someone ever tried something like this?

Thanks in adavance for your help.

Anne

noglitch, 2017-07-05:
Hi,

Your question is more related to the Yocto Project than AT91Bootstrap, so it's not the right issue tracker entry on our github.
Anyway, if you need to add packages to the basic demo, you can add them while building an "image" and you have examples of different images in our meta-atmel Yocto Project layer.
The other solution is that you build your own opkg repository and populate it with the packages that you compiled. It's described in the Yocto Project documentation and pretty easy to setup: once the packages are accessible through http for instance, that the list of packages is created and accessible and that the opkg.conf on the target got the proper information in relation with the repository that you've just setup, this remote mechanism to add or update packages works like a charm.

If you prefer to have everything already available and compiling, I would advice you to got the "Debian" route: https://eewiki.net/display/linuxonarm/SAMA5D2+Xplained+Ultra#SAMA5D2XplainedUltra-RootFileSystem

closed 2017-07-05 by noglitch

AnneNJ, 2017-07-06:
Thank you very much for your answer