linux4sam/at91bootstrap #71: Problem finding all lpddr2 memory

iveqy, 2017-09-24:
I'm having a SAMA5D26C with a LPDDR2 EDB1332BDBH-1. The CPU boots successfully ROMBOOT and is able to load at91bootstrap from QSPI memory. I get debug output from at91bootstrap, but I fail to load linux.

Investigating this further show me that the RAM initalization works fine but the CPU hangs when trying to load the device tree from flash into RAM.

So I add this loop:

volatile char * ptr = (char *)0x20000000;
while(1) {
    dbg_info("ptr: %x value: %c\n", ptr, *ptr);
    ptr += 1;
}
That gives me output as expected:

ptr: 0x20000000 value: U
ptr: 0x20000001 value: U
...
Up until:

...
ptr: 0x203ffffe value: U
ptr: 0x203fffff value:
ptr: 0x203fffff value:
ptr: 0x203fffff value:
ptr: 0x203fffff value:
ptr: 0x203fffff value:
ptr: 0x203fffff value:
It seems as I can't access memory above 4MB even though my RAM size is 128 MB. I've verified that I can write and read the written value from RAM for a value below 4MB.

Why isn't all physical memory accessible and and how come ptr isn't increased after 0x203ffffff?

wenyouyang, 2017-09-25:
Could you try to modify the ddrc configuration?
the Number of column bits: from 10 bits -> 9 bits, and the Number of Row Bits: from 14 bits -> 13 bits.

FredrikSvep, 2017-09-25:
Thanks for your answer. Yes we've. Here's the config code, commented out values are the original values. We also can't see any difference between using AT91C_DDRC2_DECOD_INTERLEAVED and not.

```
#if defined(CONFIG_LPDDR2)
static void lpddr2_reg_config(struct ddramc_register *ddramc_config)
{
        ddramc_config->mdr = (AT91C_DDRC2_DBW_32_BITS |
                              AT91C_DDRC2_MD_LPDDR2_SDRAM);

        ddramc_config->cr = (AT91C_DDRC2_NC_DDR9_SDR8 |
                             AT91C_DDRC2_NR_13 |
                             AT91C_DDRC2_CAS_3 |
                             //AT91C_DDRC2_ZQ_SHORT |
                             AT91C_DDRC2_ZQ_INIT |
                             AT91C_DDRC2_ENRDM_ENABLE |
                             AT91C_DDRC2_NB_BANKS_8 |
                             AT91C_DDRC2_UNAL_SUPPORTED);

        ddramc_config->lpddr2_lpr = AT91C_LPDDRC2_DS(0x03);

#ifdef CONFIG_BUS_SPEED_166MHZ
        /*  
         * The MT42128M32 refresh window: 32ms
         * Required number of REFRESH commands(MIN): 8192
         * (32ms / 8192) * 166MHz = 0x288.
         */
        //ddramc_config->rtr = 0x288;
        ddramc_config->rtr = 0x288;
        /* 90n short calibration: ZQCS */
        ddramc_config->tim_calr = AT91C_DDRC2_ZQCS(12);

        ddramc_config->t0pr = (AT91C_DDRC2_TRAS_(7) |
                               AT91C_DDRC2_TRCD_(3) |
                               AT91C_DDRC2_TWR_(3) |
                               AT91C_DDRC2_TRC_(11) |
                               AT91C_DDRC2_TRP_(4) |
                               AT91C_DDRC2_TRRD_(3) |
                               //AT91C_DDRC2_TRRD_(2) |
                               //AT91C_DDRC2_TWTR_(2) |
                               AT91C_DDRC2_TWTR_(3) |
                               AT91C_DDRC2_TMRD_(3));
                               //AT91C_DDRC2_TWTR_(2) |
                               //AT91C_DDRC2_TWTR_(3) |
                               //AT91C_DDRC2_TMRD_(3));

        //ddramc_config->t1pr = (AT91C_DDRC2_TRFC_(35) |
        ddramc_config->t1pr = (AT91C_DDRC2_TRFC_(22) |
                                //AT91C_DDRC2_TXSNR_(37) |
                                AT91C_DDRC2_TXSNR_(24) |
        //                      AT91C_DDRC2_TXSNR_(2) |
                                AT91C_DDRC2_TXSRD_(0) |
                                AT91C_DDRC2_TXP_(2));

        ddramc_config->t2pr = (AT91C_DDRC2_TXARD_(0) |
                               AT91C_DDRC2_TXARDS_(0) |
                              // AT91C_DDRC2_TRPA_(0) |
                               AT91C_DDRC2_TRPA_(3) |
                               AT91C_DDRC2_TRTP_(2) |
                               //AT91C_DDRC2_TFAW_(9));
                               AT91C_DDRC2_TFAW_(12));
        ddramc_config->cal_mr4r = AT91C_DDRC2_COUNT_CAL(0xC852);
#else
#error "No CLK setting defined"
#endif
}

```

wenyouyang, 2017-09-25:
It seems the configuration okay, we should suspect the hardware problem, maybe a address line is wrong connected.

FredrikSvep, 2017-09-25:
@wenyouyang We haven't found any hardware error and it seems strange that a hardware error would still give us access to 4 MB of RAM. That should indicate the we can talk to the memory OK.

Looking at the address table in https://en.wikipedia.org/wiki/Mobile_DDR I fail to find a way that an address pin could be faulty connected that would cause a 4 MB only behavior. Of course, I can be wrong here.

Is this code really tested and used? The sama5d2 explained board is using ddr3 memory so it wouldn't  surprise me if this (the lpddr2 code) is just old code laying around.

junminlin, 2017-09-25:
@FredrikSvep Is it possible for you to submit your schematics for review at Microchip's support channel: http://www.microchip.com/support/hottopics.aspx

closed 2017-09-25 by junminlin

iveqy, 2017-10-01:
The problem was that TrustZone was disabled