linux4sam/at91bootstrap #76: Newer GCC's (>= 6.x) create faulty images

stnkuehn, 2018-01-25:
A GCC with version >= 6. x seems to create faulty images (at least for an AT91SAMA5D36). The problem is that external global variables are incorrectly linked.

Example:

We add in common. c the line 

`int pahx7Iok;`

Afterwards we write in main. c

`extern int pahx7Iok;`

and 

`dbg_info("=> %p\n", &pahx7Iok);`

This produces an output saying that the pointer is zero. And the line

pahx7Iok = 0;

would cause a crash.

wenyouyang, 2018-01-26:
What version of toolchain did you use?
I tested two versions of toolchain from Linaro(i.e. gcc version 7.2.1 20171011 (Linaro GCC 7.2-2017.11) ,gcc version 6.1.1 20160711 (Linaro GCC 6.1-2016.08))
AT91bootstrap works on the sama5d3ek board.

So, please share your toolchain, we can test on our side.

stnkuehn, 2018-01-26:
Here is the build output (I build on Debian 9 Stretch):

```
CC
========
/usr/bin/arm-linux-gnueabi-gcc 6.3.0

as FLAGS
========
-g -Os -Wall -Iboard/stampa5d3x -Iinclude -DJUMP_ADDR=0x22000000 -DTOP_OF_MEMORY=0x310000 -DMACH_TYPE=9999 -Dstampa5d3x -DMACH_TYPE=9999 -DTOP_OF_MEMORY=0x310000 -DCRYSTAL_12_000MHZ -DSAMA5D3X -mcpu=cortex-a5 -mtune=cortex-a5 -DCONFIG_STAMPA5D3X

gcc FLAGS
=========
-nostdinc -isystem /usr/lib/gcc-cross/arm-linux-gnueabi/6/include -ffunction-sections -g -Os -Wall -std=c99 -fno-stack-protector -fno-common -Iboard/stampa5d3x -Iinclude -Ifs/include -I/home/sk/Arbeit/git/at91bootstrap/config/at91bootstrap-config -DAT91BOOTSTRAP_VERSION="3.7.2-00032-gc1b03df" -DCOMPILE_TIME="Fri 26 Jan 11:04:44 CET 2018" -DIMG_ADDRESS= -DIMG_SIZE= -DJUMP_ADDR=0x22000000 -DOF_OFFSET= -DOF_ADDRESS=0x21000000 -DMEM_BANK=0x20000000 -DMEM_SIZE=0x10000000 -DIMAGE_NAME="linux.bin" -DLINUX_KERNEL_ARG_STRING="" -DTOP_OF_MEMORY=0x310000 -DMACH_TYPE=9999 -DCONFIG_DEBUG -DBANNER="

AT91Bootstrap " AT91BOOTSTRAP_VERSION " (" COMPILE_TIME ")

" -DCONFIG_HW_DISPLAY_BANNER -DCONFIG_HW_INIT -Dstampa5d3x -DMACH_TYPE=9999 -DTOP_OF_MEMORY=0x310000 -DCRYSTAL_12_000MHZ -DSAMA5D3X -mcpu=cortex-a5 -mtune=cortex-a5 -DCONFIG_STAMPA5D3X -DCONFIG_SCLK -DCONFIG_CRYSTAL_12_000MHZ -DCONFIG_CPU_CLK_528MHZ -DCONFIG_BUS_SPEED_133MHZ -DCONFIG_HAS_PIO3 -DCONFIG_MMC_SUPPORT -DCONFIG_DDRC -DCONFIG_LPDDR -DCONFIG_RAM_256MB -DCONFIG_DATAFLASH_ALWAYSON -DCONFIG_SDCARD -DCONFIG_LOAD_LINUX -DCONFIG_LINUX_IMAGE -DCONFIG_SDCARD_HS -DCONFIG_OF_LIBFDT -DCONFIG_SMALL_DATAFLASH -DAT91C_SPI_CLK=33000000 -DAT91C_SPI_PCS_DATAFLASH=AT91C_SPI_PCS0_DATAFLASH -DBOOTSTRAP_DEBUG_LEVEL=DEBUG_INFO -DCONFIG_DISABLE_WATCHDOG -DCONFIG_LONG_FILENAME

ld FLAGS
========
-nostartfiles -Map=/home/sk/Arbeit/git/at91bootstrap/binaries/stampa5d3x-sdcardboot-linux-linux.bin-dt-3.7.2.map --cref -static -T elf32-littlearm.lds --gc-sections -Ttext 0x300000

  AS        /home/sk/Arbeit/git/at91bootstrap/crt0_gnu.S
  CC        /home/sk/Arbeit/git/at91bootstrap/main.c
  CC        /home/sk/Arbeit/git/at91bootstrap/board/stampa5d3x/stampa5d3x.c
  CC        /home/sk/Arbeit/git/at91bootstrap/lib/string.c
  CC        /home/sk/Arbeit/git/at91bootstrap/lib/eabi_utils.c
  CC        /home/sk/Arbeit/git/at91bootstrap/lib/div.c
  CC        /home/sk/Arbeit/git/at91bootstrap/lib/fdt.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/debug.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/at91_slowclk.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/at91_pio.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/pmc.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/at91_pit.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/at91_wdt.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/at91_usart.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/at91_rstc.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/ddramc.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/at91_mci.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/mci_media.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/sdcard.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/at91_spi.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/dataflash.c
  CC        /home/sk/Arbeit/git/at91bootstrap/driver/load_kernel.c
  CC        /home/sk/Arbeit/git/at91bootstrap/fs/src/ff.c
  CC        /home/sk/Arbeit/git/at91bootstrap/fs/src/diskio.c
  CC        /home/sk/Arbeit/git/at91bootstrap/fs/src/option/ccsbcs.c
  LD        stampa5d3x-sdcardboot-linux-linux.bin-dt-3.7.2.elf
Size of stampa5d3x-sdcardboot-linux-linux.bin-dt-3.7.2.bin is 22504 bytes
[Succeeded] It's OK to fit into SRAM area
[Attention] The space left for stack is 43032 
```

The problem also occurs with the current version of at91bootstrap.

wenyouyang, 2018-01-29:
Using the toolchain generated from Buildroot 2017.08.2 tag gcc version 6.4.0,
built the AT91Bootstrap for sama5d2 Xplained board, flashed to the board, it booted okay.

I am curious why you are doubting AT91Bootstrap source code.

closed 2018-01-29 by wenyouyang