linux4sam/at91bootstrap #84: blinking led program is not working with Bootstrap code.

tejasbhumkar, 2018-05-07:
Hello sir..
I'm using a AT91SAM9X35-evm BOARD and i'm loading simple led blinking program on 0x00 position through sam-ba software it is working fine...but when i use your bootstrap then program won't work.i'm loading bootstrap program at 0x00 position and led program at 4200 location through sam-ba software ..is their any need to change location of bootstrap program and led application program????......
**** :     I clone bootstrap repository...after that i follow your step to generate binary....i generate binary for serial flash.

ehristev, 2018-05-07:
Can you please give me a little more information:
Which exact at91bootstrap configuration did you use ?
Did you modify anything inside the at91bootstrap ?
Where did you get the blinking led program from ? Any link to it perhaps?
Thanks. I will be able to assist you once you provide me with more information.

tejasbhumkar, 2018-05-07:
thanks for the quik reply @ehristev..
I have attached one doc in that i mentioned from where i clone the code..
1) i'm compiling code for data flash..once  got binary  i'm loading that binary at 0x00 position and application program at 0x4200 position through sam-ba software
2) I got led code in keil  software package. 
3) I have not modify anything inside bootstrap code
4) i'm trying the same on at91sam9x35 evaluation board..
5) reason why i'm trying with bootstrap is that now my led program size is around 4kb but when i wanted to load the code above 24kb i need bootloader(as per datasheet at91sam9x35).
[image]

ehristev, 2018-05-07:
So, which one of those you are building ? 
sama5d3xekdf_*
linux, or u-boot?
As you can see, either of these configurations will have at91bootstrap load U-boot, or linux, not some other program that you load at 0x4200. My thinking is that bootstrap has no idea that you want to run some program based at 0x4200.
Is this correct?

tejasbhumkar, 2018-05-07:
Hello @ehristev 
i'm building for it  at91sam9x5-ek as i'm using an at91sam9x35 evaluation board.....i'm not doing a u-boot building...i'm doing following steps:
1git clone git://github.com/linux4sam/at91bootstrap.git
2)cd at91bootstrap/
3)make mrproper
4) make atsam9x5ekdf_uboot_defconfig
5)make menuconfig
6)make
7) If the building process is successful, the final .bin image is binaries/at91bootstrap.bin.

yes! @ehristev u are correct bootstarp have no idea until and unless we are not using correct location for the application code...but thing is that we use bootstrap program to download application program from serial flash to nandflash or SDRAM..and execute it there.

please correct me if i'm wrong.

ehristev, 2018-05-07:
If you are building the U-boot version of bootstrap, then, bootstrap will expect to load from dataflash at a specific address . Are you writing your program binary to dataflash ? Does bootstrap detect dataflash correctly? 
Do you have a console output from your board ?

tejasbhumkar, 2018-05-08:
Sorry for late replay @ehristev 
1)yes!, i wanted to know that specific address..
2)yes i'm writing my program binary to dataflash....
3)yes Bootstrap detect data flash correctly...
4)yes i'm getting console output form my board.

ehristev, 2018-05-08:
As it looks to me, AT91Bootstrap will load the U-boot image from dataflash from address according to this configuration:

```
config CONFIG_IMG_ADDRESS                                                        
        string "Flash Offset for U-Boot"                                         
        depends on CONFIG_DATAFLASH || CONFIG_FLASH || CONFIG_NANDFLASH          
        default "0x00008000" if CONFIG_FLASH                                     
        default "0x00008000" if CONFIG_DATAFLASH                                 
        default "0x00040000" if CONFIG_NANDFLASH                                 
        default "0x00000000" if CONFIG_SDCARD                                    
        help  
```

So, either use _make menuconfig_ to change this to your preferred value, or put your binary where the at91bootstrap is loading from.
On a side note, at91bootstrap might check a **magic value** on loading the image to see if it's a real U-boot image. Make sure you bypass that or have it there, if it's the case. 
Let me know if it worked out.
Good luck!

tejasbhumkar, 2018-05-09:
Hello @ehristev !!!
I try with the same addresses. what i was do is at 0x00 location i load second level bootloader(for data flash) and at 8000 location i load blink led program(toggling led program)..but i'm not getting correct output.

tejasbhumkar, 2018-05-10:
YES!! @ehristev  inside data-flash at 0x00 position i'm writing second level boot-loader of data flash and at 0x8000 location of DATA-FLASH i'm writing my blinky led program.

ehristev, 2018-05-10:
Can you please copy paste your console output ? Thanks.

tejasbhumkar, 2018-05-10:
Hi @ehristev 
[image]

Please find the attached doc.

Thank you!!!!

ehristev, 2018-05-10:
So, it looks your program is loaded from dataflash to external RAM. Do you use console in your program ? To try to print out something to see if your program runs or not ?

tejasbhumkar, 2018-05-10:
Hi @ehristev.....
I'm trying to add printf statement in simple led program but i'm not getting the prints. I think their is need of printf source code so their i can mention debug port pins, that source code i don't have. if you have any code related to same please let me know.

ehristev, 2018-05-10:
First of all, printf would print in your debugger console, not on serial output. To do that, you need to have something similar with a serial driver (look in bootstrap code, there is one I believe in atmel_serial.c).
Second, is your binary position independent code ? Is it aware that it has to run from 0x20000000 ? When you don't use At91bootstrap, it would run from SRAM at a different address. 
Third, did you try to compile and write a U-boot for your board at that address in dataflash, and see if U-boot will start correctly ? If U-boot runs, then there is a problem with your code.
As it looks to me, bootstrap is loading your file correctly, but something is wrong there. You need to debug, have a way to figure out if your code is actually running , and what is happening in there. 
Look at the simple serial driver and try to write to the serial device from your program. If it works, then your program is indeed running.

ehristev, 2018-05-16:
Closing issue as fixed because lack of response. 
Reopen if further assistance is needed.

closed 2018-05-16 by ehristev