linux4sam/at91bootstrap #86: Why the pin NANDRDY is not initialized in nandflash_hw_init()?

antonsviridenko, 2018-10-19:
https://github.com/linux4sam/at91bootstrap/blob/master/board/sama5d2_ptc_ek/sama5d2_ptc_ek.c#L375

I was unable to grep NANDRDY in at91booststrap sources anywhere, but it seems this pin is required for NAND flash controller operations. Is it missing deliberately or by mistake?

Quoting SAMA5D2 and NAND chip datasheets:
``
NANDRDY NAND Flash Ready/Busy Input –
``
``
READY/BUSY OUTPUT
The R/B# output indicates the status of the device operation. When low, it indicates
that a program, erase or random read operation is in progress and returns to high
state upon completion. It is an open drain output and does not float to high-z
condition when the chip is deselected or when outputs are disabled.
``

noglitch, 2018-10-21:
Hi,
To my knowledge the RDY/BSY# pin is not a requirement from NAND flash chips.

ehristev, 2018-10-22:
Looking at the board schematic, this pin is connected to PC8 I/O pin, so , the pin is available to the SoC.
I looked over this [1] Micron documentation w.r.t. NAND memories, and there are two possible methods of checking the flashes: polling status or this R/B# pin.
There are advantages and disadvantages for using any of the methods. 
Does this answer your question @antonsviridenko ?
It is likely to use in the future this feature in At91Bootstrap and we might consider implementing.

[1] https://www.micron.com/~/media/documents/products/technical-note/nand-flash/tn2913_ready_busy.pdf

antonsviridenko, 2018-10-24:
ok, thank you for explanation. I guess that NANDRDY not being initialized in at91bootstrap does not prevent using it in later stages, in u-boot and Linux, right?

ehristev, 2018-10-25:
@antonsviridenko that's correct. It is a GPIO , and Linux and U-boot will completely reset all the controllers for these, since they cannot rely on the initial bootloader for any sort of initialization...
When the linux driver probes, it will initialize it's hardware accordingly.

closed 2018-10-25 by antonsviridenko