linux4sam/at91bootstrap #90: Are configurations with "mem=" in kernel command line known to work with recent kernels (>=4.9)?

antonsviridenko, 2018-11-16:
Are you aware if following configurations exist and successfully work in production?
Like this one
https://github.com/linux4sam/at91bootstrap/blob/8b9064beb6169f2c3be089fa99c5080b2182639f/board/at91sam9261ek/Config.in.linux_arg#L2

and others with "mem=" in command line. Btw, why "mem=NN" is used when RAM size can be set in the device tree?

- at91bootstrap boots Linux directly, without u-boot

- Linux version is 4.9 or higher

- Kernel command line is passed by at91bootstrap, i.e. CONFIG_LINUX_KERNEL_ARG_STRING is enabled

- SoC is SAMA5D2

I observe kernel crashes when "mem=64M" is appended to command line, please see this post

http://lists.infradead.org/pipermail/linux-arm-kernel/2018-November/613095.html

ehristev, 2018-11-16:
Hi Anton,

Thank you for reporting this,

It is possible that the DTB memory node is modified by the bootloader, but I am not sure.
I will have a look and let you know of my findings.

PS. The kernel cmd line in general is used to easily override the DT nodes... such that you dont need to rebuild the DTB, just tweak from cmd line. Cmdline takes precedence over the DT nodes
Eugen

antonsviridenko, 2018-11-16:
> It is possible that the DTB memory node is modified by the bootloader, but I am not sure.
I will have a look and let you know of my findings.

Yes, according to sources, memory size is passed by modifying DTB:
https://github.com/linux4sam/at91bootstrap/blob/master/driver/load_kernel.c#L85

> PS. The kernel cmd line in general is used to easily override the DT nodes... such that you dont need to rebuild the DTB, just tweak from cmd line. Cmdline takes precedence over the DT nodes

I expect this behavior too