linux4sam/at91bootstrap #98: problem loading uboot from emmc

ChevalierDeBalibari, 2019-07-25:
Hello, 
we have a board based on SAMA5D27-SOM1-EK, (SAMA5D27-SOM1 + our baseboard) and we have SD card slot connected to MMC1 and eMMC on MMC0 interface. We have aplan to use eMMC on our final system and remove SD slot. 

Problem is that we can't boot from eMMC:

- at91bootstrap is compiled for  mmc0 interface
- u-boot is compiled for mmc0 interface

eMMC is partitioned via linux running on SD card using fdisk and mkfs:

```
/dev/mmcblk0: PTUUID="af1781b8" PTTYPE="dos"
/dev/mmcblk0p1: SEC_TYPE="msdos" UUID="7927-AEB6" TYPE="vfat" PARTUUID="af1781b8-01"
/dev/mmcblk0p2: UUID="c37028ac-7bda-4392-a260-93f2148fd5fa" TYPE="ext4" PARTUUID="af1781b8-02"

mmcblk0      179:0    0  7.3G  0 disk 
├─mmcblk0p1  179:1    0  256M  0 part /mnt/emmc1
└─mmcblk0p2  179:2    0    7G  0 part /mnt/emmc2

```

When we try to boot from eMMC, we are getting next message:

```
AT91Bootstrap 3.8.12-00001-g9a4293f (чет, 25. јул 2019.  13:32:32 CEST)

All interrupts redirected to AIC
SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image
```

Conncetions between eMMC and MCU (PA0 ~ PA5 and PA10 for MMC_RST):

[image]

ehristev, 2019-07-25:
Can you enable CONFIG_DEBUG_VERY_LOUD, and also update your bootstrap to 3.8.13 
then copy the boot log ?

Thanks,
Eugen

ChevalierDeBalibari, 2019-07-25:
```
reboot: Power down
RomBOOT
 ba_offset = 0xb ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf000c000: 0x10 0x300511 0xd00139 0x22239337
0xf000c010: 0x2c81716 0x82482 0x33338 0x10000
0xf000c020: 0x16 0x50008 0x0 0x0
0xf000c030: 0x6 0xf6504 0x0 0x0
0xf000c040: 0x0 0x0 0x0 0x0
0xf000c050: 0x0 0x0 0x0 0x1
0xf000c060: 0x0 0x0 0x0 0x0
0xf000c070: 0x0 0x0 0x0 0x0
0xf000c080: 0x0 0x0 0x0 0x0
0xf000c090: 0x0 0x0 0x0 0x0
0xf000c0a0: 0x0 0x0 0x0 0x0
0xf000c0b0: 0x0 0x0 0x1 0x0
0xf000c0c0: 0x0 0x0 0x0 0x0
0xf000c0d0: 0x0 0x0 0x0 0x0
0xf000c0e0: 0x0 0x0 0x0 0x4000
0xf000c0f0: 0x484d5044 0x44524320 0x0 0x20301
0xf000c100: 0x0 0x0 0x4040404 0x3030303
0xf000c110: 0x0 0x1 0x3f03f01 0x0
0xf000c120: 0x0 0x0 0xc00 0xc00
0xf000c130: 0xc00 0xc00 0xd00 0xd00
0xf000c140: 0xd00 0xd00 0x10 0x10
0xf000c150: 0x10 0x10 0xf 0x0

AT91Bootstrap 3.8.12-00001-g9a4293f (чет, 25. јул 2019.  15:35:46 CEST)

All interrupts redirected to AIC
SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image
```

ChevalierDeBalibari, 2019-07-25:
Here you go

```
 ba_offset = 0xb ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf000c000: 0x10 0x3004ff 0xd00139 0x22239337
0xf000c010: 0x2c81716 0x82482 0x33338 0x10000
0xf000c020: 0x16 0x50008 0x0 0x0
0xf000c030: 0x6 0xf6504 0x0 0x0
0xf000c040: 0x0 0x0 0x0 0x0
0xf000c050: 0x0 0x0 0x0 0x1
0xf000c060: 0x0 0x0 0x0 0x0
0xf000c070: 0x0 0x0 0x0 0x0
0xf000c080: 0x0 0x0 0x0 0x0
0xf000c090: 0x0 0x0 0x0 0x0
0xf000c0a0: 0x0 0x0 0x0 0x0
0xf000c0b0: 0x0 0x0 0x1 0x0
0xf000c0c0: 0x0 0x0 0x0 0x0
0xf000c0d0: 0x0 0x0 0x0 0x0
0xf000c0e0: 0x0 0x0 0x0 0x4000
0xf000c0f0: 0x484d5044 0x44524320 0x0 0x20301
0xf000c100: 0x0 0x0 0x4040404 0x3030303
0xf000c110: 0x0 0x1 0x3e03f02 0x0
0xf000c120: 0x0 0x0 0xc00 0xc00
0xf000c130: 0xc00 0xc00 0xd00 0xd00
0xf000c140: 0xd00 0xd00 0x10 0x10
0xf000c150: 0x10 0x10 0xf 0x0
Applying VDDSDMMC errata to ID: 0x33

AT91Bootstrap 3.8.13 (чет, 25. јул 2019.  16:09:34 CEST)

All interrupts redirected to AIC
SDHC: fix in place for SAMA5D2 SoM VDDSDMMC over-consumption errata
SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
SDHC: Timeout waiting for command complete
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image
```

Thanks!

ehristev, 2019-07-25:
Do you have a ticket opened on microchip support ?

As it looks to me, either you are using an SD-card (not eMMC), or your eMMC is not responding whatsoever to anything... is the voltage OK ? I do not see the card detect pin on your schematic.
Did you run the MPUcheck utility ? they have it on microchip support

ehristev, 2019-07-25:
As I see it, card identification fails... as if your card/eMMC is dead in the waters.

ehristev, 2019-07-25:
Where is at91bootstrap loaded from ? 
Do you load it from SDMMC0 or SDMMC1 ? How do you write it there ?

ChevalierDeBalibari, 2019-07-26:
No, I'm sure that I'm using eMMC (on SDMMC0 interface, SDMMC1 is standard microSD card slot).

According to documentation, ROM code tries to boot from SDMMC1 first, than from SDMMC0 and than it pass to other interfaces. So, when the SD card is inserted it boots from SDMMC1, and this is working. Without SD card inserted, ROM code loads at91bootstrap from eMMC, and it hangs after that.

Also, I'm sure that eMMC is working because i can access it once the system is booted from SDMMC1. I can create partitions, format them, mount fs, copy files and so on.

mattwood2000, 2019-07-26:
what rev of the silicon are you using, rev b, or rev c?

ehristev, 2019-07-26:
Try with this patch, again with debug very loud activated.
Check what messages are printed

diff --git a/driver/mci_media.c b/driver/mci_media.c
index 0941bf3..61d0cf9 100644
--- a/driver/mci_media.c
+++ b/driver/mci_media.c
@@ -906,8 +906,10 @@ static int sdcard_identification(struct sd_card *sdcard)
                 */
                dbg_info("Unusable Card\n");
                return -1;
-       } else
+       } else {
+               dbg_info("Unknown card error\n");
                return ret;
+       }
 
        sdcard->highcapacity_card = (sdcard->reg->ocr & OCR_HCR_CCS) ? 1 : 0;
 
@@ -942,6 +944,8 @@ static int sdcard_identification(struct sd_card *sdcard)
 
        sdcard->read_bl_len = DEFAULT_SD_BLOCK_LEN;
 
+       dbg_info("operating conditions ok");
+
        return 0;
 }
 
@@ -1131,15 +1135,19 @@ int sdcard_initialize(void)
 
        /* Card Indentification Mode */
        ret = sdcard_identification(sdcard);
-       if (ret)
+       if (ret) {
+               dbg_info("error identifying card\n");
                return ret;
+       }
 
        if (sdcard->card_type == CARD_TYPE_SD)
                ret = sd_initialization(sdcard);
        else
                ret = mmc_initialization(sdcard);
-       if (ret)
+       if (ret) {
+               dbg_info("error initializing mmc/sd\n");
                return ret;
+       }
 
        return 0;
 }

ChevalierDeBalibari, 2019-07-31:
Hi, here is the output:

```RomBOOT
 ba_offset = 0xb ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf000c000: 0x10 0x3004ff 0xd00139 0x22239337
0xf000c010: 0x2c81716 0x82482 0x33338 0x10000
0xf000c020: 0x16 0x50008 0x0 0x0
0xf000c030: 0x6 0xf6504 0x0 0x0
0xf000c040: 0x0 0x0 0x0 0x0
0xf000c050: 0x0 0x0 0x0 0x1
0xf000c060: 0x0 0x0 0x0 0x0
0xf000c070: 0x0 0x0 0x0 0x0
0xf000c080: 0x0 0x0 0x0 0x0
0xf000c090: 0x0 0x0 0x0 0x0
0xf000c0a0: 0x0 0x0 0x0 0x0
0xf000c0b0: 0x0 0x0 0x1 0x0
0xf000c0c0: 0x0 0x0 0x0 0x0
0xf000c0d0: 0x0 0x0 0x0 0x0
0xf000c0e0: 0x0 0x0 0x0 0x4000
0xf000c0f0: 0x484d5044 0x44524320 0x0 0x20301
0xf000c100: 0x0 0x0 0x4040404 0x3030303
0xf000c110: 0x0 0x1 0x3f03f02 0x0
0xf000c120: 0x0 0x0 0xc00 0xc00
0xf000c130: 0xc00 0xc00 0xd00 0xd00
0xf000c140: 0xd00 0xd00 0x10 0x10
0xf000c150: 0x10 0x10 0xf 0x0
Applying VDDSDMMC errata to ID: 0x33

AT91Bootstrap 3.8.13-dirty (Wed 31 Jul 2019 10:53:43 AM CEST)

All interrupts redirected to AIC
SDHC: fix in place for SAMA5D2 SoM VDDSDMMC over-consumption errata
SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
SDHC: Timeout waiting for command complete
error identifying card
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image```

P.S
```
+       dbg_info("operating conditions ok");
```

you forgot `\n`

ehristev, 2019-07-31:
I have pushed a commit on the master branch that enables more debug messages. Can you pull that commit and retest and copy paste the messages ? (with debug very loud).
I have a feeling that your eMMC 's CID register is being read as 0xffff.....fffff , which is not right.
I have one eMMC that behaves like this. I do not know yet why.
I want to know if your eMMC has the same issue

ChevalierDeBalibari, 2019-07-31:
You are right...
```
 ba_offset = 0xb ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf000c000: 0x10 0x3004ff 0xd00139 0x22239337
0xf000c010: 0x2c81716 0x82482 0x33338 0x10000
0xf000c020: 0x16 0x50008 0x0 0x0
0xf000c030: 0x6 0xf6504 0x0 0x0
0xf000c040: 0x0 0x0 0x0 0x0
0xf000c050: 0x0 0x0 0x0 0x1
0xf000c060: 0x0 0x0 0x0 0x0
0xf000c070: 0x0 0x0 0x0 0x0
0xf000c080: 0x0 0x0 0x0 0x0
0xf000c090: 0x0 0x0 0x0 0x0
0xf000c0a0: 0x0 0x0 0x0 0x0
0xf000c0b0: 0x0 0x0 0x1 0x0
0xf000c0c0: 0x0 0x0 0x0 0x0
0xf000c0d0: 0x0 0x0 0x0 0x0
0xf000c0e0: 0x0 0x0 0x0 0x4000
0xf000c0f0: 0x484d5044 0x44524320 0x0 0x20301
0xf000c100: 0x0 0x0 0x4040404 0x3030303
0xf000c110: 0x0 0x1 0x3e03f02 0x0
0xf000c120: 0x0 0x0 0xc00 0xc00
0xf000c130: 0xc00 0xc00 0xd00 0xd00
0xf000c140: 0xd00 0xd00 0x10 0x10
0xf000c150: 0x10 0x10 0xf 0x0
Applying VDDSDMMC errata to ID: 0x33

AT91Bootstrap 3.8.13-dirty (Wed 31 Jul 2019 10:53:43 AM CEST)

All interrupts redirected to AIC
SDHC: fix in place for SAMA5D2 SoM VDDSDMMC over-consumption errata
SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
mmc_verify_operating_condition
mmc_verify_operating_condition success OCR = 0xc0ff8080
Card type is MMC
sd card identified with CID = 0x1fffffff 0xffffffff 0xffffffff 0xffffffff
SDHC: Timeout waiting for command complete
sd_cmd_send_relative_addr failed
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image
```

ehristev, 2019-07-31:
Is your eMMC a toshiba ?
I have the same issue on one of mine.
However, the exact same eMMC on another of our boards (not som1 EK) , works fine, with the same at91bootstrap.
I have a feeling there is something wrong with this setup, the eMMC does not like something, but I did not figure out what. 
Maybe you can do more debugging on your side ?
The CID should be something else not 0xfff... it's like the card is keeping the line to 3v3 when transmitting. maybe it's busy signaling or something ?

ChevalierDeBalibari, 2019-07-31:
Yep it is toshiba: THGBMJG6C1LBAU7

Maybe RESET sequence?

[image]

ehristev, 2019-07-31:
The card does provide a valid and functional OCR. So , during the initialization sequence, negotiated operating conditions are done OK.
Next step is the host to issue SEND_ALL_CID
and it looks like something is wrong when reading CID from the card. Maybe the card is not ready ? But once the OCR is sent, the card should enter READY state. There is a possibility that the card finds some incompatibility with the controller, in which case goes to Inactive state? Maybe the card goes to INACTIVE ?

ChevalierDeBalibari, 2019-07-31:
I tried to repeat mmc_verify_operating_condition command twice and it fails at second one.

ehristev, 2019-07-31:
So it looks like the eMMC goes to some weird state. Perhaps inactive state. 
what OCR you receive on the second run ?
For me it's also weird that the OCR is now 
mmc_verify_operating_condition success OCR = 0xc07f8000

and this OCR does not comply with the eMMC standard. your OCR is correct

ehristev, 2019-07-31:
Hi Chevalier,

I believe this will solve your issue completely:

`diff --git a/driver/sdhc.c b/driver/sdhc.c
index 656bca6..1c27823 100644
--- a/driver/sdhc.c
+++ b/driver/sdhc.c
@@ -386,7 +386,7 @@ static int sdhc_set_clock(struct sd_card *sdcard, unsigned int clock)
                }
                clk_div >>= 1;
        }
-
+       clk_div--;
        sdhc_writew(SDMMC_CCR, sdhc_readw(SDMMC_CCR) & ~SDMMC_CCR_SDCLKEN);
 
        sdhc_writew(SDMMC_CCR, SDMMC_CCR_INTCLKEN | clk_gen_sel
@@ -577,7 +577,7 @@ static int sdhc_init(struct sd_card *sdcard)
        sdhc_writew(SDMMC_NISIER, 0);
        sdhc_writew(SDMMC_EISIER, 0);
 
-       sdhc_set_clock(sdcard, 400000);
+       sdhc_set_clock(sdcard, 200000);
        sdhc_set_bus_width(sdcard, 1);
 
        return 0;
`

Also, you may need this depending on your board:

`diff --git a/board/sama5d27_som1_ek/sama5d27_som1_ek.c b/board/sama5d27_som1_ek/sama5d27_som1_ek.c
index 0297eed..7ee184a 100644
--- a/board/sama5d27_som1_ek/sama5d27_som1_ek.c
+++ b/board/sama5d27_som1_ek/sama5d27_som1_ek.c
@@ -517,16 +517,16 @@ void at91_sdhc_hw_init(void)
 
 #ifdef CONFIG_SDHC0
        const struct pio_desc sdmmc_pins[] = {
-               {"SDMMC0_CK",   AT91C_PIN_PA(0), 0, PIO_DEFAULT, PIO_PERIPH_A},
-               {"SDMMC0_CMD",  AT91C_PIN_PA(1), 0, PIO_DEFAULT, PIO_PERIPH_A},
-               {"SDMMC0_DAT0", AT91C_PIN_PA(2), 0, PIO_DEFAULT, PIO_PERIPH_A},
-               {"SDMMC0_DAT1", AT91C_PIN_PA(3), 0, PIO_DEFAULT, PIO_PERIPH_A},
-               {"SDMMC0_DAT2", AT91C_PIN_PA(4), 0, PIO_DEFAULT, PIO_PERIPH_A},
-               {"SDMMC0_DAT3", AT91C_PIN_PA(5), 0, PIO_DEFAULT, PIO_PERIPH_A},
-               {"SDMMC0_DAT4", AT91C_PIN_PA(6), 0, PIO_DEFAULT, PIO_PERIPH_A},
-               {"SDMMC0_DAT5", AT91C_PIN_PA(7), 0, PIO_DEFAULT, PIO_PERIPH_A},
-               {"SDMMC0_DAT6", AT91C_PIN_PA(8), 0, PIO_DEFAULT, PIO_PERIPH_A},
-               {"SDMMC0_DAT7", AT91C_PIN_PA(9), 0, PIO_DEFAULT, PIO_PERIPH_A},
+               {"SDMMC0_CK",   AT91C_PIN_PA(0), 0, PIO_DRVSTR_HI, PIO_PERIPH_A},
+               {"SDMMC0_CMD",  AT91C_PIN_PA(1), 0, PIO_DRVSTR_HI, PIO_PERIPH_A},
+               {"SDMMC0_DAT0", AT91C_PIN_PA(2), 0, PIO_DRVSTR_HI, PIO_PERIPH_A},
+               {"SDMMC0_DAT1", AT91C_PIN_PA(3), 0, PIO_DRVSTR_HI, PIO_PERIPH_A},
+               {"SDMMC0_DAT2", AT91C_PIN_PA(4), 0, PIO_DRVSTR_HI, PIO_PERIPH_A},
+               {"SDMMC0_DAT3", AT91C_PIN_PA(5), 0, PIO_DRVSTR_HI, PIO_PERIPH_A},
+               {"SDMMC0_DAT4", AT91C_PIN_PA(6), 0, PIO_DRVSTR_HI, PIO_PERIPH_A},
+               {"SDMMC0_DAT5", AT91C_PIN_PA(7), 0, PIO_DRVSTR_HI, PIO_PERIPH_A},
+               {"SDMMC0_DAT6", AT91C_PIN_PA(8), 0, PIO_DRVSTR_HI, PIO_PERIPH_A},
+               {"SDMMC0_DAT7", AT91C_PIN_PA(9), 0, PIO_DRVSTR_HI, PIO_PERIPH_A},
                {"SDMMC0_RSTN", AT91C_PIN_PA(10), 0, PIO_DEFAULT, PIO_PERIPH_A},
                {"SDMMC0_VDDSEL", AT91C_PIN_PA(11), 0, PIO_DEFAULT, PIO_PERIPH_A},
                {"SDMMC0_WP",   AT91C_PIN_PA(12), 1, PIO_DEFAULT, PIO_PERIPH_A},
`

Please let me know if this fixes your issue

ehristev, 2019-07-31:
Hi, 

I made a much better and cleaner patch, fixing the exact issue in the code.
Can you please pull the latest master and try .

Thanks !!

ChevalierDeBalibari, 2019-07-31:
I tried previous one, it works!

```
RomBOOT
 ba_offset = 0xb ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf000c000: 0x10 0x3004ff 0xd00139 0x22239337
0xf000c010: 0x2c81716 0x82482 0x33338 0x10000
0xf000c020: 0x16 0x50008 0x0 0x0
0xf000c030: 0x6 0xf6504 0x0 0x0
0xf000c040: 0x0 0x0 0x0 0x0
0xf000c050: 0x0 0x0 0x0 0x1
0xf000c060: 0x0 0x0 0x0 0x0
0xf000c070: 0x0 0x0 0x0 0x0
0xf000c080: 0x0 0x0 0x0 0x0
0xf000c090: 0x0 0x0 0x0 0x0
0xf000c0a0: 0x0 0x0 0x0 0x0
0xf000c0b0: 0x0 0x0 0x1 0x0
0xf000c0c0: 0x0 0x0 0x0 0x0
0xf000c0d0: 0x0 0x0 0x0 0x0
0xf000c0e0: 0x0 0x0 0x0 0x4000
0xf000c0f0: 0x484d5044 0x44524320 0x0 0x20301
0xf000c100: 0x0 0x0 0x4040404 0x3030303
0xf000c110: 0x0 0x1 0x3e03f02 0x0
0xf000c120: 0x0 0x0 0xc00 0xc00
0xf000c130: 0xc00 0xc00 0xd00 0xd00
0xf000c140: 0xd00 0xd00 0x10 0x10
0xf000c150: 0x10 0x10 0xf 0x0
Applying VDDSDMMC errata to ID: 0x33

AT91Bootstrap 3.8.13-dirty (Wed 31 Jul 2019 10:53:43 AM CEST)

All interrupts redirected to AIC
SDHC: fix in place for SAMA5D2 SoM VDDSDMMC over-consumption errata
SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
mmc_verify_operating_condition
mmc_verify_operating_condition success OCR = 0xc0ff8080
Card type is MMC
sd card identified with CID = 0x11010030 0x30384742 0x3004d0c1 0x383395ff
sdcard_identification success
MMC: Specification Version 4.0 or higher
MMC: v5.1 detected
MMC: highspeed supported
MMC: Dual Data Rate supported
MMC: detecting buswidth...
SDHC: Error detected in status: 0x8000, 0x10
MMC: 4-bit bus width detected
SD/MMC: Done to load image
<debug_uart> 
```

Let me see how it works from master...

ChevalierDeBalibari, 2019-07-31:
Looks fine now from master! Thanks Eugen!

closed 2019-07-31 by ChevalierDeBalibari