linux4sam/at91bootstrap #101: sam9x60_early branch report undefined reference to `get_cm_rev'

zizimumu, 2019-10-23:
I configured bootstrap using :
$make at91sam9x60eksd_uboot_defconfig
then use menuconfig to chose boot memory to nand flash, compiled the codes, it shew error as below:

/home/emy/microchip/at91bootstrap/board/at91sam9x60ek/at91sam9x60ek.o: In function `nandflash_hw_init':
/home/emy/microchip/at91bootstrap/board/at91sam9x60ek/at91sam9x60ek.c:309: undefined reference to `get_cm_rev'
/home/emy/microchip/at91bootstrap/board/at91sam9x60ek/at91sam9x60ek.c:342: undefined reference to `get_cm_rev'

ehristev, 2019-10-23:
Can you try this on the master branch ?
The early branch is no longer supported

zizimumu, 2019-10-23:
is that to say that all the sam9x60_early branch in bootrap , u-boot and kernel are  no longer supported? the master branch in bootrap , u-boot and kernel can already fully support the SAM9X60-EK?

ehristev, 2019-10-23:
> is that to say that all the sam9x60_early branch in bootrap , u-boot and kernel are no longer supported? the master branch in bootrap , u-boot and kernel can already fully support the SAM9X60-EK?

Yes this is correct.
However the master branches support only the official rev of the SAM9X60 CPU and the official rev of the SAM9X60-EK as described on microchip website:
https://www.microchip.com/developmenttools/ProductDetails/DT100126
Early adopters hardware versions are no longer supported (if you have such version)
Also, I strongly believe that NAND flash booting was never supported in early version branch (supports start from IAR)

zizimumu, 2019-10-23:
OK, Thanks for your information.

closed 2019-10-23 by zizimumu