linux4sam/at91bootstrap #103: sam9x60 : qspi boot error

FabienLehoussel, 2020-01-13:
On v3.9.0 branch, sam9x60 doesn't boot from qspi.

Boot stop on EEPROM read.

Boot ok when # CONFIG LOAD EEPROM is not set

Boot ok on NAND with CONFIG LOAD EEPROM activated

ehristev, 2020-01-13:
RomBOOT

AT91Bootstrap 3.9.1-rc1-00017-g9c90157 (luni 13 ianuarie 2020, 14:55:41 +0200)

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
  #0  SAM9X60-EK [BB2]      MCHIP RDC
EEPROM: BoardDate | Year | Week
EEPROM:             2019    30

EEPROM: Board sn: 0x11500000 revision: 0x400400

SF: Got Manufacturer and Device ID: bf2643
SF: Copy 0xc0000 bytes from 0x40000 to 0x23f00000
SF: Done to load image
<debug_uart>

U-Boot 2020.01-00284-gcdde326 (Jan 13 2020 - 15:00:55 +0200)

CPU: SAM9X60
Crystal frequency:       24 MHz
CPU clock        :      600 MHz
Master clock     :      200 MHz
DRAM:  256 MiB
MMC:   sdhci-host@80000000: 0
Loading Environment from FAT... Card did not respond to voltage select!
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@f802c000
Hit any key to stop autoboot:  0
U-Boot>
U-Boot>
U-Boot>

ehristev, 2020-01-13:
AT91Bootstrap 3.9.0 (luni 13 ianuarie 2020, 15:04:49 +0200)

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
  #0  SAM9X60-EK [BB2]      MCHIP RDC
EEPROM: BoardDate | Year | Week
EEPROM:             2019    30

EEPROM: Board sn: 0x11500000 revision: 0x400400

SF: Got Manufacturer and Device ID: bf2643
SF: Copy 0xc0000 bytes from 0x40000 to 0x23f00000
SF: Done to load image
<debug_uart>

U-Boot 2020.01-00284-gcdde326 (Jan 13 2020 - 15:00:55 +0200)

CPU: SAM9X60
Crystal frequency:       24 MHz
CPU clock        :      600 MHz
Master clock     :      200 MHz
DRAM:  256 MiB
MMC:   sdhci-host@80000000: 0
Loading Environment from FAT... Card did not respond to voltage select!
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@f802c000
Hit any key to stop autoboot:  0
U-Boot>

ehristev, 2020-01-13:
Can you help with reproducing your issue. Copy/paste your console log so we can debug your issue, maybe there is something wrong with your board/setup. Thanks !

FabienLehoussel, 2020-01-13:
## Test 1 : NOK 

Conf sam9x60ekdf_qspi_uboot_defconfig  : 
```
CONFIG_SAM9X60EK=y
CONFIG_RAM_256MB=y
CONFIG_SPI_CLK=50000000
CONFIG_IMG_ADDRESS="0x00040000"
CONFIG_IMG_SIZE="0x000c0000"
CONFIG_DEBUG=y
```

Log : 
```console
AT91Bootstrap 3.90 (lundi 13 janvier 2020, 14:23:39 (UTC+0100))

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
```

## Test 2 : OK 

Conf sam9x60ekdf_qspi_uboot_defconfig -   CONFIG_LOAD_EEPROM  : 
```
CONFIG_SAM9X60EK=y
CONFIG_RAM_256MB=y
CONFIG_SPI_CLK=50000000
CONFIG_IMG_ADDRESS="0x00040000"
CONFIG_IMG_SIZE="0x000c0000"
CONFIG_DEBUG=y
# CONFIG_LOAD_EEPROM is not set
```

Log : 
```console
AT91Bootstrap 3.9.0 (lundi 13 janvier 2020, 14:17:36 (UTC+0100))

SF: Got Manufacturer and Device ID: bf2643
SF: Copy 0xc0000 bytes from 0x40000 to 0x23f00000
SF: Done to load image
<debug_uart>

U-Boot 2019.04-linux4sam_6.2-icp-rc1-dirty (Jan 13 2020 - 09:32:53 +0100)

CPU: SAM9X60
Crystal frequency:       24 MHz
CPU clock        :      600 MHz
Master clock     :      200 MHz
DRAM:  256 MiB
NAND:  512 MiB
MMC:
```

## Test 3 : OK

Conf sam9x60ekdf_qspi_uboot_defconfig +  CONFIG_DEBUG_VERY_LOUD : 
```
CONFIG_SAM9X60EK=y
CONFIG_RAM_256MB=y
CONFIG_SPI_CLK=50000000
CONFIG_IMG_ADDRESS="0x00040000"
CONFIG_IMG_SIZE="0x000c0000"
CONFIG_DEBUG=y
CONFIG_DEBUG_VERY_LOUD=y
```
log : 
```console 
AT91Bootstrap 3.9.0 (lundi 13 janvier 2020, 14:37:24 (UTC+0100))

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
EEPROM Buff:
@address: 0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf
0x30461c: 0x20 0x4d 0x43 0x48 0x49 0x50 0x20 0x52 0x44 0x43 0x12 0x1f 0x42 0x42 0x32 0xff        MCHIP RDC..BB2.
0x30462c: 0x0 0x0 0x53 0x41 0x4d 0x39 0x58 0x36 0x30 0x2d 0x45 0x4b 0x0 0x0 0x43 0xff   ..SAM9X60-EK..C.
▒▒▒▒▒▒▒▒▒▒▒▒▒^▒: Size of byte is incorrect

EEPROM: Using default information

EEPROM: Board sn: 0x11500000 revision: 0x0

SF: Got Manufacturer and Device ID: bf2643
SF: Copy 0xc0000 bytes from 0x40000 to 0x23f00000
SF: Done to load image
<debug_uart>

U-Boot 2019.04-linux4sam_6.2-icp-rc1-dirty (Jan 13 2020 - 09:32:53 +0100)

CPU: SAM9X60
Crystal frequency:       24 MHz
CPU clock        :      600 MHz
Master clock     :      200 MHz
DRAM:  256 MiB
NAND:  512 MiB
MMC:
Loading Environment from SPI 
```

ehristev, 2020-01-14:
So, if you enable DEBUG_VERY_LOUD, it works?

ehristev, 2020-01-14:
I can only imagine that at91bootstrap crashes after reading the EEPROM in the first case... Unfortunately I cannot reproduce this on my board. 
Can you try to add print messages in that code, to let me know where it's failing. 
Then we can see where exactly at91bootstrap stops executing and try to figure out what is wrong

FabienLehoussel, 2020-01-15:
Log after #104 correction : 

```console

AT91Bootstrap 3.9.1-rc1-00001-g888c3a1-dirty (mercredi 15 janvier 2020, 12:04:07 (UTC+0100))

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
EEPROM Buff:
@address: 0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf
0x30462c: 0x20 0x4d 0x43 0x48 0x49 0x50 0x20 0x52 0x44 0x43 0x12 0x1f 0x42 0x42 0x32 0xff        MCHIP RDC..BB2.
0x30463c: 0x0 0x0 0x53 0x41 0x4d 0x39 0x58 0x36 0x30 0x2d 0x45 0x4b 0x0 0x0 0x43 0xff   ..SAM9X60-EK..C.
parse_alt_board_hw_info: Size of byte is incorrect

EEPROM: Using default information

EEPROM: Board sn: 0x11500000 revision: 0x0

SF: Got Manufacturer and Device ID: bf2643
SF: Copy 0xc0000 bytes from 0x40000 to 0x23f00000
SF: Done to load image
<debug_uart>
```

What is wrong with my EEPROM ?

FabienLehoussel, 2020-01-15:
I also have problem with uboot ...

Where can I add issue ?

ehristev, 2020-01-15:
> Log after #104 correction :
> 
> ```
> AT91Bootstrap 3.9.1-rc1-00001-g888c3a1-dirty (mercredi 15 janvier 2020, 12:04:07 (UTC+0100))
> 
> EEPROM: Loading AT24xx information ...
> EEPROM: BoardName | [Revid] | VendorName
> EEPROM Buff:
> @address: 0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf
> 0x30462c: 0x20 0x4d 0x43 0x48 0x49 0x50 0x20 0x52 0x44 0x43 0x12 0x1f 0x42 0x42 0x32 0xff        MCHIP RDC..BB2.
> 0x30463c: 0x0 0x0 0x53 0x41 0x4d 0x39 0x58 0x36 0x30 0x2d 0x45 0x4b 0x0 0x0 0x43 0xff   ..SAM9X60-EK..C.
> parse_alt_board_hw_info: Size of byte is incorrect
> 
> EEPROM: Using default information
> 
> EEPROM: Board sn: 0x11500000 revision: 0x0
> 
> SF: Got Manufacturer and Device ID: bf2643
> SF: Copy 0xc0000 bytes from 0x40000 to 0x23f00000
> SF: Done to load image
> <debug_uart>
> ```
> 
> What is wrong with my EEPROM ?
> 
> Can you print your EEPROM buff ?

Can you modify the first byte of your eeprom '0x20' to '0x41' . This value looks to be wrong. Did you modify it ?
You can use Linux or something else to access the eeprom through i2c bus and rewrite it.

ehristev, 2020-01-15:
> I also have problem with uboot ...
> 
> Where can I add issue ?

e-mail it to me for now eugen.hristev@microchip.com

ehristev, 2020-01-15:
> > I also have problem with uboot ...
> > Where can I add issue ?
> 
> e-mail it to me for now [eugen.hristev@microchip.com](mailto:eugen.hristev@microchip.com)

You can open your issue here now:
https://github.com/linux4sam/u-boot-at91

I also merged your fix. Thanks again !

ehristev, 2020-02-17:
For your EEPROM bad values, there will be a special software available that can rewrite your memory with proper values. Meantime, you can rewrite this memory yourself, if you need to . (using linux for example).

closed 2020-02-17 by ehristev