linux4sam/at91bootstrap #105: problem loading uboot from emmc

IvanZholubak7, 2020-01-15:
Hello,
we have a board based on SAMA5D4 xplained and we have SD card slot connected to MMC1 and eMMC on MMC0 interface. eMMC - SDINBDG4-8G.

Problem is that we can't boot from eMMC:

    at91bootstrap is compiled for mmc0 interface
    u-boot is compiled for mmc0 interface

eMMC is partitioned via linux running on SD card using fdisk and mkfs:
mmcblk0p1
mmcblk0p2
When we try to boot from eMMC, we are getting next message:
RomBOOT

MATRIX64:
MATRIX_SRTSR0: 0x0, MATRIX_SASSR0: 0x0, MATRIX_SSR0: 0x0
MATRIX_SRTSR1: 0x0, MATRIX_SASSR1: 0xf, MATRIX_SSR1: 0x10101
MATRIX_SRTSR2: 0x0, MATRIX_SASSR2: 0x8, MATRIX_SSR2: 0x10101
MATRIX_SRTSR3: 0x0, MATRIX_SASSR3: 0xf, MATRIX_SSR3: 0x0
MATRIX_SRTSR4: 0xf, MATRIX_SASSR4: 0xffff, MATRIX_SSR4: 0xf0f0f
MATRIX_SRTSR5: 0xf, MATRIX_SASSR5: 0xffff, MATRIX_SSR5: 0xf0f0f
MATRIX_SRTSR6: 0xf, MATRIX_SASSR6: 0xffff, MATRIX_SSR6: 0xf0f0f
MATRIX_SRTSR7: 0xf, MATRIX_SASSR7: 0xffff, MATRIX_SSR7: 0xf0f0f
MATRIX_SRTSR8: 0xf, MATRIX_SASSR8: 0xffff, MATRIX_SSR8: 0xf0f0f
MATRIX_SRTSR9: 0xf, MATRIX_SASSR9: 0xffff, MATRIX_SSR9: 0xf0f0f
MATRIX_SRTSR10: 0xf, MATRIX_SASSR10: 0xffff, MATRIX_SSR10: 0xf0f0f
MATRIX_SRTSR11: 0x0, MATRIX_SASSR11: 0x4, MATRIX_SSR11: 0x0
MATRIX_SRTSR12: 0x0, MATRIX_SASSR12: 0xf, MATRIX_SSR12: 0x10101

MATRIX32:
MATRIX_SRTSR0: 0x0, MATRIX_SASSR0: 0xf, MATRIX_SSR0: 0x10101
MATRIX_SRTSR1: 0x0, MATRIX_SASSR1: 0xf, MATRIX_SSR1: 0x10101
MATRIX_SRTSR2: 0x0, MATRIX_SASSR2: 0xf, MATRIX_SSR2: 0x10101
MATRIX_SRTSR3: 0xf000000, MATRIX_SASSR3: 0xff000000, MATRIX_SSR3: 0xc0c0c0
MATRIX_SRTSR4: 0x0, MATRIX_SASSR4: 0x1, MATRIX_SSR4: 0x10101
MATRIX_SRTSR5: 0x0, MATRIX_SASSR5: 0x888, MATRIX_SSR5: 0x70707
MATRIX_SRTSR6: 0x0, MATRIX_SASSR6: 0x8, MATRIX_SSR6: 0x10101

MATRIX32
MATRIX_SPSELR(0): 0xe7cd2504
MATRIX_SPSELR(1): 0x21dfd37d
MATRIX_SPSELR(2): 0xfffffffa

_MATRIX64
MATRIX_SPSELR(0): 0xfffadafb
MATRIX_SPSELR(1): 0xffffffff
MATRIX_SPSELR(2): 0xffffffe7
ba_offset = 0xc ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf0010000: 0x0 0x30030e 0xd0003d 0x2233b338
0xf0010010: 0xb20a928 0x92328 0x33338 0x10000
0xf0010020: 0x6 0x50008 0x0 0x0
0xf0010030: 0x6 0x60514 0x0 0x0
0xf0010040: 0x0 0x0 0x0 0x0
0xf0010050: 0x0 0x0 0x0 0x2
0xf0010060: 0x8030a 0x8030a 0x8030a 0x8030a
0xf0010070: 0x0 0x0 0x0 0x0
0xf0010080: 0x0 0x1 0x0 0x0
0xf0010090: 0x2 0x0 0x0 0x0
0xf00100a0: 0x0 0x0 0x0 0x0
0xf00100b0: 0x0 0x0 0x0 0x0
0xf00100c0: 0x0 0x0 0x0 0x0
0xf00100d0: 0x0 0x0 0x0 0x0
0xf00100e0: 0x0 0x0 0x0 0x4000
0xf00100f0: 0x484d5044 0x44524320 0x0 0x210
0xf0010100: 0x0 0x0 0xc0d0c0c 0xa0a0a0a
0xf0010110: 0x3040203 0x7 0x8008201 0x0
0xf0010120: 0x0 0x0 0x1400 0x1400
0xf0010130: 0x1300 0x1400 0x1600 0x1600
0xf0010140: 0x1600 0x1600 0x1d 0x1e
0xf0010150: 0x1c 0x1d 0x19 0x0
Auto-Config the TWI Bus by the board

AT91Bootstrap 3.8.13-dirty (п║я─пЄ я▐пҐп╡ 15 12:01:34 EET 2020)

All interrupts redirected to AIC
EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
twi read: timeout to wait RXRDY bit on bus 0
EEPROM: Failed to read

EEPROM: Using default information

EEPROM: Board sn: 0x1012420 revision: 0x680820

twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
SiI9022: Not found
PHY: ETH0 PHY: Enter power down mode
twi read: timeout to wait RXRDY bit on bus 3
twi read: timeout to wait RXRDY bit on bus 3
twi read: timeout to wait RXRDY bit on bus 3
Zh_121: main.c Call function ret = (*load_image)(&image)
Zh_127: sdcard.c Call function at91_mci0_hw_init()
Zh_703: sama5d4_xplained.c const struct pio_desc mci_pins[]
xplained mci0 v4
SD/MMC: Image: Read file u-boot.bin to 0x26f00000
Zh_43: sdcard.c We are in function sdcard_loadimage()
Zh_2256: ff.c We are in function f_open()
Zh_2269: ff.c Call function chk_mounted()
Zh_2023: ff.c *p == u-boot.bin
Zh_2056: ff.c fs->drv == 0
Zh_24: diskio.c Call function disk_initialize
Zh_1111: mci_media.c We are in function sdcard_initialize()
Zh_1123: diskio.c Call function sdcard_register_at91_mci(sdcard)
Zh_398: at91.mci.c    1
Zh_403: at91.mci.c    2
Zh_1134: mci_media.c   ret = sdcard_identification(sdcard)
Zh_553: mci_media.c ocr = command->resp[0] | OCR_ACCESS_MODE_SECTOR;  = 1073741824
Zh_879: mci_media.c   mmc_verify_operating_condition(sdcard) == -12
Unusable Card
Zh_1136: mci_media.c   ret == -1
Zh_2272: ff.c res == 3
Zh_51: sdcard.c fret == 3
*** FATFS: f_open, filename: [u-boot.bin]: error
Zh_132: main.c Call function load_image_done(ret);
SD/MMC: Failed to load image

I added many outputs to know how my program work.
And I did all from https://github.com/linux4sam/at91bootstrap/issues/98.
[image]

ehristev, 2020-01-16:
Hi,
Can you please upgrade to 3.9.1 version of bootstrap. There have been some fixes in that area since 3.8.13.
Let me know how this version work

ehristev, 2020-01-16:
Hi,
Are you loading at91bootstrap from this eMMC , with the RomCode ? 
meaning you have RomCode configured to boot application from eMMC ? or it loads it from another media? You have any other media inserted ? like an Sd-Card on MCI1 .
Thanks

IvanZholubak7, 2020-01-16:
I can load Linux from this eMMC when bootstrap and uboot is in MCI1. All wokr correct. Bootstrap start from eMMC but can not load uboot from this eMMC. Error - unusable card. I will try 3.9.1 version in near hour and will write result. Thank you for help.

IvanZholubak7, 2020-01-16:
Not work.

?ЭRomBOOT

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf0010000: 0x0 0x30030e 0xd0003d 0x2233b338
0xf0010010: 0xb20a928 0x92328 0x33338 0x10000
0xf0010020: 0x6 0x50008 0x0 0x0
0xf0010030: 0x6 0x70514 0x0 0x0
0xf0010040: 0x0 0x0 0x0 0x0
0xf0010050: 0x0 0x0 0x0 0x2
0xf0010060: 0x8030a 0x8030a 0x8030a 0x8030a
0xf0010070: 0x0 0x0 0x0 0x0
0xf0010080: 0x0 0x1 0x0 0x0
0xf0010090: 0x2 0x0 0x0 0x0
0xf00100a0: 0x0 0x0 0x0 0x0
0xf00100b0: 0x0 0x0 0x0 0x0
0xf00100c0: 0x0 0x0 0x0 0x0
0xf00100d0: 0x0 0x0 0x0 0x0
0xf00100e0: 0x0 0x0 0x0 0x4000
0xf00100f0: 0x484d5044 0x44524320 0x0 0x210
0xf0010100: 0x0 0x0 0xc0d0c0c 0xa0a0a0a
0xf0010110: 0x3040203 0x7 0x8108101 0x0
0xf0010120: 0x0 0x0 0x1400 0x1400
0xf0010130: 0x1300 0x1400 0x1600 0x1600
0xf0010140: 0x1600 0x1600 0x1d 0x1e
0xf0010150: 0x1c 0x1d 0x19 0x0
Auto-Config the TWI Bus by the board

AT91Bootstrap 3.9.1-dirty (пїя┌п╡ я▐пҐп╡ 16 12:40:50 EET 2020)

All interrupts redirected to AIC
EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
twi read: timeout to wait RXRDY bit on bus 0
EEPROM: Failed to read

EEPROM: Using default information

EEPROM: Board sn: 0x1012420 revision: 0x680820

twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
twi read: timeout to wait RXRDY bit on bus 0
SiI9022: Failed to read on TWI bus: 0
SiI9022: Failed to 0x1b
SiI9022: Not found
PHY: ETH0 PHY: Enter power down mode
twi read: timeout to wait RXRDY bit on bus 3
twi read: timeout to wait RXRDY bit on bus 3
twi read: timeout to wait RXRDY bit on bus 3
xplained mci0 v1 bootstrap 3.9.1
SD/MMC: Image: Read file u-boot.bin to 0x26f00000
Unusable Card
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image

My code:

#ifdef CONFIG_SDCARD
#ifdef CONFIG_OF_LIBFDT
void at91_board_set_dtb_name(char *of_name)
{
	strcpy(of_name, "at91-sama5d4_xplained.dtb");
}
#endif

void at91_mci0_hw_init(void)
{
	dbg_info("xplained mci0 v2 bootstrap 3.9.1\n");
	const struct pio_desc mci0_pins[] = {
		{"MCI0_CK", AT91C_PIN_PC(4), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_CDA", AT91C_PIN_PC(5), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},

		{"MCI0_DA0", AT91C_PIN_PC(6), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA1", AT91C_PIN_PC(7), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA2", AT91C_PIN_PC(8), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA3", AT91C_PIN_PC(9), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA4", AT91C_PIN_PC(10), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA5", AT91C_PIN_PC(11), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA6", AT91C_PIN_PC(12), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA7", AT91C_PIN_PC(13), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{(char *)0, 0, 0, PIO_DEFAULT, PIO_PERIPH_A},
	};

	pio_configure(mci0_pins);
	pmc_enable_periph_clock(AT91C_ID_PIOC);
	pmc_enable_periph_clock(AT91C_ID_HSMCI0);
}

#endif /* #ifdef CONFIG_SDCARD */

ehristev, 2020-01-16:
`static int mmc_verify_operating_condition(struct sd_card *sdcard)                
{                                                                                
        struct sd_command *command = sdcard->command;                            
        unsigned int ocr;                                                        
        unsigned int retries = 1000;                                             
        unsigned int i;                                                          
        int ret;                                                                 
                                                                                 
        dbg_very_loud("mmc_verify_operating_condition\n");                       
                                                                                 
        /* Query the card and determine the voltage type of the card */          
        ret = mmc_cmd_send_op_cond(sdcard, 0);                                   
        if (ret)                                                                 
                return ret;                                                      
                                                                                 
        ocr = command->resp[0] | OCR_ACCESS_MODE_SECTOR;                         
                                                                                 
        for (i = 0; i < retries; i++) {                                          
                ret = mmc_cmd_send_op_cond(sdcard, ocr);                         
                if (ret)                                                         
                        return ret;                                              
                                                                                 
                if (command->resp[0]  & (0x01 << 31))                            
                        break;                                                   
                                                                                 
                udelay(1000);                                                    
        };                                                                       
                                                                                 
        if (i == retries)                                                        
                return ERROR_UNUSABLE_CARD;                                      
                                                                                 
        sdcard->reg->ocr = command->resp[0];                                     
                                                                                 
        dbg_very_loud("mmc_verify_operating_condition success OCR = %x\n",       
                        sdcard->reg->ocr);                                       
        return 0;                                                                
}                    `

ehristev, 2020-01-16:
From the above code, why isn't anything executed anymore ?

ehristev, 2020-01-16:
Can you enable DEBUG_VERY_LOAD and reprint ?

IvanZholubak7, 2020-01-16:
I see that commands that send to eMMC is not correct. mmc_cmd_send_op_cond return 0.

IvanZholubak7, 2020-01-16:
I output commands in cycles in function  mmc_verify_operating_condition. And commands is not corect.

RomBOOT

MATRIX64:
MATRIX_SRTSR0: 0x0, MATRIX_SASSR0: 0x0, MATRIX_SSR0: 0x0
MATRIX_SRTSR1: 0x0, MATRIX_SASSR1: 0xf, MATRIX_SSR1: 0x10101
MATRIX_SRTSR2: 0x0, MATRIX_SASSR2: 0x8, MATRIX_SSR2: 0x10101
MATRIX_SRTSR3: 0x0, MATRIX_SASSR3: 0xf, MATRIX_SSR3: 0x0
MATRIX_SRTSR4: 0xf, MATRIX_SASSR4: 0xffff, MATRIX_SSR4: 0xf0f0f
MATRIX_SRTSR5: 0xf, MATRIX_SASSR5: 0xffff, MATRIX_SSR5: 0xf0f0f
MATRIX_SRTSR6: 0xf, MATRIX_SASSR6: 0xffff, MATRIX_SSR6: 0xf0f0f
MATRIX_SRTSR7: 0xf, MATRIX_SASSR7: 0xffff, MATRIX_SSR7: 0xf0f0f
MATRIX_SRTSR8: 0xf, MATRIX_SASSR8: 0xffff, MATRIX_SSR8: 0xf0f0f
MATRIX_SRTSR9: 0xf, MATRIX_SASSR9: 0xffff, MATRIX_SSR9: 0xf0f0f
MATRIX_SRTSR10: 0xf, MATRIX_SASSR10: 0xffff, MATRIX_SSR10: 0xf0f0f
MATRIX_SRTSR11: 0x0, MATRIX_SASSR11: 0x4, MATRIX_SSR11: 0x0
MATRIX_SRTSR12: 0x0, MATRIX_SASSR12: 0xf, MATRIX_SSR12: 0x10101

MATRIX32:
MATRIX_SRTSR0: 0x0, MATRIX_SASSR0: 0xf, MATRIX_SSR0: 0x10101
MATRIX_SRTSR1: 0x0, MATRIX_SASSR1: 0xf, MATRIX_SSR1: 0x10101
MATRIX_SRTSR2: 0x0, MATRIX_SASSR2: 0xf, MATRIX_SSR2: 0x10101
MATRIX_SRTSR3: 0xf000000, MATRIX_SASSR3: 0xff000000, MATRIX_SSR3: 0xc0c0c0
MATRIX_SRTSR4: 0x0, MATRIX_SASSR4: 0x1, MATRIX_SSR4: 0x10101
MATRIX_SRTSR5: 0x0, MATRIX_SASSR5: 0x888, MATRIX_SSR5: 0x70707
MATRIX_SRTSR6: 0x0, MATRIX_SASSR6: 0x8, MATRIX_SSR6: 0x10101

MATRIX32
MATRIX_SPSELR(0): 0xe7cd2504
MATRIX_SPSELR(1): 0x21dfd37d
MATRIX_SPSELR(2): 0xfffffffa

_MATRIX64
MATRIX_SPSELR(0): 0xfffadafb
MATRIX_SPSELR(1): 0xffffffff
MATRIX_SPSELR(2): 0xffffffe7
ba_offset = 0xc ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf0010000: 0x0 0x30030e 0xd0003d 0x2233b338
0xf0010010: 0xb20a928 0x92328 0x33338 0x10000
0xf0010020: 0x6 0x50008 0x0 0x0
0xf0010030: 0x6 0x70514 0x0 0x0
0xf0010040: 0x0 0x0 0x0 0x0
0xf0010050: 0x0 0x0 0x0 0x2
0xf0010060: 0x8030a 0x8030a 0x8030a 0x8030a
0xf0010070: 0x0 0x0 0x0 0x0
0xf0010080: 0x0 0x1 0x0 0x0
0xf0010090: 0x2 0x0 0x0 0x0
0xf00100a0: 0x0 0x0 0x0 0x0
0xf00100b0: 0x0 0x0 0x0 0x0
0xf00100c0: 0x0 0x0 0x0 0x0
0xf00100d0: 0x0 0x0 0x0 0x0
0xf00100e0: 0x0 0x0 0x0 0x4000
0xf00100f0: 0x484d5044 0x44524320 0x0 0x210
0xf0010100: 0x0 0x0 0xc0d0c0c 0xa0a0a0a
0xf0010110: 0x3040203 0x7 0x8208202 0x0
0xf0010120: 0x0 0x0 0x1400 0x1400
0xf0010130: 0x1300 0x1400 0x1600 0x1600
0xf0010140: 0x1600 0x1600 0x1d 0x1e
0xf0010150: 0x1c 0x1d 0x19 0x0

AT91Bootstrap 3.9.1-dirty (пїя┌п╡ я▐пҐп╡ 16 16:34:34 EET 2020)

All interrupts redirected to AIC
twi read: timeout to wait RXRDY bit on bus 0
twi read: timeout to wait RXRDY bit on bus 0
xplained mci0 v3 bootstrap 3.9.1
SD/MMC: Image: Read file u-boot.bin to 0x26f00000
Zh_309: at91_mci.c command->resp_type   = 0
Zh_335: at91_mci.c command->cmd = 0
Zh_343: at91_mci.c command->argu = 0
Zh_346: at91_mci.c cmdreg = 4096
Zh_346: status = mci_readl(MCI_SR); = 201326885
Zh_364: at91_mci.c command->resp_type = 0
mmc_verify_operating_condition
Zh_309: at91_mci.c command->resp_type   = 48
Zh_335: at91_mci.c command->cmd = 1
Zh_343: at91_mci.c command->argu = 0
Zh_346: at91_mci.c cmdreg = 4161
Zh_346: status = mci_readl(MCI_SR); = 201326885
Zh_364: at91_mci.c command->resp_type = 48
Zh_553: mci_media.c ocr = command->resp[0] | OCR_ACCESS_MODE_SECTOR;  = 1073741824
Zh_309: at91_mci.c command->resp_type   = 48
Zh_335: at91_mci.c command->cmd = 1
Zh_343: at91_mci.c command->argu = 1073741824
Zh_346: at91_mci.c cmdreg = 4161
Zh_346: status = mci_readl(MCI_SR); = 201326885
Zh_364: at91_mci.c command->resp_type = 48
cycle  = 0
Zh_309: at91_mci.c command->resp_type   = 48
Zh_335: at91_mci.c command->cmd = 1
Zh_343: at91_mci.c command->argu = 1073741824
Zh_346: at91_mci.c cmdreg = 4161
Zh_346: status = mci_readl(MCI_SR); = 201326885
Zh_364: at91_mci.c command->resp_type = 48
cycle  = 1
Zh_309: at91_mci.c command->resp_type   = 48
Zh_335: at91_mci.c command->cmd = 1
Zh_343: at91_mci.c command->argu = 1073741824
Zh_346: at91_mci.c cmdreg = 4161
Zh_346: status = mci_readl(MCI_SR); = 201326885
Zh_364: at91_mci.c command->resp_type = 48
cycle  = 2
Zh_309: at91_mci.c command->resp_type   = 48
Zh_335: at91_mci.c command->cmd = 1
Zh_343: at91_mci.c command->argu = 1073741824
Zh_346: at91_mci.c cmdreg = 4161
Zh_346: status = mci_readl(MCI_SR); = 201326885
Zh_364: at91_mci.c command->resp_type = 48
cycle  = 3
Zh_309: at91_mci.c command->resp_type   = 48
........................................................................
........................................................................
..........................................................................
Zh_309: at91_mci.c command->resp_type   = 48
Zh_335: at91_mci.c command->cmd = 1
Zh_343: at91_mci.c command->argu = 1073741824
Zh_346: at91_mci.c cmdreg = 4161
Zh_346: status = mci_readl(MCI_SR); = 201326885
Zh_364: at91_mci.c command->resp_type = 48
cycle  = 996
Zh_309: at91_mci.c command->resp_type   = 48
Zh_335: at91_mci.c command->cmd = 1
Zh_343: at91_mci.c command->argu = 1073741824
Zh_346: at91_mci.c cmdreg = 4161
Zh_346: status = mci_readl(MCI_SR); = 201326885
Zh_364: at91_mci.c command->resp_type = 48
cycle  = 997
Zh_309: at91_mci.c command->resp_type   = 48
Zh_335: at91_mci.c command->cmd = 1
Zh_343: at91_mci.c command->argu = 1073741824
Zh_346: at91_mci.c cmdreg = 4161
Zh_346: status = mci_readl(MCI_SR); = 201326885
Zh_364: at91_mci.c command->resp_type = 48
cycle  = 998
Zh_309: at91_mci.c command->resp_type   = 48
Zh_335: at91_mci.c command->cmd = 1
Zh_343: at91_mci.c command->argu = 1073741824
Zh_346: at91_mci.c cmdreg = 4161
Zh_346: status = mci_readl(MCI_SR); = 201326885
Zh_364: at91_mci.c command->resp_type = 48
cycle  = 999
Unusable Card
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image

IvanZholubak7, 2020-01-16:
/*
 * Response Types
 */
#define	SD_RESP_TYPE_NO_RESP	0x00
#define	SD_RESP_TYPE_R1		0x10
#define	SD_RESP_TYPE_R1B	0x11
#define	SD_RESP_TYPE_R2		0x20
#define	SD_RESP_TYPE_R3		0x30
#define	SD_RESP_TYPE_R4		0x40
#define	SD_RESP_TYPE_R5		0x50
#define	SD_RESP_TYPE_R6		0x60
#define	SD_RESP_TYPE_R7		0x70

but my responde type is Zh_309: at91_mci.c command->resp_type = 48
Why so can be. What I can do bad?

ehristev, 2020-01-17:
response type you print as decimal or hexadecimal ?

ehristev, 2020-01-17:
i tried to connect an emmc soldered to an SD-card adaptor , and insert it in the MCI1 slot of the SAMA5d4_xplained board, something similar like this : https://www.antratek.com/microsd-to-emmc-adapter
The RomCode refuses to boot At91Bootstrap from this.. so I cannot reach your point where at91bootstrap dies. I will try to write it on NAND and boot from there. Are you sure that RomCode loads your at91boostrap from your eMMC ?

ehristev, 2020-01-17:
```
RomBOOT

MATRIX64:
MATRIX_SRTSR0: 0x0, MATRIX_SASSR0: 0x0, MATRIX_SSR0: 0x0
MATRIX_SRTSR1: 0x0, MATRIX_SASSR1: 0xf, MATRIX_SSR1: 0x10101
MATRIX_SRTSR2: 0x0, MATRIX_SASSR2: 0x8, MATRIX_SSR2: 0x10101
MATRIX_SRTSR3: 0x0, MATRIX_SASSR3: 0xf, MATRIX_SSR3: 0x0
MATRIX_SRTSR4: 0xf, MATRIX_SASSR4: 0xffff, MATRIX_SSR4: 0xf0f0f
MATRIX_SRTSR5: 0xf, MATRIX_SASSR5: 0xffff, MATRIX_SSR5: 0xf0f0f
MATRIX_SRTSR6: 0xf, MATRIX_SASSR6: 0xffff, MATRIX_SSR6: 0xf0f0f
MATRIX_SRTSR7: 0xf, MATRIX_SASSR7: 0xffff, MATRIX_SSR7: 0xf0f0f
MATRIX_SRTSR8: 0xf, MATRIX_SASSR8: 0xffff, MATRIX_SSR8: 0xf0f0f
MATRIX_SRTSR9: 0xf, MATRIX_SASSR9: 0xffff, MATRIX_SSR9: 0xf0f0f
MATRIX_SRTSR10: 0xf, MATRIX_SASSR10: 0xffff, MATRIX_SSR10: 0xf0f0f
MATRIX_SRTSR11: 0x0, MATRIX_SASSR11: 0x4, MATRIX_SSR11: 0x0
MATRIX_SRTSR12: 0x0, MATRIX_SASSR12: 0xf, MATRIX_SSR12: 0x10101

MATRIX32:
MATRIX_SRTSR0: 0x0, MATRIX_SASSR0: 0xf, MATRIX_SSR0: 0x10101
MATRIX_SRTSR1: 0x0, MATRIX_SASSR1: 0xf, MATRIX_SSR1: 0x10101
MATRIX_SRTSR2: 0x0, MATRIX_SASSR2: 0xf, MATRIX_SSR2: 0x10101
MATRIX_SRTSR3: 0xf000000, MATRIX_SASSR3: 0xff000000, MATRIX_SSR3: 0xc0c0c0
MATRIX_SRTSR4: 0x0, MATRIX_SASSR4: 0x1, MATRIX_SSR4: 0x10101
MATRIX_SRTSR5: 0x0, MATRIX_SASSR5: 0x888, MATRIX_SSR5: 0x70707
MATRIX_SRTSR6: 0x0, MATRIX_SASSR6: 0x8, MATRIX_SSR6: 0x10101

MATRIX32
MATRIX_SPSELR(0): 0xe7cd2504
MATRIX_SPSELR(1): 0x21dfd37d
MATRIX_SPSELR(2): 0xfffffffa

_MATRIX64
MATRIX_SPSELR(0): 0xfffadafb
MATRIX_SPSELR(1): 0xffffffff
MATRIX_SPSELR(2): 0xffffffe7
 ba_offset = 0xc ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf0010000: 0x0 0x30030e 0xd0003d 0x2233b338
0xf0010010: 0xb20a928 0x92328 0x33338 0x10000
0xf0010020: 0x6 0x50008 0x0 0x0
0xf0010030: 0x6 0x880514 0x0 0x0
0xf0010040: 0x0 0x0 0x0 0x0
0xf0010050: 0x0 0x0 0x0 0x2
0xf0010060: 0x8030a 0x8030a 0x8030a 0x8030a
0xf0010070: 0x0 0x0 0x0 0x0
0xf0010080: 0x0 0x1 0x0 0x0
0xf0010090: 0x2 0x0 0x0 0x0
0xf00100a0: 0x0 0x0 0x0 0x0
0xf00100b0: 0x0 0x0 0x0 0x0
0xf00100c0: 0x0 0x0 0x0 0x0
0xf00100d0: 0x0 0x0 0x0 0x0
0xf00100e0: 0x0 0x0 0x0 0x4000
0xf00100f0: 0x484d5044 0x44524320 0x0 0x210
0xf0010100: 0x0 0x0 0xc0d0c0c 0xa0a0a0a
0xf0010110: 0x3040203 0x7 0x7407502 0x0
0xf0010120: 0x0 0x0 0x1100 0x1100
0xf0010130: 0x1000 0x1100 0x1300 0x1300
0xf0010140: 0x1300 0x1300 0x1a 0x1b
0xf0010150: 0x19 0x1a 0x16 0x0
Auto-Config the TWI Bus by the board

AT91Bootstrap 3.9.1-00029-ga5ad6b5-dirty (vineri 17 ianuarie 2020, 10:06:02 +020                                                                                                                                                                                                                                             0)

All interrupts redirected to AIC
EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
EEPROM Buff:
@address: 0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf
0x2054bc: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff                                                                                                                                                                                                                                              0xff 0xff       ................
0x2054cc: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff                                                                                                                                                                                                                                              0xff 0xff       ................
HW Info: The total size: 0xff isn't correct

EEPROM: Using default information

EEPROM: Board sn: 0x1012420 revision: 0x680820

HDMI SiI9022: Enter D3 Cold mode
PHY: ETH0 PHY: Enter power down mode
ACT8865: Set REG1/REG2/REG3 Power-saving mode
twi read: timeout to wait RXRDY bit on bus 3
ACT8865: Disable ACT8865's I2C interface
SD/MMC: Image: Read file u-boot.bin to 0x26f00000
mmc_verify_operating_condition
mmc_verify_operating_condition success OCR = 0xc0ff8080
Card type is MMC
sd card identified with CID = 0xd932c136 0xba005a31 0x31303170 0x10053ff
sdcard_identification success
MMC: Specification Version 1.2
MMC: v5.0/5.01 detected
MMC: highspeed supported
MMC: Dual Data Rate supported
SD/MMC: Done to load image
<debug_uart>

U-Boot 2020.01-00018-gf408b4b (Jan 17 2020 - 10:07:09 +0200)

CPU: SAMA5D44
Crystal frequency:       12 MHz
CPU clock        :      600 MHz
Master clock     :      200 MHz
DRAM:  512 MiB
NAND:  512 MiB
MMC:   Atmel mci: 0
Loading Environment from FAT... OK
In:    serial@fc00c000
Out:   serial@fc00c000
Err:   serial@fc00c000
Net:
Error: ethernet@f8020000 address not set.
No ethernet found.

Hit any key to stop autoboot:  0
=>
=>
=>

```

ehristev, 2020-01-17:
I wrote at91bootstrap on NAND flash, RomCode will load at91bootstrap from NAND and then At91bootstrap will load the application (u-boot.bin ) from eMMC which is inserted with adapter in MCI1 slot, you can see it detectes my eMMC correctly version 5.0/5.01 and loads correctly

ehristev, 2020-01-17:
I do not have your specific memory, my memory used now is KINGSTON 16G-S110 , you have other eMMC you can try ?

ehristev, 2020-01-17:
```
RomBOOT

MATRIX64:
MATRIX_SRTSR0: 0x0, MATRIX_SASSR0: 0x0, MATRIX_SSR0: 0x0
MATRIX_SRTSR1: 0x0, MATRIX_SASSR1: 0xf, MATRIX_SSR1: 0x10101
MATRIX_SRTSR2: 0x0, MATRIX_SASSR2: 0x8, MATRIX_SSR2: 0x10101
MATRIX_SRTSR3: 0x0, MATRIX_SASSR3: 0xf, MATRIX_SSR3: 0x0
MATRIX_SRTSR4: 0xf, MATRIX_SASSR4: 0xffff, MATRIX_SSR4: 0xf0f0f
MATRIX_SRTSR5: 0xf, MATRIX_SASSR5: 0xffff, MATRIX_SSR5: 0xf0f0f
MATRIX_SRTSR6: 0xf, MATRIX_SASSR6: 0xffff, MATRIX_SSR6: 0xf0f0f
MATRIX_SRTSR7: 0xf, MATRIX_SASSR7: 0xffff, MATRIX_SSR7: 0xf0f0f
MATRIX_SRTSR8: 0xf, MATRIX_SASSR8: 0xffff, MATRIX_SSR8: 0xf0f0f
MATRIX_SRTSR9: 0xf, MATRIX_SASSR9: 0xffff, MATRIX_SSR9: 0xf0f0f
MATRIX_SRTSR10: 0xf, MATRIX_SASSR10: 0xffff, MATRIX_SSR10: 0xf0f0f
MATRIX_SRTSR11: 0x0, MATRIX_SASSR11: 0x4, MATRIX_SSR11: 0x0
MATRIX_SRTSR12: 0x0, MATRIX_SASSR12: 0xf, MATRIX_SSR12: 0x10101

MATRIX32:
MATRIX_SRTSR0: 0x0, MATRIX_SASSR0: 0xf, MATRIX_SSR0: 0x10101
MATRIX_SRTSR1: 0x0, MATRIX_SASSR1: 0xf, MATRIX_SSR1: 0x10101
MATRIX_SRTSR2: 0x0, MATRIX_SASSR2: 0xf, MATRIX_SSR2: 0x10101
MATRIX_SRTSR3: 0xf000000, MATRIX_SASSR3: 0xff000000, MATRIX_SSR3: 0xc0c0c0
MATRIX_SRTSR4: 0x0, MATRIX_SASSR4: 0x1, MATRIX_SSR4: 0x10101
MATRIX_SRTSR5: 0x0, MATRIX_SASSR5: 0x888, MATRIX_SSR5: 0x70707
MATRIX_SRTSR6: 0x0, MATRIX_SASSR6: 0x8, MATRIX_SSR6: 0x10101

MATRIX32
MATRIX_SPSELR(0): 0xe7cd2504
MATRIX_SPSELR(1): 0x21dfd37d
MATRIX_SPSELR(2): 0xfffffffa

_MATRIX64
MATRIX_SPSELR(0): 0xfffadafb
MATRIX_SPSELR(1): 0xffffffff
MATRIX_SPSELR(2): 0xffffffe7
 ba_offset = 0xc ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf0010000: 0x0 0x30030e 0xd0003d 0x2233b338
0xf0010010: 0xb20a928 0x92328 0x33338 0x10000
0xf0010020: 0x6 0x50008 0x0 0x0
0xf0010030: 0x6 0x880514 0x0 0x0
0xf0010040: 0x0 0x0 0x0 0x0
0xf0010050: 0x0 0x0 0x0 0x2
0xf0010060: 0x8030a 0x8030a 0x8030a 0x8030a
0xf0010070: 0x0 0x0 0x0 0x0
0xf0010080: 0x0 0x1 0x0 0x0
0xf0010090: 0x2 0x0 0x0 0x0
0xf00100a0: 0x0 0x0 0x0 0x0
0xf00100b0: 0x0 0x0 0x0 0x0
0xf00100c0: 0x0 0x0 0x0 0x0
0xf00100d0: 0x0 0x0 0x0 0x0
0xf00100e0: 0x0 0x0 0x0 0x4000
0xf00100f0: 0x484d5044 0x44524320 0x0 0x210
0xf0010100: 0x0 0x0 0xc0d0c0c 0xa0a0a0a
0xf0010110: 0x3040203 0x7 0x7407502 0x0
0xf0010120: 0x0 0x0 0x1100 0x1100
0xf0010130: 0x1000 0x1100 0x1300 0x1300
0xf0010140: 0x1300 0x1300 0x1a 0x1b
0xf0010150: 0x19 0x1a 0x16 0x0
Auto-Config the TWI Bus by the board

AT91Bootstrap 3.9.1-00029-ga5ad6b5-dirty (vineri 17 ianuarie 2020, 10:06:02 +0200)

All interrupts redirected to AIC
EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
EEPROM Buff:
@address: 0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf
0x2054bc: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff       ................
0x2054cc: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff       ................
HW Info: The total size: 0xff isn't correct

EEPROM: Using default information

EEPROM: Board sn: 0x1012420 revision: 0x680820

HDMI SiI9022: Enter D3 Cold mode
PHY: ETH0 PHY: Enter power down mode
twi read: timeout to wait RXRDY bit on bus 3
twi read: timeout to wait RXRDY bit on bus 3
twi read: timeout to wait RXRDY bit on bus 3
SD/MMC: Image: Read file u-boot.bin to 0x26f00000
mmc_verify_operating_condition
mmc_verify_operating_condition success OCR = 0xc0ff8080
Card type is MMC
sd card identified with CID = 0xf9b5f952 0x7e3bd4a 0x4e423415 0x10041ff
sdcard_identification success
MMC: Specification Version 1.2
MMC: v5.1 detected
MMC: highspeed supported
MMC: Dual Data Rate supported
Cmd: 12, error check: 0x5f0000, status: 0xc050025
*** FATFS: f_read: error
SD/MMC: Failed to load image

```

This is a memory from SAMSUNG but it;s still better than what you are experiencing , is it correct ?

ehristev, 2020-01-17:
the difference between you and me is that you are trying on MCI0 interface and I am using MCI1, my board does not have a slot soldered on MCI0

ShonkyCH, 2020-01-17:
> Are you sure that RomCode loads your at91boostrap from your eMMC ?

This may or may not be relevant as the @IvanZholubak7 seems to have at91bootstrap booting from something but the boot ROM of Rev A parts cannot direct boot from MCI0 - noted in the device errata.

ehristev, 2020-01-17:
> > Are you sure that RomCode loads your at91boostrap from your eMMC ?
> 
> This may or may not be relevant as @IvanZholubak7 seems to have at91bootstrap booting from something but the boot ROM of Rev A parts cannot direct boot from MCI0 - noted in the device errata.

Thanks for this information , @IvanZholubak7 to confirm boot media , and part Rev

IvanZholubak7, 2020-01-17:
I solved this problem changing #define CONFIG_SYS_BASE_MCI	AT91C_BASE_HSMCI1 
to 
#define CONFIG_SYS_BASE_MCI	AT91C_BASE_HSMCI0
in file sama5d4_xplained.h

and 

void at91_mci0_hw_init(void)
{
	const struct pio_desc mci0_pins[] = {
		{"MCI0_CK", AT91C_PIN_PC(4), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_CDA", AT91C_PIN_PC(5), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},

		{"MCI0_DA0", AT91C_PIN_PC(6), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA1", AT91C_PIN_PC(7), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA2", AT91C_PIN_PC(8), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA3", AT91C_PIN_PC(9), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA4", AT91C_PIN_PC(10), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA5", AT91C_PIN_PC(11), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA6", AT91C_PIN_PC(12), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{"MCI0_DA7", AT91C_PIN_PC(13), 0, PIO_DRVSTR_HI, PIO_PERIPH_B},
		{(char *)0, 0, 0, PIO_DEFAULT, PIO_PERIPH_A},
	};

	pio_configure(mci0_pins);
	pmc_enable_periph_clock(AT91C_ID_PIOC);
	pmc_enable_periph_clock(AT91C_ID_HSMCI1);
}

to 
pmc_enable_periph_clock(AT91C_ID_HSMCI0);
in file sama5d4_xplained.c

But I do not understand why it not worked. This variable is program and how i can see not influence. 
Thay was same in two files.

IvanZholubak7, 2020-01-17:
But I have another problem - Card did not respond to voltage select!

RomBOOT

AT91Bootstrap 3.9.1-dirty (п÷я┌пҐ я▐пҐп╡ 17 15:34:37 EET 2020)

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName

EEPROM: Using default information

EEPROM: Board sn: 0x1012420 revision: 0x680820

SD/MMC: Image: Read file u-boot.bin to 0x26f00000
MMC: Specification Version 1.2
MMC: v5.1 detected
MMC: highspeed supported
MMC: Dual Data Rate supported
SD/MMC: Done to load image
<debug_uart>

U-Boot 2017.03-linux4sam_5.7 (Feb 23 2019 - 20:00:00 +0000)

CPU: SAMA5D42
Crystal frequency:       12 MHz
CPU clock        :      600 MHz
Master clock     :      200 MHz
DRAM:  512 MiB
NAND:  atmel_nand: Fail to initialize #0 chip0 MiB
MMC:   Atmel mci: 0
Card did not respond to voltage select!
mmc_init: -95, time 12
** Bad device mmc 0 **
Using default environment

In:    serial
Out:   serial
Err:   serial
Failed to read ethernet address from AT24MAC
Net:
Error: ethernet@f8020000 address not set.
No ethernet found.
Hit any key to stop autoboot:  0
Card did not respond to voltage select!
mmc_init: -95, time 12
** Bad device mmc 0 **
Card did not respond to voltage select!
mmc_init: -95, time 12
** Bad device mmc 0 **
=>

ehristev, 2020-01-17:
Hi, 
It looks you already reached U-boot.
U-boot fails.
So At91bootstrap is working fine ;)
Don't you agree ?

ehristev, 2020-01-17:
> I solved this problem changing #define CONFIG_SYS_BASE_MCI AT91C_BASE_HSMCI1
> to
> #define CONFIG_SYS_BASE_MCI AT91C_BASE_HSMCI0
> in file sama5d4_xplained.h
>
> and
>
> void at91_mci0_hw_init(void)
> {
> const [...]

if you are booting from MCI0, then yes, you need to enable clock of MCI0 peripheral and base bus address of MCI0 must be used. What is unexpected ?

IvanZholubak7, 2020-01-17:
Yes. I agree. Bootstrap work good. Thanks.

IvanZholubak7, 2020-01-17:
Can you help with u-boot. My system normal load Linux kernel when in mci1 is sd csrd with zImage, r5_cpu.dtb,  uboot.env. But if i start from eMMC in mci0 without sd card in mci1 it  show messege - "Card did not respond to voltage select". Rather u-boot try to find file zImage from sd card in mci1. But I can not find configurations for this choose.

ehristev, 2020-01-20:
> Can you help with u-boot. My system normal load Linux kernel when in mci1 is sd csrd with zImage, r5_cpu.dtb, uboot.env. But if i start from eMMC in mci0 without sd card in mci1 it show messege - [...]

Open another issue here https://github.com/linux4sam/u-boot-at91