linux4sam/at91bootstrap #106: How feasible is adding an u-boot like environment to at91bootstrap

fberndl, 2020-02-18:
Hi,

we have a board which is similar to at91sam9m10g45ek. We currently have at91bootstrap as second stage bootloader and u-boot as third stage bootloader. We want to update u-boot whenever it is necessary and want it failsafe, so we duplicate u-boot. 

Or do you think the  u-boot bootloader is not necessary at all for at91sam9m10g45ek ( it is a old project). We use u-boot for the linux update using the swupdate tool.

We are thinking about implementing a u-boot like environment where we can set and get variables. How feasible do you think this is and maybe point us in the right direction. 

Thanks for sharing

ehristev, 2020-02-18:
Hi,

First of all, at91bootstrap does not have an input console. So, it does not have any parser that can input text, so, basically  you cannot type into console for at91bootstrap.
To have a console, would require lots of support, adding commands, etc.
The intention of at91bootstrap is to be small, quick, fast and efficient: initialize external RAM, CPU, clocks, and then pass control to a feature rich bootloader: U-boot.
I suggest that you use U-boot for anything which involves user interaction.
If you need help with something however I am happy to answer.

fberndl, 2020-02-18:
Hi,

this should not be an console. The target is to read and write variables stored in the nand flash during the boot process without interaction of a user from at91bootstrap. The variables are simplified:

- BOOT_PARTITION = 0/1
- PARTITION1_STATE = (UPDATE/GOOD/FAIL)
- PARTITION2_STATE = (UPDATE/GOOD/FAIL)
- BOOTCOUNT = 0 -3 

The target is to boot the at91bootstrap, read BOOT_PARTITION , verify bootcount < 3 increase bootcount and boot from that partition. Once we are in linux the bootcount is set to 0. Of course there is more logic applied to the swupdate process, but it should be similar to those variables in u-boot.

ehristev, 2020-02-18:
This sounds interesting at least. However it might influence the boot process in ways that look strange for people familiar with usual bootloading process

fberndl, 2020-02-18:
Thanks a lot for helping me out.

For our nand flash the manufacturer quarantees that the first page is good out of the box.
 u-boot is the second page. 

The usual bootloading process in most of our products is that u-boot is the second stage bootloader stored in the first page in the flash, for at91sam9m10g45ek we can't do this, right? 

This is the current situation:
|-----------------+
| Image               | 
|-----------------+
| Bootstrap         |
| U-BOOT           |
| U-BOOT ENV1 |
| U-BOOT ENV2 |
| Kernel              |
| Root-FS1         |
| Kernel2            |
| Root-FS2         |
|----------------+

 If we want to update u-boot and we want it failsafe, I see no other option than the suggested solution. Do think that is a bad idea to update u-boot? 

It is the same behaviour if you have u-boot and redundant linux applied to bootstrap and u-boot.

Thanks again

fberndl, 2020-02-18:
Is chainloading with u-boot possible?

ehristev, 2020-02-18:
> Thanks a lot for helping me out.
>
> For our nand flash the manufacturer quarantees that the first page is good out of the box.
> u-boot is the second page.
>
> The usual bootloading process in most of our [...]
Sorry but I am not familiar with this specific CPU
> This is the current situation:
> |-----------------+
> | Image |
> |-----------------+
> | Bootstrap |
> | U-BOOT |
> | U-BOOT ENV1 |
> | U-BOOT ENV2 |
> | Kernel |
> | Root-FS1 |
> | Kernel2 |
> | Root-FS2 | [...]

It is always a good idea to update U-boot. But I do not understand what you mean by failsafe ? You update u-boot and you are afraid that new U-boot will not work ? And you want at91bootstrap to change the u-boot loading offset if it failed several times ?

> 
> It is the same behaviour if you have u-boot and redundant linux applied to bootstrap and u-boot.
> 
> Thanks again

fberndl, 2020-02-18:
>  But I do not understand what you mean by failsafe ? You update u-boot and you are afraid that new U-boot will not work ? And you want at91bootstrap to change the u-boot loading offset if it failed several times ?
Exactly that.

ehristev, 2020-02-19:
I had give it some thought, and how I see it at this moment:
You can have a special config for bootstrap, and if this config is enabled, Bootstrap will read from flash your specific value, and if reaches a limit, it will go and read U-boot from a failsafe location, which is configured with another Kconfig entry.
So, something like:
CONFIG_BOOTSTRAP_CHECK_FAILSAFE_LOG (y/n)
CONFIG_BOOTSTRAP_FAILSAFE_LOG_ADDR (u32 value)
CONFIG_BOOTSTRAP_UBOOT_FAILSAFE_OFFSET (u32 value)

Or something similar.
You can implement this scenario and then we can review and merge it.
What do you think ? Would this solve your issues ?

ehristev, 2020-03-03:
Please reopen if you have more questions, or open issue per question
Thanks !

closed 2020-03-03 by ehristev