linux4sam/at91bootstrap #110: we are using ATSAMA5D27-D1G-CU in our prject. we used eMMC as a boot media. SDMMC0_VDDSEL is this mandatory to use? can we use VDDSDHC for 3.3V alone for booting and other read and write operation?

shiva-kumar35, 2020-07-21:
we are using ATSAMA5D27-D1G-CU in our prject. we used eMMC as a boot media. SDMMC0_VDDSEL is this mandatory to use? can we use VDDSDHC for 3.3V alone for booting and other read and write operation?

attached screen shot of emmc interface. we would like to omit the 
[image]

shiva-kumar35, 2020-07-21:
[image]
we would like to omit the attached circuit

noglitch, 2020-07-21:
Hi,
It's not the proper location for asking such a support question, moreover a HW related one.
Please use either [https://www.at91.com](https://www.at91.com) forum or maybe more appropriate is the Microchip support interface at:
[http://support.microchip.com](http://support.microchip.com)
Login to your "myMicrochip" account or creating one.
Then by clicking on "My Support" and finally "My Cases" button.

I close the "issue" then. Best regards

closed 2020-07-21 by noglitch