linux4sam/at91bootstrap #116: Compile error (clk-common.c:265:10: error: ‘AT91C_ID_SMD’ undeclared)

ruslan-gennadievich, 2020-12-16:
```
ld FLAGS
========
-nostartfiles -Map=binaries/sama5d27_som1_ek-sdcardboot-uboot-3.10.1-rc1.map --cref -static -T elf32-littlearm-tz.lds --gc-sections -Ttext 0x200000

  AS        crt0_gnu.S
  CC        main.c
  CC        board/sama5d27_som1_ek/sama5d27_som1_ek.c
  CC        lib/string.c
  CC        lib/eabi_utils.c
  CC        lib/div.c
  CC        lib/consttime_memequal.c
  CC        driver/at91_slowclk.c
  CC        driver/common.c
  CC        driver/at91_pio.c
  CC        driver/pmc/clk-common.c
driver/pmc/clk-common.c: In function ‘sys_mask_to_per_id’:
driver/pmc/clk-common.c:265:10: error: ‘AT91C_ID_SMD’ undeclared (first use in this function); did you mean ‘AT91C_ID_HSMC’?
   return AT91C_ID_SMD;
          ^~~~~~~~~~~~
          AT91C_ID_HSMC
driver/pmc/clk-common.c:265:10: note: each undeclared identifier is reported only once for each function it appears in
make: *** [Makefile:343: driver/pmc/clk-common.o] Error 1
```

ehristev, 2020-12-17:
Hi,

I have a feeling that you are not using one of the predefined defconfig files, but you changed something inside menuconfig, and then tried to build your board. What have you changed?
In other words, what is your current defconfig, such that I can reproduce this build error ?

Thanks,
Eugen

ruslan-gennadievich, 2020-12-17:
My config
https://pastebin.com/y09kan8t

ruslan-gennadievich, 2020-12-17:
I try build for sama5d2 board, but in file sama5d2.h i can't see define AT91C_ID_SMD
Maybe it's bug?

ehristev, 2021-01-07:
Can you try to use a default *_defconfig file for your board ? (and modify it according to your needs)
Or, as a first step, deselect CONFIG_ENTER_NWD from your config if you have it.

ehristev, 2021-01-15:
Will be fixed in at91bootstrap4. Meanwhile, try to use default defconfigs , as not all combinations of boards and drivers are properly tested

closed 2021-01-15 by ehristev