linux4sam/at91bootstrap #121: Timeouts on custom SAMA5D27 board

fabioestevam, 2021-03-24:
Hi,

I have a custom board based on SAMA5D27 D1G-CU with an 8-bit eMMC connected
to SDMMC0. There is no USB interface, so I use the serial UART port
to program at91bootstrap and U-Boot via SAM-BA.

This board has a 27MHz external clock generator connected to XIN and
XOUT is left unconnected, so I selected CONFIG_MCK_BYPASS=y and adapted
at91bootsrap code as below:

diff --git a/board/sama5d27_som1_ek/sama5d27_som1_ek.h b/board/sama5d27_som1_ek/sama5d27_som1_ek.h
index 912fcfd328fe..324dbb33c574 100644
--- a/board/sama5d27_som1_ek/sama5d27_som1_ek.h
+++ b/board/sama5d27_som1_ek/sama5d27_som1_ek.h
@@ -31,10 +31,10 @@
 /*
  * PMC Settings
  */
-#define BOARD_MAINOSC		24000000
+#define BOARD_MAINOSC		27000000
 
-/* PCK: 492MHz, MCK: 164MHz */
-#define BOARD_PLLA_MULA		40
+/* PCK: 486MHz, MCK: 162MHz */
+#define BOARD_PLLA_MULA		35
 
 #define BOARD_PCK		((unsigned long)((BOARD_MAINOSC * (BOARD_PLLA_MULA + 1)) / 2))
 #define BOARD_MCK		((unsigned long)((BOARD_MAINOSC * (BOARD_PLLA_MULA + 1)) / 2 / 3))
@@ -53,7 +53,7 @@
 					| AT91C_PMC_MDIV_3 \
 					| AT91C_PMC_CSS_PLLA_CLK)
 
-#define MASTER_CLOCK		164000000
+#define MASTER_CLOCK		BOARD_MCK
 
 #define PLLA_SETTINGS		(BOARD_CKGR_PLLA | \
 				BOARD_PLLACOUNT | \
diff --git a/board/sama5d27_som1_ek/sama5d27_som1_eksd1_uboot_defconfig b/board/sama5d27_som1_ek/sama5d27_som1_eksd1_uboot_defconfig
index d1e5ed0e7957..3104eaf01d0a 100644
--- a/board/sama5d27_som1_ek/sama5d27_som1_eksd1_uboot_defconfig
+++ b/board/sama5d27_som1_ek/sama5d27_som1_eksd1_uboot_defconfig
@@ -5,3 +5,5 @@ CONFIG_SDHC1=y
 CONFIG_JUMP_ADDR="0x23f00000"
 CONFIG_DEBUG=y
 # CONFIG_ENTER_NWD is not set
+CONFIG_MCK_BYPASS=y
+CONFIG_SDHC0=y
-- 
2.25.1

After re-builing at91bootsrap, I flash at91bootstrap and U-Boot to the eMMC
like this:

sam-ba -p serial -d sama5d2  -a sdmmc:0:1:1:8:5  -c writeboot:boot.bin
sam-ba -p serial -d sama5d2  -a sdmmc:0:1:1:8:5  -c write:u-boot.bin:0x8000

And after that, I see the following upon boot:

AT91Bootstrap 3.10.2-00017-g7738830d9732 (2021-03-24 12:14:34)

SDHC: fix in place for SAMA5D2 SoM VDDSDMMC over-consumption errata
The MAINCK frequency: 27000000
Timeout waiting for GCK ready!
SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
SDHC: Timeout waiting for internal clock ready

Any ideas as to how to fix the GCK and SDHC timeouts as shown above?

Thanks,

fabioestevam, 2021-03-24:
Sorry, it seems that pasting the patch content did not work.
Here is the patch: https://pastebin.com/raw/jV0Kh653

ehristev, 2021-03-24:
Hi,

According to datasheet, 24 Mhz is the maximum frequency for the external oscillator.
Have you found information that 27 Mhz would work ?
Maybe you can put a clock divider between your oscillator and the SAMA5D2 and thus try at 13.5 Mhz ?

fabioestevam, 2021-03-24:
Hi Eugen,

Please note that this board does not use an external oscillator between XIN and XOUT.

It uses a clock generator of 27MHz connected to XIN pin. XOUT is left unconnected.

According to "Table 67-15. XIN Clock Electrical Characteristics" of SAMA5D2-Series-Data-sheet-ds60001476G.pdf the maximum allowed XIN frequency is 50MHz.

Could you please confirm if it would be safe from a software perspective to use the 27MHz external clock connected to XIN?

Maybe the changes I did in at91bootstrap are not enough?

Thanks

ehristev, 2021-03-24:
As it looks, GCK is not stable , probably because the frequency is out of range. 
In the board file, the GCK for SDMMC0 is configured here:

`        pmc_enable_generic_clock(CONFIG_SYS_ID_SDHC,                             
                                 GCK_CSS_UPLL_CLK,                               
                                 ATMEL_SDHC_GCKDIV_VALUE);`

ehristev, 2021-03-24:
Maybe the UPLL is not configured correctly

ehristev, 2021-03-24:
You can try to change the source of GCK for SDMMC to be PLLA instead of UPLL , or check if UPLL is configured correctly

fabioestevam, 2021-03-24:
Excellent, thanks to your advice I changed it to:

pmc_enable_generic_clock(CONFIG_SYS_ID_SDHC, GCK_CSS_PLLA_CLK, 2);

And now  at91bootstrap does not show any timeout and goes further:

AT91Bootstrap 3.10.2-00017-g7738830d9732-dirty (2021-03-24 14:29:25)

SDHC: fix in place for SAMA5D2 SoM VDDSDMMC over-consumption errata
The MAINCK frequency: 27000000
SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
MMC: Specification Version 4.0 or higher
MMC: v5.0/5.01 detected
MMC: highspeed supported
MMC: Dual Data Rate supported
MMC: detecting buswidth...
MMC: 8-bit bus width detected
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image

I cannot use FAT in my eMMC as I don't have USB access to format it.

Is there a way to tell at91bootstrap to fetch u-boot.bin from a specified eMMC offset?

Thanks a lot

ehristev, 2021-03-24:
Bootstrap can do the following : 'copy 1 MB from sd-card/emmc' , as raw, and then jump to the designated address in SDRAM. You can configure this address ( will be called demo-app image or something inside menuconfig) , and you should configure it to the u-boot linked address...
If you are storing the at91bootstrap in the emmc user partition , this may affect you.
You could enable the boot partitions in the emmc and store the at91bootstrap there.
Then you can configure Romcode to boot it from boot partition. This will leave user partition free for your use.
AT91Bootstrap cannot read the boot partitions.
If you just then copy the raw u-boot.bin to the start of the emmc user partition, then configure at91bootstrap to copy the first 1 MB into SDRAM and jump, it might work.

fabioestevam, 2021-03-24:
I confirmed that I am booting at91bootstrap from the boot partition.

I am flashing it in the boot partition by doing:
sam-ba -p serial -d sama5d2  -a sdmmc:0:1:1:8:5  -c writeboot:boot.bin

I copied u-boot.bin in the start of the eMMC user partition:
sam-ba -p serial -d sama5d2  -a sdmmc:0:1:0:8:5  -c write:u-boot.bin

Now I am trying your suggestion of  "copying raw 1MB from eMMC" approach, but I was not able to tell at91bootstrap to load 'raw' instead of a file from the FAT partition.

Here are my changes:
https://pastebin.com/raw/CTw33tEw

and at91bootstrap is still looking for a file named "none".

What would I need to change in order to copy the raw 1MB content of the eMMC user partition to SDRAM and execute from there?

Thanks!

fabioestevam, 2021-03-24:
I missed to copy the boot message showing that it still tries to look for a file inside the FAT partition:

AT91Bootstrap 3.10.2-00018-g1697f4f672a2-dirty (2021-03-24 16:23:57)

SDHC: fix in place for SAMA5D2 SoM VDDSDMMC over-consumption errata
The MAINCK frequency: 27000000
SD/MMC: Image: Read file none to 0x23f00000
MMC: ADMA supported
MMC: Specification Version 4.0 or higher
MMC: v5.0/5.01 detected
MMC: highspeed supported
MMC: Dual Data Rate supported
MMC: detecting buswidth...
MMC: 8-bit bus width detected
*** FATFS: f_open, filename: [none]: error
SD/MMC: Failed to load image

ehristev, 2021-03-24:
it's because you altered the source code for this. remove the changes from the pastebin, it's not what you want.
do 'make menuconfig' and in image loading strategy, select the option to load 1MB into SDRAM
after that look for the option that says 'demo application storage setup' and select there the proper address in SDRAM where to copy the 1MB, 
also make sure u-boot is linked to that address in it's linker config.

fabioestevam, 2021-03-24:
Ok, I removed the source code changes and did as you suggested:

AT91Bootstrap 3.10.2-00018-g1697f4f672a2 (2021-03-24 17:52:15)

SDHC: fix in place for SAMA5D2 SoM VDDSDMMC over-consumption errata
The MAINCK frequency: 27000000
SD/MMC: Image: Read file softpack.bin to 0x22000000
MMC: ADMA supported
MMC: Specification Version 4.0 or higher
MMC: v5.0/5.01 detected
MMC: highspeed supported
MMC: Dual Data Rate supported
MMC: detecting buswidth...
MMC: 8-bit bus width detected
*** FATFS: f_open, filename: [softpack.bin]: error
SD/MMC: Failed to load image

It still tries to load from a file name in the FAT partition.

Looking at driver/sdcard.c it seems that it can only read from file and not raw. 

One more question: I am using mainline U-Boot, target sama5d27_som1_ek_mmc_defconfig as reference.

Where is the linker config that you referred to so I can see the proper SDRAM address to pass to at91bootsrap?

Sorry for my basic questions. It is my first time using this SoC.

Thanks!

ehristev, 2021-03-25:
It looks like you are right. I thought it would work. It works probably for flashes and other kinds of media.
you could hack the code to do raw copy, but as of this moment it looks to not be implemented for sd-cards.
So I think the only way is for you to make a FAT partition and add u-boot.bin there.
you can create a FAT partition by mounting a file in a loopback , in a linux system. Then, this file , you can write it to the emmc using sam-ba.
This way you can avoid actually formatting the eMMC, but rather, formatting a file and then 'burning' the resulted file with sam-ba

fabioestevam, 2021-03-25:
Hi Eugen,

Thanks. This approach works and I was able to boot U-Boot :-)

The next step is to load the dtb, zImage and initramfs and boot Linux.

Thanks for your excellent help!

Cheers

ehristev, 2021-03-25:
Hi,

I am glad I could help.
If it's of interest to you to read raw from SD-card/eMMC, feel free to submit a patch that does it, and I can review/include it in mainline bootstrap .
Thanks and good luck !

closed 2021-03-26 by ehristev