linux4sam/at91bootstrap #122: Suspicious (?) RAM performance on SAMA5D2

zougloub, 2021-04-19:
It is my understanding that u-boot and Linux don't alter the RAM controller configuration so I'm raising this "issue" here.

I am not sure that it is normal to have such low memcpy (~200 MB/s) and memset (~ 400 MB/s) performance on the sama5d2-xplained hardware;  here are numbers from tinymembench:

```
    C copy backwards                                     :    124.6 MB/s (0.6%)
    C copy backwards (32 byte blocks)                    :    134.5 MB/s (0.9%)
    C copy backwards (64 byte blocks)                    :    122.7 MB/s
    C copy                                               :    188.0 MB/s
    C copy prefetched (32 bytes step)                    :    182.5 MB/s (0.1%)
    C copy prefetched (64 bytes step)                    :    189.5 MB/s
    C 2-pass copy                                        :    179.0 MB/s
    C 2-pass copy prefetched (32 bytes step)             :    182.5 MB/s
    C 2-pass copy prefetched (64 bytes step)             :    188.9 MB/s
    C fill                                               :    428.6 MB/s (0.2%)
    C fill (shuffle within 16 byte blocks)               :    428.4 MB/s
    C fill (shuffle within 32 byte blocks)               :    428.5 MB/s (0.2%)
    C fill (shuffle within 64 byte blocks)               :    130.0 MB/s
    ---
    standard memcpy                                      :    189.6 MB/s
    standard memset                                      :    428.5 MB/s
    ---
    NEON read                                            :    348.2 MB/s
    NEON read prefetched (32 bytes step)                 :    371.5 MB/s (0.1%)
    NEON read prefetched (64 bytes step)                 :    405.4 MB/s
    NEON read 2 data streams                             :    321.4 MB/s
    NEON read 2 data streams prefetched (32 bytes step)  :    407.7 MB/s
    NEON read 2 data streams prefetched (64 bytes step)  :    407.0 MB/s
    NEON copy                                            :    181.8 MB/s (0.1%)
    NEON copy prefetched (32 bytes step)                 :    181.7 MB/s
    NEON copy prefetched (64 bytes step)                 :    189.6 MB/s
    NEON unrolled copy                                   :    188.1 MB/s (1.6%)
    NEON unrolled copy prefetched (32 bytes step)        :    186.9 MB/s (0.5%)
    NEON unrolled copy prefetched (64 bytes step)        :    189.6 MB/s
    NEON copy backwards                                  :    134.5 MB/s
    NEON copy backwards prefetched (32 bytes step)       :    134.8 MB/s (0.6%)
    NEON copy backwards prefetched (64 bytes step)       :    185.1 MB/s (0.7%)
    NEON 2-pass copy                                     :    179.5 MB/s
    NEON 2-pass copy prefetched (32 bytes step)          :    187.0 MB/s
    NEON 2-pass copy prefetched (64 bytes step)          :    196.3 MB/s
    NEON unrolled 2-pass copy                            :    181.5 MB/s
    NEON unrolled 2-pass copy prefetched (32 bytes step) :    181.9 MB/s (0.1%)
    NEON unrolled 2-pass copy prefetched (64 bytes step) :    195.9 MB/s
    NEON fill                                            :    428.5 MB/s
    NEON fill backwards                                  :    428.4 MB/s
    VFP copy                                             :    188.2 MB/s (0.2%)
    VFP 2-pass copy                                      :    180.5 MB/s (0.2%)
    ARM fill (STRD)                                      :    428.5 MB/s
    ARM fill (STM with 8 registers)                      :    428.6 MB/s (0.1%)
    ARM fill (STM with 4 registers)                      :    428.4 MB/s
    ARM copy prefetched (incr pld)                       :    187.9 MB/s (0.5%)
    ARM copy prefetched (wrap pld)                       :    188.9 MB/s
    ARM 2-pass copy prefetched (incr pld)                :    184.7 MB/s
    ARM 2-pass copy prefetched (wrap pld)                :    184.7 MB/s (0.1%)
```

tinymembench was added on this image:
 https://files.linux4sam.org/pub/demo/linux4sam-2020.10/linux4sam-poky-sama5d2_xplained-headless-2020.10.img.bz2
and I also replaced the BOOT.BIN with the latest master of at91bootstrap, getting the same results.

The sama5d2 performance that I saw would be what to expect with DDR3 clocked at 50 MHz and without DLL; should I expect more or not? I am seeing 5x this memset/memcpy performance (1000 MB/s / 2000 MB/s) on an imx7 equipped with worse DDR3 memory...

I don't have a lot of experience with this chip but a cursory review of the DDR configuration in the board files, vs. the sama5d2 datasheet and the RAM datasheet doesn't shock me.

Tagging @swapna-gg1 as I saw that they added a memcpy enhancement in dd4da885 and I suppose benchmarks were ran for the occasion.

Thank you,

ehristev, 2021-04-19:
Hi,

Have you also tried the lmbench memory performance package ? namely for bw_mem tool

ehristev, 2021-04-19:
sama5d2 datasheet states 125 Mhz max clock with DLL off for DDR3 type of memory.

ehristev, 2021-04-19:
if the memset/memcpy is done by CPU accesses only, it also involves bus access/latency, and CPU speed. It is not just about the memory itself.
I remember imx7 has a much higher CPU frequency and it's a different cortex, plus the busses should be much higher clocked.

zougloub, 2021-04-19:
`bw_mem` (using 256 MB for read and 128 MB for copies) gives similar results to those of `tinymembench`:
- read: 350 MB/s
- write: 425 MB/s
- copy: 180 MB/s

[This info](https://www.digikey.ca/en/product-highlight/a/atmel/sama5d3-arm-cortex-a5) from the [similar sama5d3](https://www.microchip.com/en-us/products/microcontrollers-and-microprocessors/32-bit-mpus/sama5) indicates “32-bit wide DDR controller running up to 166 MHz that offers up to 1328 MB/s of bandwidth” and I would expect to be closer to that performance level than what I'm seeing.

zougloub, 2021-04-19:
Side-note, I notice you said:

> sama5d2 datasheet states 125 Mhz max clock with DLL off for DDR3 type of memory.

Yet at91bootstrap's defconfig for sama5d2_xplained has a 166 MHz bus clock  / DDR clock, and more than a couple of  people have worked on the timings (eg. @noglitch) so I would expect this to work... but I'm not sure why not.

That 1328 MB/s figure from the quote above is `166 MHz * 32 bit * 2 / 8 bit/B`.

ehristev, 2021-04-19:
if you have a DDR controller at 166 Mhz, then, for each clock cycle, you transfer 2 times 32 bits right ? so that's 8 bytes.
8 * 166 = 1328 . Multiply that with 1 million (166 is Mhz), you get a pure theoretical transfer rate of 1328 MB/s maximum .
This is the physical limit of the DDR3 itself.
This would be achievable if you would just use some kind of machine to keep transferring data on the bus. In real life, even with a memcpy or memset, this is not really what you can achieve in terms of usable data transferred. 
You have caches, you have metadata, paging, you have also code that needs to be loaded from memory, you have cache trashing, you have queues, you have other masters on the bus that request lock on the bus and on the ddr controller. You have big delays when switching the memory from read to write operations, you have AXI and AHB busses that interconnect everything, and they are clocked at different speeds. each transaction is created, requested and then queued on several buffers inside the architecture. Then you have to wait for the reply , until it comes back, then update the CPU, and so on.
At copy operation for example, you first need to create a transaction to read data, a full cache line for example, this request is being sent from the CPU, it goes through the bus, it must be queued to the DDR controller. then the DDR controller switches to read mode, and then it will activate the bank row and column where the data resides. At this moment maybe it will be performed at your speed of 1328 MB/s. after this the response is created and queued back to the interconnect, it will go to the CPU, inside the cache lines. And after that, because you are performing a copy, you will need to queue another transaction to write to a different area of the DDR. Then all the process is done again, and the controller must switch the memory to write transaction , and so on.
So maybe at 180 MB/s you do not reach the maximum of the ddr bandwidth, but it would also be interesting to see what real bandwidth for the DDR would be used, if for example you are also doing image acquisition, which heavily writes the DDR using a different master ( the Image sensor controller), without CPU involvement , until the CPU has to read back the data from the DDR. Doing this in parallel might get an overall usage of the bandwidth that is higher than what you see by only using the CPU

zougloub, 2021-04-19:
I fully understand your point but I'd tend to believe that on a system doing nothing except `memset()` we should reach something that's close to the limit.
For reference, on a Raspberry Pi Zero, which has an ARM1176 (ARMv6) processor, [tinymembench gives](http://ix.io/1niO) `memcpy()` at 400 MB/s, `memset()` at 1500 MB/s, which is pretty close to the maximum value of 1600 MB/s for a 200 MHz bus clock (which is a guess because that hardware is not terribly well documented).

ehristev, 2021-04-19:
it may be that memset is optimized somehow on these architectures.

ehristev, 2021-04-19:
it would be interesting to see the rasp pi doing the same thing at half the CPU frequency. See how much impact the CPU has on the memset operation.

zougloub, 2021-04-20:
Here are the results from the Raspberry pi zero at 500 MHz:
```
Architecture:        armv6l
Byte Order:          Little Endian
CPU(s):              1
On-line CPU(s) list: 0
Thread(s) per core:  1
Core(s) per socket:  1
Socket(s):           1
Vendor ID:           ARM
Model:               7
Model name:          ARM1176
Stepping:            r0p7
CPU max MHz:         500.0000
CPU min MHz:         500.0000
BogoMIPS:            498.89
Flags:               half thumb fastmult vfp edsp java tls

 C copy backwards                                     :    126.1 MB/s (0.7%)
 C copy backwards (32 byte blocks)                    :    127.1 MB/s (0.8%)
 C copy backwards (64 byte blocks)                    :    127.2 MB/s (0.8%)
 C copy                                               :    126.2 MB/s
 C copy prefetched (32 bytes step)                    :    321.0 MB/s
 C copy prefetched (64 bytes step)                    :    189.1 MB/s
 C 2-pass copy                                        :    128.1 MB/s (0.2%)
 C 2-pass copy prefetched (32 bytes step)             :    234.3 MB/s (0.7%)
 C 2-pass copy prefetched (64 bytes step)             :    185.0 MB/s (0.9%)
 C fill                                               :    736.3 MB/s (0.7%)
 C fill (shuffle within 16 byte blocks)               :    736.7 MB/s (0.7%)
 C fill (shuffle within 32 byte blocks)               :    736.2 MB/s
 C fill (shuffle within 64 byte blocks)               :    736.5 MB/s (0.9%)
 ---
 standard memcpy                                      :    374.7 MB/s (1.1%)
 standard memset                                      :   1240.3 MB/s (0.4%)
 ---
 VFP copy                                             :    157.7 MB/s (0.9%)
 VFP 2-pass copy                                      :    131.5 MB/s
 ARM fill (STRD)                                      :    736.7 MB/s (0.1%)
 ARM fill (STM with 8 registers)                      :   1026.4 MB/s (1.5%)
 ARM fill (STM with 4 registers)                      :   1240.0 MB/s (0.6%)
 ARM copy prefetched (incr pld)                       :    374.9 MB/s (1.2%)
 ARM copy prefetched (wrap pld)                       :    215.3 MB/s
 ARM 2-pass copy prefetched (incr pld)                :    266.0 MB/s
 ARM 2-pass copy prefetched (wrap pld)                :    224.4 MB/s (0.5%)
```

zougloub, 2021-04-20:
And when downclocking at 200 MHz on the pi zero I get:
```
 C copy backwards                                     :     96.3 MB/s (1.0%)
 C copy backwards (32 byte blocks)                    :     96.0 MB/s (0.6%)
 C copy backwards (64 byte blocks)                    :     96.0 MB/s (4.9%)
 C copy                                               :    101.4 MB/s
 C copy prefetched (32 bytes step)                    :    183.4 MB/s
 C copy prefetched (64 bytes step)                    :    131.9 MB/s (0.7%)
 C 2-pass copy                                        :     81.4 MB/s (0.8%)
 C 2-pass copy prefetched (32 bytes step)             :    132.3 MB/s
 C 2-pass copy prefetched (64 bytes step)             :    105.6 MB/s
 C fill                                               :    432.6 MB/s
 C fill (shuffle within 16 byte blocks)               :    432.6 MB/s
 C fill (shuffle within 32 byte blocks)               :    432.8 MB/s (1.9%)
 C fill (shuffle within 64 byte blocks)               :    432.7 MB/s (1.6%)
 ---
 standard memcpy                                      :    252.6 MB/s (1.2%)
 standard memset                                      :    507.0 MB/s (1.2%)
 ---
 VFP copy                                             :    112.6 MB/s (1.2%)
 VFP 2-pass copy                                      :     87.5 MB/s
 ARM fill (STRD)                                      :    432.8 MB/s (2.0%)
 ARM fill (STM with 8 registers)                      :    515.8 MB/s (9.5%)
 ARM fill (STM with 4 registers)                      :    507.2 MB/s (1.5%)
 ARM copy prefetched (incr pld)                       :    253.2 MB/s (0.7%)
 ARM copy prefetched (wrap pld)                       :    211.6 MB/s
 ARM 2-pass copy prefetched (incr pld)                :    165.4 MB/s
 ARM 2-pass copy prefetched (wrap pld)                :    162.3 MB/s
```
... which is still faster than the sama5d2.

So CPU frequency does have an impact, but at 200 MHz CPU we're still not seeing the bottleneck that I see on sama5d2.

Maybe the memory controller is very "low power" but I can't help but think that there may be something going on here.

ehristev, 2021-09-27:
After internal discussions with our design team, we have several reasons to explain the lower performance compared to Raspberry PI.
First, our DDR controller does not embed buffers nor fifos and this affect the performance greatly.
If Raspberry is based on ARM AXI bus and not on AHB bus, this means point to point configuration which brings a lot of performance, but it's not what happens in real life applications.
Thus in a synthetic benchmark like tinymembench the performance will be much better, as stated above (point to point, buffers and fifos embedded ).
The expected performance for this product with this synthetic benchmark is what you have found during your tests.

closed 2021-09-27 by ehristev