linux4sam/at91bootstrap #126: v4 fails to load Linux kernel

edgar-bonet, 2021-05-27:
I am [still][124] trying to use AT91Bootstrap 4 to boot the [Acqua SoM][acqua] from Acme Systems. This time it builds smoothly, but fails to load the Linux kernel.

Here is my defconfig:

```text
CONFIG_SAMA5D3X=y
CONFIG_CPU_CLK_528MHZ=y
CONFIG_DEBUG=y
CONFIG_DEBUG_VERY_LOUD=y
CONFIG_SDCARD=y
CONFIG_LOAD_LINUX=y
CONFIG_IMAGE_NAME="zImage"
```

In order to get a complete system, I used Buildroot 2021.05-rc2 (with some extra packages):
* built with `acmesystems_acqua_a5_512mb_defconfig`, which is based on AT91Bootstrap&nbsp;3
* flashed `output/images/sdcard.img` into a microSD card
* tested this configuration → it works
* mounted the boot partition of the card on my computer, and there:
  * replaced boot.bin (AT91Bootstrap&nbsp;3) with the newly built bootloader (build/binaries/sama5d3-sdcardboot-linux-zimage-dt-4.0.0-rc2.bin)
  * renamed the device tree: at91-sama5d3\_acqua.dtb → sama5d3.dtb, as this is the name the new bootloader expects
* tested again → fail

Here is the “very loud” log of the console output:

```text
RomBOOT
 ba_offset = 0xa ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xffffea00: 0x0 0x30081e 0xc00034 0x22228226
0xffffea10: 0x21b0c0a 0x62262 0x33338 0x10000
0xffffea20: 0x16 0x8 0x0 0x0
0xffffea30: 0x6 0x70404 0x0 0x0
0xffffea40: 0x0 0x8030a 0x8030a 0x8030a
0xffffea50: 0x8030a 0x0 0x0 0x1
0xffffea60: 0x0 0x0 0x1 0x0
0xffffea70: 0x0 0x0 0x0 0x3d00
0xffffea80: 0xf00 0xf00 0xf00 0xf00
0xffffea90: 0x300 0x0 0x0 0x0
0xffffeaa0: 0x0 0x0 0x0 0x0
0xffffeab0: 0x0 0x0 0x0 0x0
0xffffeac0: 0x0 0x0 0x0 0x0
0xffffead0: 0x0 0x0 0x0 0x0
0xffffeae0: 0x0 0x0 0x0 0x200
0xffffeaf0: 0x484d5044 0x44524320 0x0 0x10140
0xffffeb00: 0x0 0x0 0x0 0x0
0xffffeb10: 0x0 0x0 0x0 0x0
0xffffeb20: 0x0 0x0 0x0 0x0
0xffffeb30: 0x0 0x0 0x0 0x0
0xffffeb40: 0x0 0x0 0x0 0x0
0xffffeb50: 0x0 0x0 0x0 0x0

AT91Bootstrap 4.0.0-rc2-00012-gffd4b694 (2021-05-27 15:46:29)

SD/MMC: Image: Read file zImage to 0x22000000
mmc_verify_operating_condition
Cmd: 1 Response Time-out
SD: Card Capacity: High or Extended
sd card identified with CID = 0x1391780 0x281dc750 0x36344703 0x574457ff
sdcard_identification success
SD: Specification Version 3.0X
SD/MMC: dt blob: Read file sama5d3.dtb to 0x21000000
KERNEL: try as zImage: magic=0x0
KERNEL: try as uImage: magic=0xff7f
KERNEL: Got unsupported magic!
	as uImage magic: 0xff7f
	as zImage magic: 0x0
SD/MMC: Failed to load image
```

It seems to be complaining about an incorrect magic number. However, the magic seems fine to me. According to [load\_kernel.c][load_kernel], the expected value is `0x016f2818` at offset `0x0024`, and this is indeed what I have (in little endian order):

```text
$ hd -v /media/edgar/1E71-A9DE/zImage | head -4
00000000  00 00 a0 e1 00 00 a0 e1  00 00 a0 e1 00 00 a0 e1  |................|
00000010  00 00 a0 e1 00 00 a0 e1  00 00 a0 e1 00 00 a0 e1  |................|
00000020  05 00 00 ea 18 28 6f 01  00 00 00 00 d0 4e 47 00  |.....(o......NG.|
00000030  01 02 03 04 45 45 45 45  80 3b 00 00 00 90 0f e1  |....EEEE.;......|
```

Any hint on how to tackle this issue would be appreciated.

[124]: ../issues/124
[acqua]: https://www.acmesystems.it/acqua
[load_kernel]: ../blob/ffd4b694a895aeb311275f45d65b3ca24ae0f9d5/driver/load_kernel.c#L261-L268

TonyHan11, 2021-05-28:
While building the bootstrap using version 4 with your defconfig, `CONFIG_DDR_SET_BY_JEDEC=y` occurs in `.config` file, so the DDRAMC parameters might be unsuitable for your board.
As in At91bootstrap 3 `board/acme/sama5d3_acqua/sama5d3_acqua.c` is quite similar to `board/sama5d3_xplained/sama5d3_xplained.c`, some configurations for sama5d3-xplained may be used for Acqua board.
I have no Acqua board to verify booting linux, would you like to add the following configurations and test again? Thanks.
```
CONFIG_DDR_SET_BY_DEVICE=y
CONFIG_DDR_MT47H64M16=y
CONFIG_NOT_DQS_DISABLED=y
CONFIG_TWI1=y
CONFIG_ACT8865=y
CONFIG_ACT8865_SET_VOLTAGE=y
CONFIG_VOLTAGE_OUT2=1250
CONFIG_VOLTAGE_OUT5=3300
```

edgar-bonet, 2021-05-28:
Thank you @TonyHan11, that did the trick! It loads the kernel. I still have to figure out how to pass kernel command line arguments, but that's another issue.

Now, I find this situation somewhat annoying, as the Acqua board comes in various versions with different brands of RAM (Micron, Samsung, Hynix, Elpida and Alliance). I assumed the JEDEC parameters were conservative enough to fit all those versions, but that seems to not be the case. If one has to tune the defconfig for the specific RAM in use, I guess I would rather stick with AT91Bootstrap&nbsp;3. :-(

ehristev, 2021-05-28:
@edgar-bonet it might be the case that one of the memory vendors are actually out of JEDEC space... I would expect the same, that the JEDEC parameters work fine on all the memories, since they are JEDEC.
So, either there is a bug in our JEDEC configuration, or the memory does not comply with it.
@TonyHan11 the JEDEC configration works on our board, sama5d3_xplained ?
Another option is try to see the exact differences between this DDR configuration and the JEDEC. Then identify which parameter causes the problem.

ehristev, 2021-05-28:
> Thank you @TonyHan11, that did the trick! It loads the kernel. I still have to figure out how to pass kernel command line arguments, but that's another issue.
@edgar-bonet check CONFIG_OVERRIDE_CMDLINE

edgar-bonet, 2021-05-28:
@ehristev: I added

```text
CONFIG_OVERRIDE_CMDLINE=y
CONFIG_LINUX_KERNEL_ARG_STRING="console=ttyS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4 rootwait"
```

to the defconfig, but that broke the build. It would seem only macros are accepted in CONFIG\_LINUX\_KERNEL\_ARG\_STRING. Is that expected or should it be reported as an issue?

ehristev, 2021-05-28:
CONFIG_LINUX_KERNEL_ARG_STRING=\"console=ttyS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4 rootwait\"

Have you tried it like this ?

edgar-bonet, 2021-05-28:
@ehristev: Yes, exactly. Combined with `CONFIG_OVERRIDE_CMDLINE`.

ehristev, 2021-05-28:
Ok, sorry, github edited my string. Here is what I meant :`
`
CONFIG_LINUX_KERNEL_ARG_STRING=\"console=ttyS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4 rootwait\"
`
notice the backslashes

edgar-bonet, 2021-05-28:
@ehristev: I just tried with the backslashes. It seems `CONFIG_LINUX_KERNEL_ARG_STRING` is then ignored. When I typed `make`, this displayed the “gcc FLAGS”. Among those I saw:

```text
-DCMDLINE="CONFIG_ARG_CONSOLE CONFIG_ARG_PRINTK CONFIG_ARG_MTD CONFIG_ARG_ROOT CONFIG_ARG_MEMTEST CONFIG_ARG_ROOTFSTYPE CONFIG_ARG_DELAY CONFIG_ARG_CMA CONFIG_ARG_RW CONFIG_ARG_ROOTWAIT"
```

Then, the build succeeded, but the `strings` command revealed this string embedded within the binary:

```text
console=ttyS0,115200 earlyprintk mtdparts=atmel_nand:12M(bootstrap/kernel)ro,-(rootfs) root=/dev/mmcblk0p2 rootfstype=ubifs ubi.mtd=1 rootdelay=2 rw rootwait
```

This is the only occurrence of “console” within the binary. Notice the wrong rootfstype, and the extra parameters. I didn't even try to boot it.

ehristev, 2021-05-28:
@TonyHan11 could you have a look at this please ?

TonyHan11, 2021-06-08:
The Linux kernel arguments can not be changed due to the string attributes of  `config ARG_*` are empty. A patch is needed to enable the configurations of these arguments. With the patch users can override Linux kernel arguments as they want.
For example:
```
CONFIG_ARG_CONSOLE="console=ttyS0,115200"
CONFIG_ARG_DELAY=" rootdelay=2"
CONFIG_ARG_MEMTEST=""
CONFIG_ARG_MTD=" mtdparts=atmel_nand:12M(bootstrap/kernel)ro,-(rootfs)"
CONFIG_ARG_PRINTK=" earlyprintk"
CONFIG_ARG_ROOT=" root=/dev/mmcblk0p2"
CONFIG_ARG_ROOTFSTYPE=" rootfstype=ubifs ubi.mtd=1"
CONFIG_ARG_ROOTWAIT=" rootwait"
CONFIG_ARG_RW=" rw"
```

--------- bellow is the patch --------
```
diff --git a/Config.in.kernel b/Config.in.kernel
index 5c290ef2..f2ffed3b 100644
--- a/Config.in.kernel
+++ b/Config.in.kernel
@@ -15,51 +15,51 @@ config LINUX_KERNEL_ARG_STRING
        default "CONFIG_ARG_CONSOLE CONFIG_ARG_PRINTK CONFIG_ARG_MTD CONFIG_ARG_ROOT CONFIG_ARG_MEMTEST CONFIG_ARG_ROOTFSTYPE CONFIG_ARG_DELAY CONFIG_ARG_CMA CONFIG_ARG_RW CONFIG_ARG_ROOTWAIT"

 config ARG_CMA
-       string
+       string "Linux bootparam: Contiguous Memory Area support"
        default " cma=256m" if SAMA7G5
        default ""

 config ARG_CONSOLE
-       string
+       string "Linux bootparam: the serial port after boot"
        default "console=ttyS0,115200"

 config ARG_DELAY
-       string
+       string "Linux bootparam: option to delay before attempting to mount the root fs"
        default " rootdelay=2"

 config ARG_MEMTEST
-       string
+       string "Linux bootparam: option to enable memory test"
        default " memtest=0" if SAMA7G5
        default ""

 config ARG_MTD
-       string
+       string "Linux bootparam: option about MTD parts"
        default " mtdparts=atmel_nand:12M(bootstrap/kernel)ro,-(rootfs)" if SDCARD
        default " mtdparts=atmel_nand:8M(bootstrap/kernel)ro,-(rootfs)" if SDCARD
        default ""

 config ARG_PRINTK
-       string
+       string "Linux bootparam: option to use earlyprintk"
        default " earlyprintk"

 config ARG_ROOT
-       string
+       string "Linux bootparam: the device to be used as the rootfs while booting"
        default " root=ubi0:rootfs" if !(SDCARD || QSPI)
        default " root=/dev/mmcblk0p2" if QSPI
        default " root=/dev/mmcblk1p2" if SDHC1
        default " root=/dev/mmcblk0p2"

 config ARG_ROOTFSTYPE
-       string
+       string "Linux bootparam: the type of the filesystem to load the rootfs"
        default " rootfstype=ubifs ubi.mtd=1"
        default " rootfstype=ext4" if SAMA7G5

 config ARG_ROOTWAIT
-       string
+       string "Linux bootparam: option to wait for root device to show up"
        default " rootwait"

 config ARG_RW
-       string
+       string "Linux bootparam: mount the rootfs with specified attribute like read-only, read/write, ..."
        default " rw"
 endif
```

ehristev, 2021-06-08:
@TonyHan11 what happens if someone wants to use a parameter not listed here ?
for example atmel.pm_modes ?
why did you add these as Kconfig options instead of passing a single Kconfig string as kernel arguments for the bootargs ?
also, what happens if someone configures for example ARG_PRINTK=" console=ttyS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4 rootwait" and no other argument option ?

TonyHan11, 2021-06-08:
The final `CMDLINE` will be the concatenation of all the configurations in `LINUX_KERNEL_ARG_STRING`.
It aims to make every parameter configurable. In at91bootstrap only the parameters occur in previous `Config.in.linux_arg` files are listed as there're too many  parameters for the kernel.
Might it is better to use a single Kconfig string here.

ehristev, 2021-06-11:
@edgar-bonet  Could you try Tony's patch and let me know how it works for you. Then I will apply it , if it suits your needs.

edgar-bonet, 2021-06-12:
I did a few tests, using both v4.0.0-rc3 and the current 4.x branch (59c8963d8e6a08222616d1b384e1009ba05e6b82), with or without @TonyHan11's patch, with three different defconfigs. That's 12&nbsp;combinations.

This is the first defconfig:

```text
CONFIG_SAMA5D3X=y
CONFIG_CPU_CLK_528MHZ=y
CONFIG_DEBUG=y
CONFIG_DEBUG_VERY_LOUD=y
CONFIG_DDR_SET_BY_DEVICE=y
CONFIG_DDR_MT47H64M16=y
CONFIG_NOT_DQS_DISABLED=y
CONFIG_SDCARD=y
CONFIG_LOAD_LINUX=y
CONFIG_OVERRIDE_CMDLINE=y
CONFIG_LINUX_KERNEL_ARG_STRING="console=ttyS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4 rootwait"
CONFIG_IMAGE_NAME="zImage"
CONFIG_TWI1=y
CONFIG_ACT8865=y
CONFIG_VOLTAGE_OUT2=1250
CONFIG_VOLTAGE_OUT5=3300
```

The second one is identical, except for the quotes around the Linux command line which were backslash-escaped:

```text
CONFIG_LINUX_KERNEL_ARG_STRING=\"console=ttyS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4 rootwait\"
```

The third one was derived from the previous ones by removing `CONFIG_LINUX_KERNEL_ARG_STRING` and adding the following lines:

```text
CONFIG_ARG_DELAY=""
CONFIG_ARG_MTD=""
CONFIG_ARG_PRINTK=""
CONFIG_ARG_ROOTFSTYPE=" rootfstype=ext4"
```

The results were independent of the version of AT91Boostrap I used (RC3 or 4.x branch).

With the first defconfig (`CONFIG_LINUX_KERNEL_ARG_STRING="..."`), the build always failed with errors like this:

```text
driver/load_kernel.c: In function ‘board_override_cmd_line’:
<command-line>: error: ‘console’ undeclared (first use in this function)
driver/load_kernel.c:374:9: note: in expansion of macro ‘CMDLINE’
  374 |  return CMDLINE;
      |         ^~~~~~~
```

With the second defconfig (escaped quotes), the build succeeded, but the provided command line was always ignored.

With the third defconfig (individual `CONFIG_ARG_*` parameters), the build succeeded and:

 - with an unpatched AT91Bootstrap, judging from the output of the `strings` command, my settings for the command line were ignored

 - with the patch applied, `strings` showed that the intended command line was embedded in the bootloader's binary.

In the second case, however, that command line was apparently not used at run time. Among the console messages during the boot I saw the following lines (which were not consecutive):

```text
AT91Bootstrap 4.0.0-rc3-00002-g59c8963d-dirty (2021-06-12 16:58:46)
Kernel command line: mem=256M 
VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
Please append a correct "root=" boot option; here are the available partitions:
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```

TonyHan11, 2021-06-15:
For the third defconfig above, there's no code to print the message `Kernel command line`. Could the message be dumped by the Linux kernel?
It is better to add the following code in main.c to dump the exact value of `CMDLINE` to see if it is the one wanted.
`usart_puts("======\r\nCMDLINE:" CMDLINE "\r\n======\r\n");`

edgar-bonet, 2021-06-15:
@TonyHan11 wrote:
> Could the message [`Kernel command line`] be dumped by the Linux kernel?

Indeed, [it comes from the kernel][kernel].

> It is better to add the following code in main.c [...]

I added that right after `display_banner();`. Selected output lines:

```text
AT91Bootstrap 4.0.0-rc3-00002-g59c8963d-dirty (2021-06-15 12:30:51)
CMDLINE:console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rw rootwait
Kernel command line: mem=256M 
```

It looks like the bootloader and the kernel do not agree on what the command line is. How is the command line provided to the kernel? Is it through the /chosen/bootargs node of the device tree, or some other means?

I noticed [this commit][] that dynamically modifies the command line. Could it be related to this bootloader/kernel disagreement?

[kernel]: /torvalds/linux/blob/v5.4/init/main.c#L607
[this commit]: ../commit/1d0bc22bd8718c30ff0518b6d6c839f035ca58b6

TonyHan11, 2021-06-16:
I have verified with “CONFIG_OVERRIDE_CMDLINE=y” on sam9x60-ek board and found the `CMDLINE` dumped by at91bootstrap is the same as the one dumped by the Linux kernel. Not know why they are different on your board. Maybe you can try to hard-code `CMDLINE` in at91bootstrap and find out whether it is the `CMDLINE` issue that causes the failure of loading Linux kernel.

When using “CONFIG_OVERRIDE_CMDLINE=y”, the `CMDLINE` is dynamically generated to a char array “cmdline_buf[256]” which locates at the data region. I’m not sure whether or not it could be destroyed during loading the kernel now.

edgar-bonet, 2021-06-16:
**Solved!** 😃

After adding quite a few calls to `dbg_info()` here and there, I think I found the cause of my problem: in device/sama5d3/sama5d3.c, [the function `board_override_cmd_line()`][src] returns a `NULL` command line, unless `CONFIG_LOAD_ANDROID` is set.

The following patch makes my board bootable:

```patch
diff --git a/device/sama5d3/sama5d3.c b/device/sama5d3/sama5d3.c
index 7c27b836..268fe0ab 100644
--- a/device/sama5d3/sama5d3.c
+++ b/device/sama5d3/sama5d3.c
@@ -311,6 +311,8 @@ char *board_override_cmd_line(void)
 		cmdline = CMDLINE " androidboot.hardware=sama5d3x-pda";
 	else
 		cmdline = CMDLINE " androidboot.hardware=sama5d3x-ek";
+#else
+	cmdline = CMDLINE;
 #endif
 	return cmdline;
 }
```

Note that this is needed in addition to @TonyHan11's patch to Config.in.kernel.

[src]: ../blob/v4.0.0-rc3/device/sama5d3/sama5d3.c#L304-L316

ehristev, 2021-06-16:
@edgar-bonet nice !
Let's see if we really need that 'load android' or we can ditch it. @noglitch what do you think ?

noglitch, 2021-06-16:
@ehristev : yes, I think we can ditch it.
@edgar-bonet :+1: Nice finding!

ehristev, 2021-06-16:
Okay, so, @edgar-bonet if you like, you can create a patch (PR if you like) and remove that load android part completely and have your board working, then I can apply it and with Tony's patch we should be able to boot fine your board.
Then you can submit another patch with a defconfig for your board and join the club of bootstrap4 boards !

edgar-bonet, 2021-06-16:
I am not quite sure how to proceed now. If, as requested, I remove the `CONFIG_LOAD_ANDROID`-specific code, then the function `board_override_cmd_line()` ends up reduced to this:

```c
char *board_override_cmd_line(void)
{
	return CMDLINE;
}
```

This is identical to [the weak default version][weak] of the function, so in this case I would rather remove this implementation altogether, and rely instead on the default.

But then, the default implementation is the _only_ implementation in the whole source tree. I would be tempted to ditch it and replace [its only call][call]:

```c
bootargs = board_override_cmd_line();
```

with:

```c
bootargs = CMDLINE;
```

Is this acceptable?

My understanding is that this function was introduced specifically for supporting the Android-specific case. At least that's what it looks to me, judging from the code of [the commit that introduced it][commit]. The commit log, however, invokes the more general idea of a facility for “allow[ing] boards to set their own command-line”. Should this facility be retained even though no board uses it anymore? Should I remove it in the name of YAGNI?

[weak]: ../blob/v4.0.0-rc3/driver/load_kernel.c#L372-L375
[call]: ../blob/v4.0.0-rc3/driver/load_kernel.c#L396
[commit]: ../commit/d3c3193173863af49f30392c1c05c4ff403d436a

ehristev, 2021-06-17:
`
> 
> This is identical to [the weak default version](../blob/v4.0.0-rc3/driver/load_kernel.c#L372-L375) of the function, so in this case I would rather remove this implementation altogether, and rely instead on the default.
> 
Yes I think that's the best way for now. remove it and rely on the weak version.
If someone wants to add extra arguments for, let's say android, they can add a new symbol like for 'root=..' or 'cma=..' which already exist

ehristev, 2021-06-23:
Prerequisites merged !
I think you can send another PR with the defconfig now, there shouldn't be any blocking point anymore.

edgar-bonet, 2021-06-23:
Great!

I still have some things to sort out (e.g. how to handle devices with various amounts of RAM), but I can close this issue now.

Tank-you all for this!

closed 2021-06-23 by edgar-bonet