linux4sam/at91bootstrap #127: -nostartfiles is not an LDFLAG

celledge, 2021-06-03:
Newer version of GCC (10.2.0 tested) have added extra error checking to the LD linker. If you now pass a flag that it doesn't recognize it will suggest an alternative. Additionally `-nostartfiles` is a GCC flag, not a LD flag. Together this causes the at91bootstrap make file to generate a linking error:

`arm-poky-linux-gnueabi-ld: Error: unable to disambiguate: -nostartfiles (did you mean --nostartfiles ?)`

Explanation available here:
https://sourceware.org/pipermail/binutils/2021-June/116826.html

I think `-nostartfiles` was just silently ignored previously.

ehristev, 2021-06-07:
Hi,

I am using this GCC version
gcc version 10.2.1 20201103 (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16))

I do not see this warning. How did you hit this problem ?
Anything special you did ?
I see you are using the yocto/poky toolchain in fact.

ehristev, 2021-06-07:
According to GCC documentation I see this parameter is available for the linker 
https://gcc.gnu.org/onlinedocs/gcc/Link-Options.html

celledge, 2021-06-07:
I'm sorry, this is actually caused by binutils version 2.36. The 10.2.1 GCC ARM toolchain you are using shipped with binutils version 2.35.1. arm-none-linux-gnueabihf-ld is just an alias for the binutils ld executable.

They have identical sha256 hashes:
788345c3d11cffae5f50f1e74d87575afbeaa8b2824cfe518d7e5f4cb23b55f7  arm-none-linux-gnueabihf-ld
788345c3d11cffae5f50f1e74d87575afbeaa8b2824cfe518d7e5f4cb23b55f7  ../arm-none-linux-gnueabihf/bin/ld

What -nostartfiles is actually doing is:
-n => Do not page align data, but this flag is already specified elsewhere on the LDFLAGS, so not a problem
-ostartfiles => Output to file named "startfiles", but this is being overridden by the later -o flag

I tried to remove the -nostartfiles from the makefile to see if it changed anything in the map file, but I discovered that if I touch the Makefile at all, even just adding a comment that the output map file changed on me. Not sure what's causing that yet, but it makes it hard to compare the differences with and without the flag. If I use a Makefile with an extra comment line, there doesn't appear to be a difference in the map files with and without the -nostartfiles.

celledge, 2021-06-07:
Oh. The difference in the map files was being generated by the addition of a '-dirty' statement to the DAT91BOOTSTRAP_VERSION. Once it was made to always be dirty, there was no apparent difference with and without the -nostartfiles when using GCC 10.2.1 / Binutils 2.35.1.

If using Binutils 2.36.1, the link with fail due to -nostartfiles being an invalid flag.

ehristev, 2021-06-08:
I believe the purpose of the -nostartfiles is to stop the toolchain from actually using start files . 
Maybe the correct place to put this flag is in CC flags and not the LDFLAGS ?
Have you tried to place it there and see what happens?

Thanks for shedding some light on this.

celledge, 2021-06-08:
I don't think -nostartfiles needed at all. It's intended to be used when using gcc as the final linker:
`gcc -nostartfiles -o foo.elf foo.o bar.o`
That would generate a binary with the start files excluded.

Instead this project is calling the binutils linker and telling it exactly what objects to include:
`ld -n -o foo.elf foo.o bar.o`
I suspect this decision was made to make it easy to pass in the assembler start sequence from crt0_gnu.S. It's possible the original project used gcc as the linker early in development until it was decided to swap to binutils which left the flag as vestigial but not throwing any error.

ehristev, 2021-06-08:
@nirvann Hi, what's your take on this one ? As you have tested bootstrap on various platforms and toolchains.
Thanks !

nirvann, 2021-06-08:
Hi, I didn't met the linker error message either.
But I agree with Nick Clifton's answer on binutils ML.

This `-nostartfiles` argument has been erroneously given to _ld_ since at least Year 2010 in commit 6ea5bf7cdbe0befdbe799b1f26409d5c7e185895. There are traces that `-Wl,` was used prior to that, showing that the linker was called indirectly in the past.
Now `-nostartfiles` simply has to be removed from _LDFLAGS_.
As pointed out by @celledge in our _ld_ command line there are explicit `-n -o $(BINDIR)/$(BOOT_NAME).elf` options, hence removing `-nostartfiles` will not alter the build.

ehristev, 2021-06-08:
@celledge Do you wish to make a patch for this, or you wish me to do it ? 
You would either be the author (first case) or the suggester (Suggested-by) (second case)

Thanks !

celledge, 2021-06-10:
@ehristev Is this pull request sufficient or do you need anything else?

ehristev, 2021-06-10:
Yes it's fine. I will apply this soon.

ehristev, 2021-06-11:
Fixed by applying patch. Thanks again !

closed 2021-06-11 by ehristev

tpetazzoni, 2021-08-07:
Could you make a new release of at91bootstrap 3.x with this fix? Thanks!

ehristev, 2021-08-16:
3.10.3 tagged and pushed.
Thanks for the heads up.
(I was out of office)