linux4sam/at91bootstrap #144: Contrib folder removed in v4.x

elockman, 2021-12-30:
I am building code for a custom board using the sama5d2.  My builds have broken after the at91bootstrap v4.x release.
In v3.x, I added my custom board config to a contrib/subfolder, much like Acme Systems did.  That folder has been removed in v4.x.

Is this an oversight, or is there now a new way to properly add custom board configs?

elockman, 2021-12-30:
I see the transition made in v4.x.  The bootstrap is now device-centric, instead of board-centric.

ehristev, 2021-12-30:
You are correct, now being a device centric approach, all you have to do is resubmit a new defconfig  for your custom boards, you do not have to duplicate code anymore.
Normally you shouldn't have to write any new line of code, but if there is something like a quirk, we can try to sort it out.

Contrib subdirectory was not removed, it was not moved to 4.x because it didn't make sense... it was a lot of code duplication. We expect people who want to move to 4.x to submit a patch with a new defconfig for their board. We could not do it since we lack the boards to test it.
But nothing stops people to continue using 3.x if they are happy with it. 3.x will not be removed and we will continue to take patches for boards and fixes for 3.x series.

closed 2021-12-30 by ehristev

elockman, 2021-12-31:
@ehristev can you point me to a document that explains adding a board configuration for yocto builds in v4.x?

I've added **newboard** to the COMPATIBLE_MACHINE definition.

Next, I've added a **newboard_qspi_uboot_defconfig** file to the at91bootstrap/configs folder.

Finally, I've added to these definitions in the at91bootstrap.inc file
   **AT91BOOTSTRAP_CONFIG_newboard ??= "${AT91BOOTSTRAP_MACHINE}_qspi_uboot"
   AT91BOOTSTRAP_LOAD_newboard ??= "qspiboot-uboot"**

My build is failing with the following error:
ERROR: at91bootstrap-4.0.1+gitAUTOINC+905ba56436-r0 do_configure: **No config files found**

What am I missing?

ehristev, 2022-01-02:
@elockman unfortunately I am not very experienced with yocto. I suggest you open an issue here : 
https://github.com/linux4sam/meta-atmel
Someone with more yocto experience may help.

my simple suggestion is, have you changed the yocto at91bootstrap recipe to point to your repository, or add a patch, or how did you add the defconfig to the bootstrap project ?

elockman, 2022-01-03:
I forked the repo and added the files to my repo.  I'll open an issue in meta-atmel.

elockman, 2022-01-03:
Outside of a new COMPATIBLE_MACHINE definition, a defconfig file, and the changes to the inc file, are there any other changes required for a new board?

ehristev, 2022-01-03:
> Outside of a new COMPATIBLE_MACHINE definition, a defconfig file, and the changes to the inc file, are there any other changes required for a new board?

I don't know. Better ask this in meta-atmel

ehristev, 2022-01-03:
> I forked the repo and added the files to my repo. I'll open an issue in meta-atmel.

have you changed yocto recipe to download your repo instead of the original repo ?

elockman, 2022-01-03:
I am currently downloading my repo to build.

I'm a little unclear why you don't know what files are used to define a board in the new bootstrap.  That seems to be the main point of the changes to v4.x.  Maybe I'm not asking my question well.

It seems that only a new defconfig is created for a new board now, instead of the /contrib/subfolder. Is that correct?

ehristev, 2022-01-03:
> I am currently downloading my repo to build.
> 
> I'm a little unclear why you don't know what files are used to define a board in the new bootstrap. That seems to be the main point of the changes to v4.x. Maybe I'm not asking my question well.

A board in the new bootstrap is only a defconfig file. I do not know what COMPATIBLE_MACHINE or inc file you refer to.

> 
> It seems that only a new defconfig is created for a new board now, instead of the /contrib/subfolder. Is that correct?

That is true.