linux4sam/at91bootstrap #146: bootstrap on eMMC - 2GB (skyhigh v5.1 - SkyHighMemory_S40FC002_DS)

khaledkrayem, 2022-01-19:
i am trying to boot to bootstrap and u-boot on eMMC :

i am using a custom board - based on Sama5D2

RomBOOT
 ba_offset = 0xa ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf000c000: 0x10 0x300a21 0xc00034 0x2223a337
0xf000c010: 0x2220f0d 0x72362 0x33338 0x10000
0xf000c020: 0x16 0x50008 0x0 0x0
0xf000c030: 0x6 0x166404 0x0 0x0
0xf000c040: 0x0 0x0 0x0 0x0
0xf000c050: 0x0 0x0 0x0 0x1
0xf000c060: 0x0 0x0 0x0 0x0
0xf000c070: 0x0 0x0 0x0 0x0
0xf000c080: 0x0 0x0 0x0 0x0
0xf000c090: 0x0 0x0 0x0 0x0
0xf000c0a0: 0x0 0x0 0x0 0x0
0xf000c0b0: 0x0 0x0 0x1 0x0
0xf000c0c0: 0x0 0x0 0x0 0x0
0xf000c0d0: 0x0 0x0 0x0 0x0
0xf000c0e0: 0x0 0x0 0x0 0x4000
0xf000c0f0: 0x484d5044 0x44524320 0x0 0x20301
0xf000c100: 0x0 0x0 0x4040404 0x3030303
0xf000c110: 0x0 0x1 0x3c03c02 0x0
0xf000c120: 0x0 0x0 0xb00 0xb00
0xf000c130: 0xb00 0xb00 0xc00 0xc00
0xf000c140: 0xc00 0xc00 0xf 0xf
0xf000c150: 0xf 0xf 0xe 0x0

AT91Bootstrap 4.0.1-00006-g9a0208a4-dirty (2022-01-19 18:06:52)

All interrupts redirected to AIC
SD/MMC: Image: Read file u-boot.bin to 0x26f00000
MMC: ADMA supported
mmc_verify_operating_condition
mmc_verify_operating_condition success OCR = 0xc0ff8080
Card type is MMC
sd card identified with CID = 0x1010053 0x34303030 0x3210c960 0xfc9a38ff
sdcard_identification success
MMC: Specification Version 4.0 or higher
MMC: v5.1 detected
MMC: highspeed supported
MMC: Dual Data Rate supported
MMC: detecting buswidth...
MMC: 4-bit bus width detected
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image

ehristev, 2022-01-20:
Hi,

Have you written the u-boot.bin file correctly into the first partition of the eMMC ?
(and this partition is formatted with fat32 file system)
Where are you loading the AT91bootstrap from ? from the same eMMC device ?

khaledkrayem, 2022-01-20:
[image]
u-boot is written into '/dev/mmcblk1p1'
AT91bootstrap  is loaded from eMMC 
qml file:
config {
		sdmmcInstance: 1		
		sdmmcPartition: 1
		}

ehristev, 2022-01-20:
Hi,

Can you mount the SD-Card first partition on your computer and the list the contents please

ehristev, 2022-01-20:
Could you apply this change and retry ?

```
diff --git a/driver/sdcard.c b/driver/sdcard.c
index 284b6141..0130421d 100644
--- a/driver/sdcard.c
+++ b/driver/sdcard.c
@@ -62,8 +62,8 @@ static int sdcard_read_cmd(char *cmdline_file, char *cmdline_args)
 
        fret = f_open(&file, cmdline_file, FA_OPEN_EXISTING | FA_READ);
        if (fret != FR_OK) {
-               dbg_info("*** FATFS: f_open, filename: [%s]: error\n", 
-                                                                  cmdline_file);
+               dbg_info("*** FATFS: f_open, filename: [%s]: error %d\n",
+                         cmdline_file, fret);
                 ret = -1;
                goto open_fail;
        }
```

khaledkrayem, 2022-01-20:
[image]
[image]

khaledkrayem, 2022-01-20:
i am writing AT91bootstrap  on /dev/mmcblk1boot1
it seems that it cann't read/access from /dev/mmcblk1p1
[image]

ehristev, 2022-01-20:
For your information, AT91Bootstrap *cannot* read boot emmc partitions. So, it cannot read from mmcblkXbootY

Can you try to wipe mmcblk1boot1 and copy your boot.bin into mmcblk1p1 .
This way we can try to test if the first stage bootloader (ROM code) can read from mmcblk1p1 .

ehristev, 2022-01-20:
FR_NO_FILESYSTEM,               /* (13) There is no valid FAT volume */ 

it looks like you don't have any file system on this partition.
How are you connecting this to your laptop ? if it's an eMMC ?
isn't it soldered on your board ?

ehristev, 2022-01-20:
> Hi,
> 
> Can you mount the SD-Card first partition on your computer and the list the contents please

So this was stupid of me since you said you have an eMMC. 
Are you sure you formatted the eMMC with file system ?

khaledkrayem, 2022-01-20:
it an eMMC - soldered on my board
i formatted mmcblk1p1 to fat32 - using mkdosfs command
boot.bin is on mmcblk1p1, it stuck on:
RomBOOT

ehristev, 2022-01-20:
How did you format it ? are you booting Linux on the board ?

so it looks like neither ROMcode can load from user partition 1

khaledkrayem, 2022-01-20:
i boot linux on the board
i want to boot linux from eMMC not from flash
i want to put : at91bootstrap, ubootenv, u-boot, DTB , zImage on first partition (as a FAT32) and ROOTFS on second partition (as EXT4)

ehristev, 2022-01-20:
> can i put 'at91bootstrap, ubootenv, u-boot' on mmcblkXbootY for booting and then read (the rest) from mmcblk1p1/mmcblk1p2

No, because at91bootstrap cannot read from *boot* partition, only from *user* partition.
We have to understand why your partition does not work.

ehristev, 2022-01-20:
> i boot linux on the board i want to boot linux from eMMC not from nor flash i want to put : at91bootstrap, ubootenv, u-boot, DTB , zImage on first partition (as a FAT32) and ROOTFS on second partition (as EXT4)

Can you show me the boot log , to see how does Linux detect your eMMC . Linux detects it, so it should work .

khaledkrayem, 2022-01-20:
[image]

ehristev, 2022-01-20:
Okay, what is found as mmc0 ? Do you have another eMMC ? or what is it ?

khaledkrayem, 2022-01-20:
[image]

ehristev, 2022-01-20:
Linux allocates your eMMC as mmc1, so this means that it finds another mmc, and it allocates it as mmc0. This should be visible in the boot log.

khaledkrayem, 2022-01-20:
mmc0: new high speed SDIO card at address 0001

ehristev, 2022-01-20:
okay... so on mmc0

> mmc0: new high speed SDIO card at address 0001
> 
> do you want the full log?

okay so on mmc0 you have an SDIO device, got it.

ehristev, 2022-01-20:
Device       Boot StartCHS    EndCHS        StartLBA     EndLBA    Sectors  Size Id Type
/dev/mmcblk0p1 *  0,32,33     1,70,5            2048      20479      18432 9216K  c Win95 FAT32 (LBA)
#

ehristev, 2022-01-20:
Can you try to make your partition similar to mine.
Id to be 'c' first, and try again

khaledkrayem, 2022-01-20:
can't find /dev/mmcblk0 - it look like it failed to create it

khaledkrayem, 2022-01-20:
how could i create this partition?

khaledkrayem, 2022-01-20:
> try first to format with mkfs.vfat , and check if the id has changed. to recreate the partition, use fdisk, delete the partition then create a new one... there are plenty of tutorials on the internet. It doesn't come out of my head now, so it's best to find the options online

i mean om /dev/mmcblk0
i know how to create partitions :) - but i mean on this specific one (because i don't have it)
or you mean to create it on /dev/mmcblk1

[image]

khaledkrayem, 2022-01-20:
> 9216K

[image]

ehristev, 2022-01-20:
okay so after you do this, try to mkfs.vfat and copy the boot.bin and u-boot.bin there and give it a try

khaledkrayem, 2022-01-20:
i format it as FAT32 and put these files on it
didn't work :( 

[image]

ehristev, 2022-01-20:
I have no idea why. Maybe the user partition is blocked... however bootstrap will report that there is no file system on it. Maybe the partition table cannot be read correctly by bootstrap/ROMcode.

nirvann, 2022-01-21:
@khaledkrayem Hi,

> [image]

1. Why is UBI mentioned?
2. Is the e.MMC partitioned with a MBR (vs a GPT)?
3. Is the mmcblk0p1 partition formatted with FAT16 or FAT32 (vs exFAT)?

At this point I suspect a limitation of ChaN FatFs R0.09. Is it feasible to give a try to the [Software Package](https://github.com/atmelcorp/atmel-software-package/tree/master/examples/sdmmc_sdcard), which uses FatFs R0.12 instead?

khaledkrayem, 2022-01-21:
> @khaledkrayem Hi,
>
>
> 1. Why is UBI mentioned?
> 2. Is the e.MMC partitioned with a MBR (vs a GPT)?
> 3. Is the mmcblk0p1 partition formatted with FAT16 or FAT32 (vs exFAT)?
>
> At this point I suspect a [...]

Hi @nirvann 
You can see that i created a new one (as @ehristev asked from med)
mmcblk1p1is formatted with Fat32
Regarding MBR vs GPT, which one should i use?

nirvann, 2022-01-24:
Currently a MBR shall be used.
[FatFs Change Log](http://www.elm-chan.org/fsw/ff/updates.txt) says support for GPT requires FatFs R0.14 or greater.

khaledkrayem, 2022-01-24:
@ehristev : i think that ba_offset should be "ba_offset = 0xb ..." not "ba_offset = 0xa ..." (i want to use SDHC1)
although i config it:
CONFIG_SDHC=y
# CONFIG_SDHC0 is not set
CONFIG_SDHC1=y

but i got "ba_offset = 0xa ..."
1. is it ok?
2.  if not - what i should do to get ba_offset = 0xb ...?

ehristev, 2022-01-25:
Yes it could be that the partition table is in a format that it cannot be read by Romcode/at91bootstrap.

ehristev, 2022-01-25:
> @ehristev : i think that ba_offset should be "ba_offset = 0xb ..." not "ba_offset = 0xa ..." (i want to use SDHC1) although i config it: CONFIG_SDHC=y
ba_offset is related to DDR configuration, not SDHC.

khaledkrayem, 2022-01-25:
Hi @ehristev : i make the changes and got ba_offset = 0xb ...
but:
[image]

on u-boot still read from "mmc dev 0" instead of "mmc dev 1"

ehristev, 2022-01-25:
> on u-boot still read from "mmc dev 0" instead of "mmc dev 1"

This depends on how u-boot is configured. Maybe you are using a u-boot configuration that wants to read from mmc dev 0. I can't tell what you have in your U-boot devicetree to see what is on mmc0, and how is u-boot configured.
If both mmc's are configured in DT, then you have to change u-boot configuration to store the environment in mmc1 and try to boot Linux from mmc1.

khaledkrayem, 2022-01-26:
@ehristev 
i fix the issue and got this error:
[image]

ehristev, 2022-01-26:
@khaledkrayem  Honestly I have not seen this error before in U-boot, so I can't say what is wrong there. 
However, I see now that you are able to load U-boot from your MMC device ??
So your problem with bootstrap being unable to load u-boot.bin is fixed ?

ehristev, 2022-01-26:
> > @khaledkrayem Honestly I have not seen this error before in U-boot, so I can't say what is wrong there. However, I see now that you are able to load U-boot from your MMC device ?? So your problem with bootstrap being unable to load u-boot.bin is fixed ?
> 
> yes, i am booting u-boot from eMMC i put u-boot.bin on "/dev/mmcblk1p1"(i formatted it with FAT32) , and i put uboot.env file their too but i got this error

How did you make it work ? What was wrong previously ? For the whole discussion, the problem is that Bootstrap was unable to read u-boot.bin from your eMMC. How did you fix that ?

khaledkrayem, 2022-01-27:
> How did you make it work ? What was wrong previously ? For the whole discussion, the problem is that Bootstrap was unable to read u-boot.bin from your eMMC. How did you fix that ?

Hi @ehristev 
I formatted mmcblk1boot0 with Fat32
This give bootsrap the ability to read from it
I also create  a Fat32 partition on mmcblk1p1 and put their u-boot.bin and uboot.env (i fix the ddr too on bootstrap)
Now, i have an issue with reading mmc1 dev on uboot
Any ideas for what can cause this issue and suggestion for solving these issues?

ehristev, 2022-01-28:
bootstrap cannot read from mmcblk1bootX partitions.
Maybe you recreated mmcblk1p1 which bootstrap *can* read indeed.
So the problem was the user1 partition? or the partition table ?

About u-boot, you can open a new issue on u-boot project here : https://github.com/linux4sam/u-boot-at91 
meanwhile, if bootstrap is working, I am closing this issue.

closed 2022-01-28 by ehristev