linux4sam/at91bootstrap #148: make oldconfig infinite loop with SPI_IOSET

celledge, 2022-01-28:
If you try to use `make oldconfig` on a device that uses dataflash, but does not set SPI_IOSET_MAX (e.g. sama5d3xekdf_uboot_defconfig) you end up stuck in an infinite loop when reaching SPI_IOSET. No value can be accepted, not even the default of 0. This problem will cause the Yocto meta-atmel/at91bootstrap recipe to fail spectacularly by appending to an infinitely long log file until your disk fills up... (I caught it at 8GB before doing so, thankfully)

The culprit is in `driver/Config.in.dataflash`:
https://github.com/linux4sam/at91bootstrap/blob/9a0208a4cededad8730ff5e409925c64e6ecbe65/driver/Config.in.dataflash#L57-L62

That does the wrong thing if SPI_IOSET_MAX is undefined, and creates an unsolvable range (1, 0). It needs to be corrected to check for the existence of SPI_IOSET_MAX first:
>if SPI_IOSET_MAX
>...
>endif

This correction worked for me, but I don't know if there is better syntax.

ehristev, 2022-02-04:
Hi,

Thank you for your report. We will have a look and come back to you

ehristev, 2022-02-14:
Applied a fix. Thank you !

closed 2022-02-14 by ehristev