linux4sam/at91bootstrap #149: SAMA5D2 Xplained SD highspeed sd cards issue using at91bootstrap v4.0.1

afinfante, 2022-02-16:
the current code has some anomaly that I am investigating and wanted to report. The sd card used is a [Kingston canvas select plus 32 GB](https://www.kingston.com/unitedkingdom/en/memory-cards/canvas-select-plus-microsd-card). After the SW reboot of the card the SSBL is unable to load the UBoot as it happens in issues such as [44](https://github.com/linux4sam/at91bootstrap/issues/44), and [98](https://github.com/linux4sam/at91bootstrap/issues/98).

I attach the log obtained as well as the patch applied on this version to obtain the traces as follows.
```
AT91Bootstrap 4.0.1-dirty (2021-12-09 14:45:31)

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
  #0  SAMA5D2-XULT [AC1]      ATMEL-RFO
EEPROM: BoardDate | Year | Week
EEPROM:             2014    1

EEPROM: Board sn: 0xd300000 revision: 0x600000

SD/MMC: Image: Read file u-boot.bin to 0x26f00000
MMC: ADMA supported
SD: Card Capacity: High or Extended
sdcard identification success
SD: Specification Version 3.0X
Specification version reported
High speed device
Error checking switched function
Error switching to high speed capabilities
Error initializing mmc/sd
*** FATFS: f_open, filename: [u-boot.bin]: error code 3
SD/MMC: Failed to load image
```

```
diff --git a/driver/mci_media.c b/driver/mci_media.c
index ff2cce2..191d344 100644
--- a/driver/mci_media.c
+++ b/driver/mci_media.c
@@ -438,7 +438,10 @@ static int sd_switch_func_high_speed(struct sd_card *sdcard)
 
 	ret = switch_check_hs_busy_status_supported(sdcard, &support);
 	if (ret)
+        {
+                dbg_info("Error checking busy status supported\n");
 		return ret;
+        }
 
 	if (!support) {
 		dbg_info("SD: Not support hs function switch\n");
@@ -452,15 +455,23 @@ static int sd_switch_func_high_speed(struct sd_card *sdcard)
 				SD_SWITCH_FUNC_HS_SDR25,
 				switch_func_status);
 	if (ret)
+        {
+                dbg_info("Error setting operation mode\n");
 		return ret;
+        }
+
 
 	/* Check Switched function */
 	status = swap_uint32(switch_func_status[4]);
 	if ((status >> 24) & 0x01) {
 		sdcard->highspeed_card = 1;
 		return 0;
-	} else
+	}
+        else
+        {
+                dbg_info("Error checking switched function\n");
 		return -1;
+        }
 }
 
 static int sd_card_set_bus_width(struct sd_card *sdcard)
@@ -890,8 +901,12 @@ static int sdcard_identification(struct sd_card *sdcard)
 		 */
 		dbg_info("Unusable Card\n");
 		return -1;
-	} else
+	}
+        else
+        {
+                dbg_info("Unknow card error\n");
 		return ret;
+        }
 
 	sdcard->highcapacity_card = (sdcard->reg->ocr & OCR_HCR_CCS) ? 1 : 0;
 
@@ -936,7 +951,7 @@ static int sdcard_identification(struct sd_card *sdcard)
 
 	sdcard->read_bl_len = DEFAULT_SD_BLOCK_LEN;
 
-	dbg_very_loud("sdcard_identification success\n");
+	dbg_info("sdcard identification success\n");
 	return 0;
 }
 
@@ -985,11 +1000,17 @@ static int sd_initialization(struct sd_card *sdcard)
 		dbg_info("1.0 and 1.01\n");
 	}
 
+        dbg_info("Specification version reported\n");
+
 	if (host->caps_high_speed) {
-		if (sdcard->sd_spec_version != SD_VERSION_1_0) {
+		dbg_info("High speed device\n");
+                if (sdcard->sd_spec_version != SD_VERSION_1_0) {
 			ret = sd_switch_func_high_speed(sdcard);
 			if (ret)
+                        {
+                                dbg_info("Error switching to high speed capabilities\n");
 				return ret;
+                        }
 		}
 	}
 
@@ -1000,10 +1021,14 @@ static int sd_initialization(struct sd_card *sdcard)
 			host->ops->set_clock(sdcard, 25000000);
 	}
 
+
 	/* Change the bus mode */
 	ret = sd_card_set_bus_width(sdcard);
 	if (ret)
+        {
+                dbg_info("Error setting bus width\n");
 		return ret;
+        }
 
 	return 0;
 }
@@ -1132,14 +1157,21 @@ int sdcard_initialize(void)
 	/* Card Indentification Mode */
 	ret = sdcard_identification(sdcard);
 	if (ret)
+        {
+                dbg_info("Error identifying card\n");
 		return ret;
+        }
 
 	if (sdcard->card_type == CARD_TYPE_SD)
-		ret = sd_initialization(sdcard);
+                ret = sd_initialization(sdcard);
 	else
 		ret = mmc_initialization(sdcard);
-	if (ret)
+
+        if (ret)
+        {
+                dbg_info("Error initializing mmc/sd\n");
 		return ret;
+        }
 
 	return 0;
 }
diff --git a/driver/sdcard.c b/driver/sdcard.c
index b63149f..01fa931 100644
--- a/driver/sdcard.c
+++ b/driver/sdcard.c
@@ -25,7 +25,7 @@ static int sdcard_loadimage(char *filename, BYTE *dest)
 
 	fret = f_open(&file, filename, FA_OPEN_EXISTING | FA_READ);
 	if (fret != FR_OK) {
-		dbg_info("*** FATFS: f_open, filename: [%s]: error\n", filename);
+		dbg_info("*** FATFS: f_open, filename: [%s]: error code %d\n", filename, fret);
 		ret = -1;
 		goto open_fail;
 	}
@@ -62,8 +62,7 @@ static int sdcard_read_cmd(char *cmdline_file, char *cmdline_args)
 
 	fret = f_open(&file, cmdline_file, FA_OPEN_EXISTING | FA_READ);
 	if (fret != FR_OK) {
-		dbg_info("*** FATFS: f_open, filename: [%s]: error\n", 
-                                                                  cmdline_file);
+		dbg_info("*** FATFS: f_open, filename: [%s]: error code %d\n", cmdline_file, fret);
                 ret = -1;
 		goto open_fail;
 	}
```

Thank you in advance for your help @ehristev 

https://github.com/linux4sam/at91bootstrap/blob/646977aeace88b04041bf9e4c33d998a5a71ad3e/driver/mci_media.c#L459

afinfante, 2022-02-16:
I just checked that the same artifacts on another [sd](https://www.westerndigital.com/en-us/products/memory-cards/sandisk-ultra-uhs-i-microsd#SDSQUA4-032G-GN6MA) do work properly. Clearly it is due to the SD model itself, any ideas?

ehristev, 2022-02-16:
Hi @afinfante 

I would need this card to test myself , to be able to reproduce your issue. I will see if I can find it for purchase in a shop.
I would like to ask, your board is exactly sama5d2_xplained ? and you are trying on the SD slot which is connected to SDMMC1 ? or you are using a modified board.
It;s interesting to see why on the first boot you have no problems, but on a second boot it fails.
It is possible that the Card is being changed to a mode from which it does not recover on reset.
the SD-card has no reset line, and unless there is a power cycle on the Card, it is not reset at all when you do software reset, and it's just reinitialized.

ehristev, 2022-02-16:
Is your card SDHC or SDXC ?

afinfante, 2022-02-16:
Hello @ehristev, thank you for getting back to me so quickly. It is SDHC, specifically I have put the model above, I leave it reflected [here](https://www.kingston.com/unitedkingdom/en/memory-cards/canvas-select-plus-microsd-card) if it is helpful.

afinfante, 2022-02-16:
Hello again @ehristev  and thank you in advance,
I have left you the exact model and capacity in my previous comment. I bought it on amazon spain, but from what I've seen it is distributed normally in other countries.

That's it, it's exactly the same board. The plans are to make serveral modifications to it but for the moment it is as I received it. 
but for the moment it's exactly as I received it. Answering to your next question, and following the above, that is, it is connected to the `SDMMC1`.

The truth is that I've also found this behaviour quite curious, that's why I've decided to get in touch with you. 

That's, there is no power cycle, just a reboot ordered by SW. However in this same scenario with the other model of Sandisk that I left in the first post everything works properly It should be the same if that was happening right?

ehristev, 2022-02-16:
Well let me say this, sandisk is part of the original Secure Digital that invented the SD protocol... while Kingston, well.
There are many types of SDcards out there, and some of the vendors do not really respect the specs, or some vendors work fine even if the spec is 'bent' a little.
I have some kingston cards but not exactly your model. I have some older one 8 GB capacity, which work as far as I know.
I will try to order one card and try it.

Looking through your logs, it looks like the card refuses to switch to high speed mode.
Could you enable the 'debug very loud' option in menuconfig , just go to debugging and choose the one with the most verbosity. Then try to run again and copy the logs for me.
I need to see what kind of communication works with the card and which communication does not

afinfante, 2022-03-03:
Hello @ehristev, sorry for the delay.
During today I will leave you the log you asked me for a few days ago to evaluate it.

afinfante, 2022-03-03:
Hello @ehristev, I'm trying to get the traces from you by modifying the recipe, as I usually do when working with Yocto (I've customized it a bit to make it easier for me to switch between adding traces to my whole distro or removing them by simply putting HMI_DEBUG = 1 in my local.conf or commenting it out respectively. This way it's more convenient for me to look at both at91bootstrap and uboot traces, among others).

**at91bootstrap_4.0.1.bb**
```
require at91bootstrap.inc

LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://LICENSES/MIT.txt;md5=12c44f58fe16bd407f016e45950c2f3d"

COMPATIBLE_MACHINE = '(sama5d2-xplained|sama5d2-xplained-mmc|sama5d2-xplained-emmc)'

FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

SRC_URI = " \
   git://github.com/linux4sam/at91bootstrap.git;protocol=https;branch=at91bootstrap-4.x \
   ${@bb.utils.contains('HMI_DEBUG', '1', 'file://0001-f_open_error_report.patch', '', d)} \
   ${@bb.utils.contains('HMI_DEBUG', '1', 'file://debug.cfg', '', d)} \
"

PV = "4.0.1+git${SRCPV}"
SRCREV = "4d41296e9ae12379555fb46a941897e7264600a2"

S = "${WORKDIR}/git"

EXTRA_OEMAKE = 'CROSS_COMPILE=${TARGET_PREFIX} CC=${TARGET_PREFIX}gcc EXTRA_CC_ARGS="${TOOLCHAIN_OPTIONS}"'

AT91BOOTSTRAP_BIN_PATH = "${S}/build/binaries"

AT91BOOTSTRAP_CONFIG_PATH = "${S}/configs"

do_configure_append() {
   if [ ${HMI_DEBUG} -eq 1 ]; then
      bbplain "Debug settings will be added to ${S}/configs/${AT91BOOTSTRAP_TARGET}" 
      cat ${THISDIR}/files/debug.cfg >> ${S}/configs/${AT91BOOTSTRAP_TARGET}
      cml1_do_configure
   fi
}
```

**debug.cfg**
```
# CONFIG_DEBUG_INFO is not set
CONFIG_DEBUG_VERY_LOUD=y
```

However I see that even though the compile date changes in the at91bootstrap header and therefore my binary is being updated, I don't get the most verbose level. Is it possible that there is some anomaly when putting these traces in this version?  (I have already put traces correctly in the U-Boot in this way successfully)

afinfante, 2022-03-03:
Good afternoon @ehristev!
Here are the traces at the most verbose level for the case we talked about earlier. I got it from the reboot, which is when I was not able to load the U-Boot.

```
�RomBOOTg... [   58.543300] reboot: Restarting system
 ba_offset = 0xc ...

Dump DDRAMC Registers:
@address: 0x0 0x4 0x8 0xc
0xf000c000: 0x10 0x300511 0xd0035d 0x44439436
0xf000c010: 0x3001d1b 0x74000 0x33338 0x10000
0xf000c020: 0x4 0x50008 0x0 0xc852
0xf000c030: 0x40 0x746404 0x0 0x0
0xf000c040: 0x0 0x0 0x0 0x0
0xf000c050: 0x0 0x0 0x0 0x2
0xf000c060: 0x0 0x0 0x0 0x0
0xf000c070: 0x0 0x0 0x0 0x0
0xf000c080: 0x0 0x0 0x0 0x0
0xf000c090: 0x0 0x0 0x0 0x0
0xf000c0a0: 0x0 0x0 0x0 0x0
0xf000c0b0: 0x0 0x0 0x1 0x0
0xf000c0c0: 0x0 0x0 0x0 0x0
0xf000c0d0: 0x0 0x0 0x0 0x0
0xf000c0e0: 0x0 0x0 0x0 0x4000
0xf000c0f0: 0x484d5044 0x44524320 0x0 0x20301
0xf000c100: 0x0 0x0 0x4040404 0x3030303
0xf000c110: 0x0 0x1 0x3b03c01 0x0
0xf000c120: 0x0 0x0 0xb00 0xb00
0xf000c130: 0xb00 0xb00 0xc00 0xc00
0xf000c140: 0xc00 0xc00 0xf 0xf
0xf000c150: 0xf 0xf 0xe 0x0
BKP: backup_mode_resume, resuming = 0
Backup mode enabled

AT91Bootstrap 4.0.1-dirty (2021-12-09 14:45:31)

All interrupts redirected to AIC
EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
EEPROM Buff:
@address: 0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf
0x206670: 0x41 0x41 0x54 0x4d 0x45 0x4c 0x2d 0x52 0x46 0x4f 0xe 0x1 0x41 0x43 0x31 0x3e AATMEL-RFO..AC1>
0x206680: 0x0 0x0 0x53 0x41 0x4d 0x41 0x35 0x44 0x32 0x2d 0x58 0x55 0x4c 0x54 0x43 0xac ..SAMA5D2-XULTC.
  #0  SAMA5D2-XULT [AC1]      ATMEL-RFO
EEPROM: BoardDate | Year | Week
EEPROM:             2014    1

EEPROM: Board sn: 0xd300000 revision: 0x600000

SD/MMC: Image: Read file u-boot.bin to 0x26f00000
MMC: ADMA supported
mmc_verify_operating_condition
SDHC: Timeout waiting for command complete
SD: Card Capacity: High or Extended
sd card identified with CID = 0x27504853 0x44333247 0x60000009 0x60155ff
sdcard_identification success
SD: Specification Version 3.0X
*** FATFS: f_open, filename: [u-boot.bin]: error code 3
SD/MMC: Failed to load image
```

agen007niaga, 2022-09-02:
Ejen007niaga

elockman, 2023-08-28:
@ehristev is there a solution to this? Is there a list of approved SD cards that can be found on Amazon, or otherwise?

I am having a similar issue with a couple SD cards on the SAMA5d27-SOM1-EK1. I've downloaded multiple demos from linux4sam.org and the issue persists. I've also went back to look at different versions.

I thought I may have changed the boot fuses, so I bought a new SAMA5d27-SOM1-EK1 to make sure it was a virgin board.

```
RomBOOT

AT91Bootstrap 4.0.6 (2023-05-05 12:39:18)

SDHC: fix in place for SAMA5D2 SiP VDDSDMMC over-consumption errata
SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
SDHC: Error: No Card Inserted
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image
RomBOOT

AT91Bootstrap 3.10.2 (2021-01-15 07:29:56)

SDHC: fix in place for SAMA5D2 SoM VDDSDMMC over-consumption errata
SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
SDHC: Error: No Card Inserted
*** FATFS: f_open, filename: [u-boot.bin]: error
SD/MMC: Failed to load image

```
Can you point me to a supported SD card?

afinfante, 2025-04-26:
@elockman use any Sandisk or Samsung uSD, I have tried several with no problems. I recommend [this](https://amzn.eu/d/0O3QkyU) one.