linux4sam/at91bootstrap #152: Why no THUMB mode for SAMA5DX?

lnksz, 2022-02-28:
I came across this configuration, and don't understand, why THUMB is disabled for e.g. SAMA5D3X, even if its [datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/SAMA5D3-Series-Data-sheet-DS60001609b.pdf) mentions support for the thumb instructions set.

https://github.com/linux4sam/at91bootstrap/blob/2eb1a1efc00fd46db52a3c41b00cf79eaf96fd77/Kconfig#L206-L210

Could you explain? Thank you!

ehristev, 2022-02-28:
@noglitch  @nirvann @junminlin Do you have any knowledge of why thumb is not available for these MPUs ?
Thanks !

lnksz, 2022-02-28:
FYI: I forced `-mthumb` manually into the `CPPFLAGS` variable, and the compiled binary was ridiculously smaller.

- `-marm` : 65k (only 448 bytes left from the SRAM)
- `-mthumb` : 42k

Without any other changes to our code-base.

lnksz, 2022-02-28:
Also because of the default of our toolchain (linaro v6.5) being `-mthumb` we have been compiling and using the bootstrap for the SAMA5D35 board with the thumb instruction set (for at least 2-3 years now). I only noticed the change because of a toolchain update where the default may have been changed.

ehristev, 2022-03-01:
@lnksz Since you discovered this and you tested it on your sama5d3 based board, I suggest you create a patch that removes the restriction for sama5d3, such that thumb can be enabled for these boards.
Open a Pull Request and I will merge it.

lnksz, 2022-03-03:
Gladly. What I could identify as a possible source for this is 177af40cd745eb5ddda83bb282fddb94d8cc770a from 2013 which was aiming to turn off `-mthumb-interwork` through this.

That flag is still added to the flags through this option:
https://github.com/linux4sam/at91bootstrap/blob/97e5fe2193d87cd4a44b15d7ae28eea3b2416ac8/device/device_cpp.mk#L15-L17

So I will test whether with `-mthumb-interwork` our build still succeeds.

lnksz, 2022-11-23:
Hi there @ehristev, thanks for the ping, I will take care of this until the end of this year.

lnksz, 2022-11-24:
I verified it, and all still works for us with `-mthumb-interwork` as well.
PR follows shortly.

ehristev, 2022-11-25:
Thanks again for figuring this out. Applied your patch.

closed 2022-11-25 by ehristev

lnksz, 2022-11-25:
Thanks for the patience ;)