linux4sam/at91bootstrap #153: Debugging with Eclipse on Linux

lnksz, 2022-05-11:
Hi there!

I am trying to get more elaborate debugging experience to contribute to the at91bootstrap project.

I found a [tutorial on Linux4SAM](https://www.linux4sam.org/bin/view/Linux4SAM/AT91BootstrapDebugEclipse) on how to configure Eclipse CDT and a JTAG debugger to be able to set use GDB features.

But following the tutorial I noticed, that it is quite dated and some links don't work, or the Zylin plug-in fails to be installed.
(Using latest Eclipse 2022-03)

I also found the [jlink-plugin](https://eclipse-embed-cdt.github.io/debug/jlink/) which is relying on the cPack manager.

After some try-and-error I got the setup working so far that I can connect to my SAMA5A3Xplained board by clicking on Debug and the download of the built binary is successful.
But after the reset is issued the CPU starts running and the registers cannot be read out.

Anyway.. it may be a of a more general interest as well to have an updated version of the Linux4SAM tutorial to enable easier contribution to the project.

Can any of the maintainers help out? You surely have a working set-up at hand, right?

ehristev, 2022-05-11:
Hi,

> Can any of the maintainers help out? You surely have a working set-up at hand, right?

While I would like to say yes... I can only say no. I never used Eclipse with at91bootstrap, actually last time I used Eclipse was around 2011 I think...

I will ask around if someone tried it, but don't get your hopes up.

Eugen

lnksz, 2022-05-11:
Thanks! I have installed Eclipse only for this... So I if you have a different tool I am more then open for alternatives!

lnksz, 2022-05-17:
With all respect and knowing how non-critical this is, may I ask you if there is any news in the maintainer circle about this?
Thanks!

ehristev, 2022-05-17:
No news.
@noglitch  or @nirvann , @TonyHan11 @junminlin  maybe have some suggestions to share about how they work with debugging at91bootstrap

nirvann, 2022-05-17:
@lnksz, Hi,
The reference debugger is... GDB on the command line.
[MPLAB X IDE](https://www.microchip.com/mplab/mplab-x-ide) too may be used to debug AT91Bootstrap, but not with SAMA5D3x MPUs.
Regarding Eclipse, for sure it would be nice to spot the necessary updates to the tutorial.

The toolchain first should be updated to a newer version, though not necessarily the latest.
E.g. [GNU Arm Embedded Toolchain 8-2018-q4-major](https://developer.arm.com/-/media/Files/downloads/gnu-rm/8-2018q4/gcc-arm-none-eabi-8-2018-q4-major-linux.tar.bz2?revision=ab7c81a3-cba3-43be-af9d-e922098961dd?product=Downloads,64-bit,,Linux,8-2018-q4-major) (GCC 8.2.1).

The SEGGER J-Link GDB Server may be updated to the latest [J-Link Software Pack](https://www.segger.com/downloads/jlink/).
Typical options are:
`JLinkGDBServer -select USB -device Cortex-A5 -endian little -if SWD -speed auto -ir -LocalhostOnly`
If options aren't given to J-Link GDB Server as arguments, it offers a GUI to set them.
The _Target interface_ may also be set to the 4-wire _JTAG_, alternatively.
According to SEGGER, adaptive clocking is only available for ARM 7/9 targets, **not** for Cortex-A/R/M targets. Therefore, _Speed_ should be assigned _Auto Selection_ as above.
At this point it would help to start `arm-none-eabi-gdb`, and check it can start a debug session successfully.

As an alternative to Eclipse, one may give a try to [Visual Studio Code](https://code.visualstudio.com) with the [gnu-debugger](https://marketplace.visualstudio.com/items?itemName=metalcode-eu.gnu-debugger) extension.

lnksz, 2022-05-18:
Hi @nirvann ,

Thank you for the aid!

With the JLinkGDBServer settings and the linked toolchain's gdb I could establish a debugging session.

Following the instructions in the [Segger Wiki](https://wiki.segger.com/J-Link_GDB_Server#Debugging_with_J-Link_GDB_Server) for the GDB CLI I can enjoy the familiar GDB experience.

After this adding a different front-end is secondary for me.

closed 2022-05-18 by lnksz